Koenkk/zigbee2mqtt #4944: Lonsonho Zigbee 3.0 Tuya Smart Led Lamp Licht E27 220V Rgb + W + C Werkt Met Smartthings echo Plus Hub Alexa Google Thuis

ariekanarie2005, 2020-11-11:
## Information about the device + link

please add Lonsonho zigbee tuya smart led lamp rgb/w/c

https://nl.aliexpress.com/item/4001242163772.html?spm=a2g0o.productlist.0.0.ef7b604fi23ZaV&algo_pvid=cc06c2fd-5b05-43e4-8cf9-d7c13f32e084&algo_expid=cc06c2fd-5b05-43e4-8cf9-d7c13f32e084-1&btsid=0b0a119a16051076520816766ed409&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

## data/database.db entry of the device:
{"id":22,"type":"Router","ieeeAddr":"0x00124b00221d029c","nwkAddr":15503,"manufId":0,"manufName":"eWeLight","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1,13],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-CL01","manufacturerName":"eWeLight","powerSource":1,"zclVersion":1,"appVersion":7,"stackVersion":2,"hwVersion":1,"dateCode":"20200914","swBuildId":"1.0.7"}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":7,"stackVersion":2,"hwVersion":1,"dateCode":"20200914","swBuildId":"1.0.7","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1605043921015}

thanks in advance,
..

fufar, 2020-11-30:
got same lamp, you manage to create supportive devices.js section?

ariekanarie2005, 2020-11-30:
No, i tried to remove the paired device in device.js. I let it pair again but still no devices reported.
I hope someone can help us, some lonsonho devices are already supported in zigbee2mqtt.

fufar, 2020-12-01:
My seller answer me that i should try to connect it as https://www.zigbee2mqtt.io/devices/rgbw2.zbee27.html?spm=a2g0s.imconversation.0.0.22cc3e5fxuEG0W

so i add to device.js
```
    {
        zigbeeModel: ['ZB-RGBCW'],
        model: 'rgbw2.zbee27',
        vendor: 'Lonsonho',
        description: 'RGBW LED bulb with dimmer',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
```
lamp worked now 
maybe it will helpful for your case too

Hacker-CB, 2020-12-21:
I have same Lonsonho RGB bulb and currently z2m recognize it as `YSR-MINI-01_rgbcct` dimmer. 
* [AliExpress link](https://aliexpress.com/item/4001242163772.html)
* [Current converter source code link](https://github.com/Koenkk/zigbee-herdsman-converters/blob/4ec1778b372ab3cd9063140ad1d26d181324ff5a/devices.js#L6735)

For my Lonosnho RGB bulb Manufacturer is `eWeLight` and modelId is `ZB-CL01`.

I can add next converter but I'm not sure If it can break existing YSRSAI dimmer, because I don't know its `manufacturerName` or other fingerprint details:

```javascript
    {
        fingerprint: [{modelID: 'ZB-CL01', manufacturerName: 'eWeLight'}],
        model: 'ZB-CL01_lonsonho',
        vendor: 'Lonsonho',
        description: 'Smart Led Lamp Bulb Light E27',
        extend: preset.light_onoff_brightness_colortemp_colorxy,
    },
```

@Koenkk How do you think, which is better way to split this bulb from dimmer?

P.S. It will be better to not accept PR without fingerprint anymore ;)

Koenkk, 2020-12-21:
@Hacker-CB this is fine, fingerprint is preferred over the zigbeeModel, so if the modelID is ZB-CL01 and manufacturerName is eWeLight it will use this definition.

Hacker-CB, 2020-12-21:
@Koenkk i know this, but I don’t know `manufacturerName` of the existing existing YSRSAI dimmer. I don’t have it and can’t test it.
May be it is also same as in Lonsonho bulb (eWeLight)...

Koenkk, 2020-12-21:
@Hacker-CB true, asked for it here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1379

Koenkk, 2020-12-22:
@Hacker-CB  manufacturer name is `YSRSAI` so good luck here. Can you make a pr? (also use a fingerprint for the ZB-CL01 please)

`{"id":6,"type":"Router","ieeeAddr":"0x00124b001cd6a473","nwkAddr":39018,"manufId":0,"manufName":"YSRSAI","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1,13],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-CL01","manufacturerName":"YSRSAI","powerSource":1,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":1,"dateCode":"20200507","swBuildId":"1.0.6"}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":6,"stackVersion":2,"hwVersion":1,"dateCode":"20200507","swBuildId":"1.0.6","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1608617396877} `

Hacker-CB, 2020-12-22:
@Koenkk, I made PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/1958

Koenkk, 2020-12-22:
Merged, assuming  this can be closed now.

closed 2020-12-22 by Koenkk