Koenkk/zigbee2mqtt #4949: Ikea Tradfri LED1903C5/LED1835C6 won't pair (nothing in logs)

nielsrune, 2020-11-11:
## What happened
I'm switching from tradfri gateway to zigbee2mqtt. All other bulb has been paired but my single Tradfri LED1835C6.
Nothing at all is shown in the log of zigbee2mqtt.
I have followed the suggestions at https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all
The bulbs firmware was up to date on the Tradfri Gateway.

## What did you expect to happen
Expected the bulb to start pairing.

## How to reproduce it (minimal and precise)
Try to pair new (< 1 month old) Ikea Tradfri LED1535C6.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1 (Electrolama zig-a-zig-ah, newly bought, latest firmware CC26X2R1_20201026.hex)
Adapter firmware version: CC26X2R1_20201026.hex

System is a new Raspberry Pi 4 model B running Rasberry Pi OS.
Home Assistant Supervised 0.117.5

Tried Touchlink scan. At first run nothing is found. Further runs fails with Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

io53, 2020-11-15:
The trådfri e14 is very picky with the timings when resetting. Try to wait 1 full second (or a bit more) in the off state, and keep the on state very short when doing the reset procedure.

nielsrune, 2020-11-15:
Good news and bad news:

I tried one more according to io53s suggestion waiting a little more in the off state. As before, I could get the bulb to glow, ie. paring mode. But the bulb would still not show up in zigbee2mqtt.

But then I tried - more or less in a desperate action - to switch the config to:
```
experimental:
  new_api: true
```
Then after doing the reset procedure, the bulb immediately paired.

I'm still new to the whole smart home thing, so I havn't a clue yet as to the difference in old vs new api. But it seems like a bug or at least a quirk, that maybe should be put on
https://www.zigbee2mqtt.io/devices/LED1903C5_LED1835C6.html

hdoedens, 2020-11-17:
I am having the exact same issue. Two brand new TRADFRI lamps (bulbs). I have reset the bulbs more than ten times now. I place it very close to the coordinator but there is no logging, and no paring. 
I tried the suggestion above: disabling the `new_api` but no success. I guess @nielsrune was lucky?

I think I have tried every suggestion now but nothing is working. What could I try or how can I see what is going on? Debug logging of the zigbee2mqtt container is not showing anything.

I have many more TRADFRI devices, remotes and lamps, they all pair at first try.

waldher, 2020-11-22:
I'm having similar issues. I'm able to pair a brand new LED1835C6 with the Tradfri Gateway, with which I've updated its firmware to 2.3.050, but I'm not able to pair it with zigbee2mqtt, despite having bulbs in the past do this without issue. I've also tried two different LED1835C6's now, so it's not isolated to one bulb.

waldher, 2020-11-22:
An update on my previous comment: At first, setting
```
experimental:
  new_api: false
```
in the HomeAssistant plugin configuration didn't save. After setting it again, I can confirm that it does indeed pair right away after a bulb reset using firmware 2.3.050. (If you don't know what firmware you have on your bulb, you'll need to first pair the bulb with a Tradfri Gateway to update it)

MattWestb, 2020-11-22:
I was have the same problems some time. First the bulb not being reseted but working with little tryings.
Some time its timing out after being reseted and starting one own networks as zigbee lights is doing (LL and Zigbee 3 standard). Its normal on channel 11 with default pan-ID then touchlink scanning from deCONZ. Reseting it with touchlink can helping getting it being detected.
The large problem is if not having the default LL channels in the config (11. 15. 20 and 25 is moderate for lights in LL and zigbee 3) then it can the coordinator is not scanning the channel and not finding it. Theoretically all channels should working but using one of the primary ones its making it more successful then working with lights.

nielsrune, 2020-11-22:
I'm glad to hear @waldher got his bulbs working in the same way.

I don't recall my procedure exactly, but it may have been the same:
- Set new_api to false and restart zigbee2mqtt and wait for it to settle (the frontend is not working with this setting), so the log page in the zigbee2mqtt is the place to monitor.
- Do the reset procedure on the bulb again. I'd rather count to 7 than missing out on cycle.
- If done right and the bulb is glowing, refresh the log. My bulb started to pair right away.  

@hdoedens would you care to try again, as it seems I wasn't just lucky..

Also, I can confirm that my firmware is 2.3.050 and the bulb was previously connected to a Trådfri gateway. Also, try to remove the bulb from the gateway before resetting. 

@MattWestb My dongle is the zig-a-zig-ahh. Before getting the bulb to pair, I tried both scanning and resetting over touchlink. But it seems to not work with the dongle. The seller, Omar, wrote something like he too didn't got touchlink to work. 
But I can see that I have `channel: 11` in the configurations advance section. I didn't touch the default config but setting a custom `network_key`

MattWestb, 2020-11-22:
What i have reading that is some problems with the firmware of all TI coordinators that is making TL not working all the time. Then its working its being very nicely implanted in the web front end.

hdoedens, 2020-11-22:
I tried the procedure as described by @nielsrune again, several times. I completely stopped and started zigbee2mqtt to be sure sure the new settings would have effect. As soon as the log was quiet I reset the Tradfri bulb, but nothing showed up in the log and the bulb keeps glowing. When I click the a Tradfri remote a few times, that will bind to the bulb immediately and the bulb respons to the Tradfri remote. Still nothing in the log (but that makes sense as it is not paired with zigbee2mqtt).

Maybe this has something to do with my settings or the version I am running. I'm running the edge version as the binding of remote to Zigbee devices is broken in the stable version. This is my config:
```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: *******
  password: ***********
serial:
  port: /dev/ttyACM1
advanced:
  log_level: debug
  pan_id: 6755
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: false
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
```
Note the non-default `pan_id`. As I was having trouble to pair right from the start and I migrated some software but used the same hardware I thought it would help to change the `pan_id` to have a fresh start.
I don't know the firmware version of the bulb's that won't pair as I do not have an IKEA bridge.
The hardware I am using is a Texas Instruments CC2531.

Any help is more than appreciated...

MattWestb, 2020-11-22:
@hdoedens The PAN-ID is only the "name" of the network and one resetted device is looking for all PAN-IDs if not having one hard coded from the factory (closed system) so its very OK.
Channel 11 is the first channel lights is looking at for joining and also the best if having problems.

I have only resetting the light more times until they was pairing but it was not in Z2M.

I have reading that one Z2M user with CC-2531 was enabling debug logging and the pairing wos start working and disabling it it was stopping working. You can trying it and see if its working.
Touch link resetting of the bulb can also being worth testing. And you have the new web frontend installed so testing how and if its helping.

How large is your CC2531 system ? Normally they is working up to 30 - 40 devices then hitting the wall ;-(

hdoedens, 2020-11-22:
@MattWestb Glad to hear my settings are OK :) I have tried the Touchlink several times, but I think that only works with Philips devices.
I already tried with and without debug logging on. No success...
I currently have 11 devices paired with the CC2531.

One question: How close do you have the bulb to the coordinator when pairing? I have tried both very close (a few centimeters away) en just met them hang on the ceiling.

MattWestb, 2020-11-22:
Touchlink is working on all IKEA devices. I was trying resetting one IKEA E27 WW with deCONZ and wast taking the wrong one and was resetting one IKEA GU10 WS in my kitchen 6 meters away that is paired with my IKEA GW :-(( 
Philips HUE is TL only working after being re powered if having newer firmware.

If doing touchlink pairing (that you is not doing) you need being very near but with classical pairing (that's what Z2M is doing) its good being near but its normal communication so normal working range suld being OK bet nearer is better / safer.
Your network is not too large for your CC-2531 and you have no strange things in your config that is making the world bad for it. The CC-253X have normally pretty bad radio and antenna but if you is near it should working without problem but i knowing that's not the fact in this case.

You can trying doing the normal 6 x on - of with short on and long off and then the bulb is blinking you is power it off for some seconds and then on and see if you can pairing it (some saying its need being repowered after being reseted for being in pairing mode).

nielsrune, 2020-11-23:
> When I click the a Tradfri remote a few times, that will bind to the bulb immediately and the bulb respons to the Tradfri remote. Still nothing in the log (but that makes sense as it is not paired with zigbee2mqtt).
> @hdoedens 

Maybe the bulb is somehow still paired to your remote.
Try to reset the remote with four clicks on the button next to the battery and wait 10 seconds. Maybe even remove the battery from the remote before doing the reset of the bulb.
Just to make sure the remote is detached from the bulb.

MattWestb, 2020-11-23:
If the remote (that was touchlink pared with the bulb) can controlling the bulb then the bulb was not resetted and have all network settings from the remote stored. One bulb that is not reserted (in zigbee term not factory new) can not joining one classical (with trustcenter) zigbee network. Only with touchlink network can stealing it and force it joining its to one new network.

hdoedens, 2020-11-24:
- reset the remote
- confirmed that the remote could not control the bulb anymore
- removed battery from to be sure
- reset bulb with 6 on/offs, while `new_api: false` and debug logging on
nothing...

Because I have no GUI at the moment I tried to factory reset the bulb by placing it within centimeters of the coordinator and sending `zigbee2mqtt/bridge/config/touchlink/factory_reset` whiteout payload:
```
Zigbee2MQTT:info  2020-11-24 11:45:53: Starting touchlink factory reset...
Zigbee2MQTT:info  2020-11-24 11:45:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"reset_started","meta":{"status":"started"},"type":"touchlink"}'
Zigbee2MQTT:warn  2020-11-24 11:46:05: Failed to factory reset device through Touchlink
Zigbee2MQTT:info  2020-11-24 11:46:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"reset_failed","meta":{"status":"failed"},"type":"touchlink"}'
```
I have tried to reset the bulb using touchline many times both while first resetting it using the on/off method (while it was glowing) and also when it was just on.

hdoedens, 2020-11-24:
One thing I notice: After I have reset the bulb I still can control it using the remote?!

Had the reset procedure changed on new bulbs?

hdoedens, 2020-11-24:
I think I've got it: I was reading a bit closer for the reset procedure. As some mention to click the bulb 7-times off and on 'to be sure to reset it'... that 7th time actually ruins the reset procedure. You have wait IN the 6th on state to reset it properly. Once I did that the bulb paired immediately!

Maybe good to add this little detail to the docs of the Tradfri devices.

Thanks for the help everyone!

MattWestb, 2020-11-24:
I think if you can still control the bulb with the remote both have not being reseted.
Then resetting one remote (pressing pairing button x 4 and led is blinking for some seconds) its making one new network if not being paired with one GW.
The network key and pan-id is being random generated so not colliding with other remotes around.

Great that you was getting the bulb resetted so you cud getting it paired !!!

hdoedens, 2020-11-24:
After all it had nothing to do with the new front-end as the second bulb I paired was with the `new_api` setting to `true`

closed 2021-01-01 by github-actions[bot]

stevenbrown1862, 2022-01-08:
> 

This is absolutely right.   I've spent an afternoon switching the light to make it reset and I always did that 7th time to make sure.   Whatever you do stick to the 6 times exactly.

JacekKac, 2022-10-05:
hi guys, i have the same problem as you, don't know what to do now. 
I am able to put the bulb into pairing mode: fast blinking bulb, but in zigbee2mqtt nothing happens after allowing devices to join.

hdoedens, 2022-10-07:
Have you upgraded your zitbeen stick tot the latest firmware?

cjdell, 2024-10-31:
Giving this a bump as just spend the last 2 hours trying to pair 2 bulbs (model number LED2101G4). Even my trusty Philips Hue remote which I have kept around *purely* as an instrument for re-pairing stuff wasn't working (but did make the lights flash slowly which confused me even more).

As others have said, flick ON for 5 times (about 1 second OFF in between) then leave ON the 6th time and they will flash *quickly* for several seconds and pairing will now work.

Nebula83, 2024-11-14:
Had the same experience today: switch seven times and glowing but no pairing. Switch 5 times amd a short flash and rapid glowing with an immediate pair. @Koenkk Can we add this to the documentation? I'd be willing to open up an MR for it, but notnsire if desirable.

Koenkk, 2024-11-16:
@Nebula83 yes, feel free to make a pr!