Koenkk/zigbee2mqtt #4950: Hue Bulb switch ON and last color state

ramon18, 2020-11-11:
## What happened
Philips Hue light bulbs with a command to turn on on a certain color, it turn ON using the previous color then switches to the requested color. Easier to verify with a transition = 5.

## What did you expect to happen
I was expecting it to apply the color and then turn on the light without flash/transition from the previous color.
This is my expectation because the bulb was previously turned OFF.

Maybe I'm doing something wrong, but could not find any info.
Btw, tested the scene_add + scene_recall and it works as expected (ie, it fades from "black" to the color stored in the scene).

## How to reproduce it (minimal and precise)
Step 1:
`{
  "state": "ON",
  "color": "#ff0000"
}`

Step 2:
`{
  "state": "OFF"
}`

Step 3:
`{
  "color": "#00ff00",
  "transition": 5,
  "state": "ON"
}`

## Debug info
Zigbee2MQTT version: latest dev branch
Adapter hardware: Conbee II

ramon18, 2020-11-22:
I started to investigate this issue by myself.
Digging the sources while debugging I found an interesting part in publish.js with a comment:

``// Not all bulbs support setting the color/color_temp while it is off``

This may be the reason why z2m is behaving like it is now.
In fact after removing the property sort (in publish.js), hue LCT015 bulbs completely ignores the set color, because they are off.

The question is, how is Hue Bridge capable of sorting this out?
Setting { color, brightness: 1, transition: 0 } in one initial payload, then do the final transition to the desired brightness, sounded link a good plan, but unfortunately it does not work :S

Anyone has suggestions?

sjorge, 2020-11-22:
I spend a bit of time looking into this, form my tests only trådfri supported setting color and color_temp while off, but a recensent firmware update removed that there too :(

Currently only my FLOAT panels still do this, but my updates guntarp stopped working after an update a few months ago.

So not sure what the current state is. I was not aware hue supported it with their own bridge though, like you I tired various things like setting brightness to 1, setting the color,... without success. I wonder if hue is using some custom extension that z2m does currently not implement.

sjorge, 2020-11-22:
Something I didn’t test because we didn’t have support when I did my testing is scenes...

Does a bulb allow a scene to he added when off? If so, we could add ‘startup’ scene with the color, brightness and stage we want. Wait for it to be store and then execute the scene.

If I don’t forget I will try this tomorrow.

ramon18, 2020-11-22:
Yes, scenes add/remove works while bulb is off, but the problem is how to manage scene IDs, etc
Not sure if we are able to always use ID=1 and it just overwrite the scene payload.

I had done a basic test, add a scene, then recall it. It work smooth, but carry an extra command and I noticed that's slower than usual.

My last resort plan is to add lots of scenes, managed a table of ids <-> hue in my application, then find the nearest hue and use that scene to "startup" the bulb/group, at last play the final transition to the desired color and brightness.
But that's a lot of work :)

Still investigating, I'll post here if found anything useful.

sjorge, 2020-11-22:
I though we could specify an ID with scene_add but didn’t check, didn’t think that it would be slowler, but I certainly might be on a busy network.

I guess if someone has the hub sniffing what it is doing is also a good thing to try.

ramon18, 2020-11-22:
After some more tests and dirty hacks, I found that ID reuse is perfectly accepted.
Work nicely with groups too.

Slomo17, 2020-12-22:
I have the same problem, i tried using scenes in two ways, one is saveing the set state of the lights in a file and loading it again, this works so far, but it leads to the descirbed effect, that when turning on, the last color, set bevor turning the lights off, is first loaded and than switched. Then i tried using the zigbee2mqtt scene implementation, here it works as it should but i noticed a different problem. Transition times below 1 seconds doesn't work here. It always  turn on instalntly, but i would like a time around 0.5 seconds. I don't really understand how you solved the problem with the ID reuse you discribed.

I have a Hue bridge if that helps, but i don't know how to sniff whats it's doing.

ramon18, 2020-12-22:
What I mean by group ID reuse is nothing fancy, you just publish scene_add over and over with same ID and it works naturally.

Slomo17, 2020-12-23:
Sorry, i still don't understand. You add a scene with scene_add right befor you recall the scene with recall_scene? But then i have the problem with the transition time, where values between 0 und 1 seconds doesn't work (gets rounded to 0). I compared it with the hue bridge, where the default transition times seem to be 0.4 seconds, which i would prefere, i think 1 second is a bit to slow.

sjorge, 2020-12-23:
I have a rather complex flow that does the following ( written in human understandable below ):
- if first bulb in group == on:
  - send  state/color/brightness + transition
- if first bulb in group == off
  - send scene_add with state=off, transition=0, color=whatever_i_want, brightness=whatever_i_want
  - sleep ~ 500ms - 1sec (depending on size of group)
  - send scene_recall
  - sleep ~ 250ms
  - send state=on, transition=0.5

So I'm only going through scene_add when the bulb/bulbs in the group are currently off
But I do not turn on the bulb in the scene_add, I set state=off to keep the bulb of, it will however change brightness and color.

Then I turn it on with the wanted transition afterwards.

Slomo17, 2020-12-23:
Ah you can recall a scene with state off, didn't tied this, now i get it. Thanks, i will try your approach.

Slomo17, 2020-12-23:
But what i don't understand. If its possible to set a scene with scene_add when off, souldn't it be possible to also do this via the "normal" object send to the bulb.

sjorge, 2020-12-23:
All bulbs aside very old trådfri seem to ignore everything when off, aside from scenes

~ sjorge

Slomo17, 2020-12-23:
I mostly hav hue bulbs.
I tried your suggestion, it works, but with scene_add you can only add scenes where all the bulbs have the same color and brightness, or am i missing something? A workaround would be to give every lamp its own group.

sjorge, 2020-12-23:
You do not need to use groups though,

If you just want to control one bulb, just add the scene to the bulb instead of the group.
You can do it mixed too

have a group with 4 bulbs, scene_add/recall on the group will effect all 4. 
scene_add/recall to 1 of the bulbs in the group, will only effect the group.

Slomo17, 2020-12-23:
ah good to know, thanks

closed 2021-01-31 by github-actions[bot]