Koenkk/zigbee2mqtt #4951: Woox R7054 Smart Security Remote Control

AntonioDiGennaro, 2020-11-11:
## Information about the device + link
Woox R7054 Smart Security Remote Control
https://wooxhome.com/products-c10/security-c6/woox-r7054-smart-remote-control-p53

## data/database.db entry of the device
{"id":15,"type":"EndDevice","ieeeAddr":"0xbc33acfffe289a01","nwkAddr":14441,"manufId":4098,"manufName":"_TZ3000_p6ju8myv","powerSource":"Battery","modelId":"TS0215A","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[0,10,1,1280,1281],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0215A","manufacturerName":"_TZ3000_p6ju8myv","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1605119577468}

out-of-the box it is recognized as a Tuya SOS but in that case obviously only the Emergency button works. the Arm Away, Arm Home and Disarm buttons do not work.

I have tried to modify the device.js file and by using this

    {
        zigbeeModel: ['TS0215A'],
        model: 'R7054',
        vendor: 'Woox',
        description: 'Smart security remote controller',
        supports: 'action',
        fromZigbee: [fz.command_arm, fz.command_emergency, fz.battery],
        toZigbee: [],
        exposes: [e.battery(), e.action(['disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'invalid_code', 'emergency'])],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        },
    },    

I get this in the logs for one click on the Arm button (seems like messages are duplicated)
Zigbee2MQTT:debug 2020-11-11 22:11:10: Received Zigbee message from '0xbc33acfffe289a01', type 'commandArm', cluster 'ssIasAce', data '{"armmode":3,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-11 22:11:10: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"arm_all_zones","action_code":"","action_zone":255,"linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:11:10: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"","linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:11:10: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01/action', payload 'arm_all_zones'
Zigbee2MQTT:debug 2020-11-11 22:11:11: Received Zigbee message from '0xbc33acfffe289a01', type 'commandArm', cluster 'ssIasAce', data '{"armmode":3,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-11 22:11:11: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"arm_all_zones","action_code":"","action_zone":255,"linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:11:11: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"","linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:11:11: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01/action', payload 'arm_all_zones'

clicking once on arm at home I get the messages 3x
Zigbee2MQTT:debug 2020-11-11 22:13:02: Received Zigbee message from '0xbc33acfffe289a01', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-11 22:13:02: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:13:02: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"","linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:13:02: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01/action', payload 'arm_day_zones'
Zigbee2MQTT:debug 2020-11-11 22:13:03: Received Zigbee message from '0xbc33acfffe289a01', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-11 22:13:03: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:13:03: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"","linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:13:03: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01/action', payload 'arm_day_zones'
Zigbee2MQTT:debug 2020-11-11 22:13:04: Received Zigbee message from '0xbc33acfffe289a01', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-11 22:13:04: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"arm_day_zones","action_code":"","action_zone":255,"linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:13:04: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01', payload '{"action":"","linkquality":118}'
Zigbee2MQTT:info  2020-11-11 22:13:04: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe289a01/action', payload 'arm_day_zones'

any suggestion?
thank you

AntonioDiGennaro, 2020-11-17:
Hello
in the end it works as I have tried above. I could set up simple automation to control the HA alarm panel via the remote controller.
Can anybody help on what is the best way to add support it in z2m and distinguish it from Tuya SoS button?

Koenkk, 2020-11-18:
What is logged in the Zigbee2MQTT debug logging when pressing the arm away, arm home and disarm buttons?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

AntonioDiGennaro, 2020-11-18:
arm_away
Zigbee2MQTT:debug 2020-11-18 13:24:53: Received Zigbee message from 'Woox_SmartSecurityRemoteControl_1', type 'commandArm', cluster 'ssIasAce', data '{"armmode":3,"code":"","zoneid":23}' from endpoint 1 with groupID 0

arm_home
Zigbee2MQTT:debug 2020-11-18 13:27:40: Received Zigbee message from 'Woox_SmartSecurityRemoteControl_1', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code":"","zoneid":23}' from endpoint 1 with groupID 0

disarm
Zigbee2MQTT:debug 2020-11-18 13:25:14: Received Zigbee message from 'Woox_SmartSecurityRemoteControl_1', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code":"","zoneid":23}' from endpoint 1 with groupID 0

Koenkk, 2020-11-18:
Please apply the changes from: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1780/files and check if it works

AntonioDiGennaro, 2020-11-18:
Yes it works except for battery level detection
I tried to modify device.js in the configure: part like this

    {
        fingerprint: [{modelID: 'TS0215A', manufacturerName: '_TZ3000_p6ju8myv'}],
        model: 'TS0215A_remote',
        vendor: 'TuYa',
        description: 'Security remote control',
        supports: 'action',
        fromZigbee: [fz.command_arm, fz.command_emergency, fz.battery],
        exposes: [e.battery(), e.action(['disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'invalid_code', 'emergency'])],
        toZigbee: [],
        whiteLabel: [
            {vendor: 'Woox', model: 'R7054'},
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genTime', 'genBasic', 'ssIasAce', 'ssIasZone']);
        },
    },

and it works.
thanks for your help.
I am new to zigbee and z2m... be clement :)

Koenkk, 2020-11-19:
Thanks, added the device. 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-19 by Koenkk

lawful76, 2024-10-02:
I had some trouble with adding R7054. There's a new controller for _TZ3000_jwcixnrz.

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const bind = require('zigbee-herdsman-converters/lib/reporting').bind;
const e = require('zigbee-herdsman-converters/lib/exposes').presets;
const exposes = require('zigbee-herdsman-converters/lib/exposes');

let lastAction = null;
let lastActionTime = 0;
const debounceTime = 3000;

const fzLocal = {
    TS0215A_action: {
        cluster: 'ssIasAce',
        type: ['commandArm', 'commandEmergency'],
        convert: (model, msg, publish, options, meta) => {
            const actionLookup = {
                0: 'disarm',
                1: 'arm_day_zones',
                2: 'arm_night_zones',
                3: 'arm_all_zones',
                255: 'emergency',
            };
            let action = null;
            const now = Date.now();

            if (msg.type === 'commandArm' && msg.data && msg.data.armmode !== undefined) {
                action = actionLookup[msg.data.armmode];
            } else if (msg.type === 'commandEmergency') {
                action = 'emergency';
            }

            if (action) {
                if (lastAction !== action || (now - lastActionTime) > debounceTime) {
                    lastAction = action;
                    lastActionTime = now;
                    return {action: action};
                } else {
                    // Ignore dublicate
                    return null;
                }
            } else {
                return null;
            }
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS0215A', manufacturerName: '_TZ3000_jwcixnrz'}],
    model: 'R7054',
    vendor: 'TuYa',
    description: 'Security remote control',
    fromZigbee: [fz.battery, fzLocal.TS0215A_action],
    toZigbee: [],
    exposes: [
        e.battery(),
        e.battery_voltage(),
        e.action([
            'disarm',
            'arm_day_zones',
            'arm_night_zones',
            'arm_all_zones',
            'emergency',
        ]),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'ssIasAce']);
        try {
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
        } catch (error) {
            logger.warn(`Can't read battery attibutes: ${error}`);
        }
    },
};

module.exports = definition;
`

Koenkk, 2024-10-03:
@lawful76 
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

lawful76, 2024-10-15:
> @lawful76 Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

I'd love to but I've no idea how to create a poll request. After clicking it offers me to edit tuya.ts file. Do I have to replace it with my external converter and commit?

Sorry for late reply.

Koenkk, 2024-10-16:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.