Koenkk/zigbee2mqtt #4952: Niko connected outlets turns off/on by aqara opple button press even if it shouldn't

golddragon007, 2020-11-11:
## What happened
I do a single press some of my aqara opple button 1 and all of my Niko connected outlets turns off. I press on the same device the button 2 and all my Niko connected outlets turn on, even if the coordinator is OFFLINE!
## What did you expect to happen
To generate a proper event instead of killing everything. I do not have such a rule (at least I did not create it).
## How to reproduce it (minimal and precise)
I have 22 Niko connected, 8 aqara opple, 6 mija light sensor and 4 TS0601_thermostat currently in the system, I confirmed 2 aqara opple shuts off/on everything, 2 doesn't.
## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: current
Here is the log: [syslog_cut.zip](https://github.com/Koenkk/zigbee2mqtt/files/5527135/syslog_cut.zip)
This is one of the problematic switches:
```
Nov 11 22:52:00 raspberrypi 7e200bde8e39[660]: #033[32mZigbee2MQTT:info #033[39m 2020-11-11 22:52:00: bedroom1_shutter (0x04cf8cdf3c77c750): WXCJKG12LM - Xiaomi Aqara Opple switch 2 bands (EndDevice)
```
These are working correctly:
```
Nov 11 23:09:44 raspberrypi d5cf254a587d[662]: #033[32mZigbee2MQTT:info #033[39m 2020-11-11 23:09:44: kitchen_shutter (0x04cf8cdf3c77c733): WXCJKG12LM - Xiaomi Aqara Opple switch 2 bands (EndDevice)
```

golddragon007, 2020-11-12:
@Koenkk Even this is a z2m issue? but before I've changed down my controller from cc2531 I did not have this problem...

Koenkk, 2020-11-12:
@golddragon007 I think the aqara devices did not configure correctly by default they send messages to group 0 and probably the Niko switch is in group 0 by default. You can force a reconfigure via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to wakeup the device right before sending it).

golddragon007, 2020-11-12:
@Koenkk How can I check if the issue is solved (without shut off everything)?

golddragon007, 2020-11-12:
I did it, here is the log: 
[reconfigure.zip](https://github.com/Koenkk/zigbee2mqtt/files/5532671/reconfigure.zip)
For me, it seems that it always goes to groupid 0 I have multiple tries in that file. I've also noticed that now it has only singe click according to the MQTT messages that I've got from the broker, even if I do multiple presses. I guess because of the wrong configuration?

Koenkk, 2020-11-14:
@golddragon007 did you see successfully configured in the zigbee2mqtt log?

golddragon007, 2020-11-14:
@Koenkk ?? For me, it seems like the configuration message goes, but they don't react or reconfigure themself to the same group. Because nothing changes.

golddragon007, 2020-11-14:
The aqara switch didn't configure well, it always sends single presses even if I do double or trippel or longpress...

Koenkk, 2020-11-14:
When checking the zigbee2mqtt log when configuring, does it says: 'Successfully configured' after executing https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

golddragon007, 2020-11-14:
Haha, so a lot of times it's fails with this:
```
2020-11-14T20:35:10.682Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":34836,"relaycount":0,"relaylist":[]}
2020-11-14T20:35:10.682Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-14T20:35:10.725Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0x04cf8cdf3c77c750:40752,1)
2020-11-14T20:35:10.725Z zigbee-herdsman:controller:endpoint Write 0x04cf8cdf3c77c750/1 aqaraOpple({"mode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Timeout - 40752 - 1 - 235 - 64704 - 4 after 10000ms)
[31mZigbee2MQTT:error[39m 2020-11-14 21:35:10: Failed to configure 'bedroom1_shutter', attempt 2 (Error: Write 0x04cf8cdf3c77c750/1 aqaraOpple({"mode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Timeout - 40752 - 1 - 235 - 64704 - 4 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```

After it was successfully configured it works fine. But how can I prevent this? I mean the wrong or no configuration or at least to not shut off all my sockets incl. my server one too? :D

Koenkk, 2020-11-15:
- Configure only has to succeed once (which should happen right after pairing) and will not be executed after that anymore
- The attribute report is normal, nothing strange there
- `�[34mZigbee2MQTT:debug�[39m 2020-11-14 22:05:45: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"livingroom_shutter2"}'
` indicates Zigbee2MQTT received an MQTT message (which was probably send by you?)

golddragon007, 2020-11-15:
No, my messages are like:
```
[34mZigbee2MQTT:debug[39m 2020-11-12 19:50:53: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'bedroom1_shutter'
[32mZigbee2MQTT:info [39m 2020-11-12 19:50:53: Configuring 'bedroom1_shutter'
```

Koenkk, 2020-11-15:
@golddragon007 I think it can also occur when you click the "Configure"  button in the frontend.

niqu1982, 2020-11-22:
Aquara button have this problem. Solution: reset to default, set and pair again, after this wqake up by sigle shot press reset button then send from zigbee2mqtt configure this device. Problem solved. I have this same problem and fight 2 days now it's ok

fluppie, 2020-12-23:
@golddragon007 How did you pair the socket to your zigbee network? Manual only shows how to reset the device to factory.

golddragon007, 2020-12-23:
It says for me how to put in pairing mode, you need to turn on/off 3 times. Then it will flash (I think) slowly.

fluppie, 2020-12-23:
Thanks, in the meantime I've found it in a Niko manual http://guide.niko.eu/display/HGNHC2V2/Manual+use
I've created a pull request for the docs :-) https://github.com/Koenkk/zigbee2mqtt.io/pull/519

aalbul, 2021-01-07:
Just a short update.
I had the same issue, and it appeared to be that the switch was in "command" mode by default. Taking into account, that it is bound to the coordinator by default, it was using button 1 / button 2 to turn on/off all the lights/switches.

The simplest solution was to change the operation mode to "event" by sending the message:

```json
{"operation_mode": "event"}
```

to `zigbee2mqtt/Bedroom Second Floor Bed Switch Left/set` where "Bedroom Second Floor Bed Switch Left" is a device friendly name.

closed 2021-02-15 by github-actions[bot]