Koenkk/zigbee2mqtt #4958: SNZB-02 reporting frequency

Pigotka, 2020-11-12:
## What happened
I have several SNZB-02 devices connected to Z2M and working correctly for weeks. The problem is that those devices only report temperature and humidity for changes less than some threshold  ( for temperature it is 1 degree) once per hour. Since those sensors are used indoor temperature changes are mostly below that threshold and thus regulation of the heating is quite slow.
I first expected it is FW issue or Sonoff problem but then I discussed on discord and some ppl reported much higher frequency. Also last week I repaired the device into ZHA (with same coordinator) and frequency increased to normal as well. This makes me think it is Z2M configuration problem.
Note that other temperature sensors like aquara and tuya works as expected on same coordinator and Z2M.

## What did you expect to happen
Update frequency will be at least once per 5 min.

## How to reproduce it (minimal and precise)
Pair SNZB-02 and keep the sensor in constant temperature. Watch updates frequency.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: tested on both CC2531, CC26X2R1
Adapter firmware version: 20201026

Pigotka, 2020-11-12:
[image]
First part of data reported by ZHA and second by Z2M using same coordinator and same sensor.

RoyP99, 2020-11-14:
Hi Pigotka,
just happens to have the same issue.
Received a SNZB-02 sensor yesterday and it worked fine but reported only once every hour or for a temperature change more than 1 degrees.

First saw the issue [here](https://github.com/Koenkk/zigbee2mqtt/issues/3794)
But this is already too old as the file devices.js is already different now.
Then I saw you issue with the same question I had but with no answer yet.
So I looked how to solve it and I succeeded.

Go to the directory node_modules/zigbee-herdsman-converters/ where you find the file devices.js
Now search for the SNZB/02 entry where you see the following part±
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            const bindClusters = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'];
            await bind(endpoint, coordinatorEndpoint, bindClusters);
            await configureReporting.temperature(endpoint, {min: 5, change: 100});
            await configureReporting.humidity(endpoint);
            await configureReporting.batteryVoltage(endpoint);
        },
The line configureReporting.temperature describes when the temperature must be reported.
Default is± min=10, max=1hour, change=100, this is defined at:
    temperature: async (endpoint, overrides) => {
        const payload = configureReportingPayload('measuredValue', 10, repInterval.HOUR, 100, overrides);
        await endpoint.configureReporting('msTemperatureMeasurement', payload);
    },
So this means after 1 hour it will report and also when the change is more than 100 -> 100/100=1 degrees
The min value is changed to 5.
I wanted it to report every 0.2 degrees so I changed it "change: 20".
But for once every 5 minutes add the following: "max: repInterval.MINUTES_5"
            await configureReporting.temperature(endpoint, {min: 5, change: 100, max: repInterval.MINUTES_5});

Now I hope I'm correct as I'm not an experienced user and I haven't tried the above suggestion.

After this change you have to reconfigure the module again.
For this look at the link provided above as otherwise there will be no change in the behavior.

For me there is only one question. Is the configuration done in the module itself or is it something in zigbee2mqtt?
I guess in the module itself as you have to reconfigure it.

Good luck

Pigotka, 2020-11-16:
Thank you @RoyP99 so much. Your findings was correct. I had no idea that configuration can affect reporting frequency.
Last few days I'm experimenting with good values and right now I'm satisfied with:
`{min: 5, max: repInterval.MINUTES_15, change: 25}`
I will create pull request soon.  But I think we are playing with user preferences here. Maybe someone don't need so frequent updates. This would be cool to have as a device specific conf but I have no idea if that's possible. If someone would point me to an example I would code it myself. 

While playing with the configuration of the device I should mention that reconfiguring that sensor is a pain. I did not manage to reconfigure it using payload and had to force remove and repair the device every time. Sometimes it had to be factory reset multiple times as configure failed.

The question whether configuration is done in device or in z2m remains.

Pigotka, 2020-11-18:
Sensor works fine now but it is draining the battery more than it should. I will slightly correct the settings soon.

cadavre, 2020-11-19:
This kind of changes are useless. :) @Koenkk isn't there a way to let users configure this by themselves via `confi.yml` device configuration?

Ones would like to configure this to update more often, others less often...

Changing this back and forth looks pointless. :)

Koenkk, 2020-11-19:
@cadavre this is currently being discussed here https://github.com/Koenkk/zigbee2mqtt/issues/4321

sjorge, 2020-11-19:
> Sensor works fine now but it is draining the battery more than it should. I will slightly correct the settings soon.

I dropped about 2-5% in 3 days, a lot more than before... I did spend a lot of time repairing them so it might be related to that also.