Koenkk/zigbee2mqtt #4960: Backup restore does not work on "unclear" stick, even after reflashing it with JFlash

Hacker-CB, 2020-11-12:
## What happened

Backup is not restored on stick which was used before in other z2m instance, even it was reflashed with JFlash.
 
## What did you expect to happen

Backup should be restored.

## How to reproduce it (minimal and precise)

* Take new `unclear` stick which was used on other z2m instance before.
* Flash it with JFlash
* Stop z2m
* Extract old stick
* Insert new one
* Start z2m
* Backup will not restored because `hasConfigured` is still true.

## Why this happend

* JFlash does not erase NVMEM items.

## Suggestion

* Add `instance_id` value which will be uniq for each z2m instance
* Save `instance_id` to `configuration.yaml` or to `database.db` coordinator entry.
* Save `instance_id` to coordinator NVMEM inside `znpHasConfigured` or to the new nvitem.
* Erase NVMEM if `instance_id` is different

Hacker-CB, 2020-11-12:
Here is function to erase all nvram items: https://github.com/jethome-ru/zigbee-herdsman/commit/b21bf0bf73d28e31a5b98e8ed6fe02153dba7b97

dmak, 2020-12-13:
Would be nice to have a code that erases NVMEM as part of Z2M.

closed 2021-01-20 by github-actions[bot]