Koenkk/zigbee2mqtt #4963: Ikea TRADFRI bulb LED1624G9 does not support state OFF in combination with transition != 0

baderas, 2020-11-12:
## What happened
When I send the following to one of my LED1624G9 Tradfri Bulbs, when it is on:
`{"state":"OFF","transition":2,"color":{"hex":"#9ACD32"}}`
Zigbee2mqtt reports a new state shortly after:
`{"brightness":254,"color":{"x":0.1513,"y":0.3425},"last_seen":"2020-11-12T15:08:22+01:00","linkquality":117,"state":"OFF","update":{"state":"idle"},"update_available":false}`

**But:** The bulb still is on. It is not off.
Sometimes it works, if I spam the "OFF" state 8 or more times in a row.

However after some debugging, the transition seems to be the problem. If I send:
`{"state":"OFF","transition":0,"color":{"hex":"#9ACD32"}}`
The bulb goes immediately off (reliably).

## What did you expect to happen
The bulb should slowly fade off in 2 seconds.

## How to reproduce it (minimal and precise)
Send
`{"state":"ON","transition":2,"brightness":255}`
to a LED1624G9 (FW: 1.3.013).
Wait a few minutes (>5 minutes), then send:
`{"state":"OFF","transition":2,"color":{"hex":"#9ACD32"}}`
It stays lit. Then send:
`{"state":"OFF","transition":0,"color":{"hex":"#9ACD32"}}`
--> it goes off.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: slaesh's stick (Z-Stack 3.x.0 firmware (20201026))
Adapter firmware version: LED1624G9 has   20200220  1.3.013

Not sure if it this happens with other Ikea Tradfri bulbs.
Maybe its a problem in the FW of the bulb (latest, no update available) and not zigbee2mqtt related. 
Maybe transition is generally not supported when sending state=OFF? (if that is the case: improve documentation?)

closed 2020-12-21 by github-actions[bot]

schmittner, 2021-01-04:
For the record, I have a similar problem with the same model (same FW). When turning on the bulb with `transition` > 0, and turning off the bulb shortly after, the bulb will stay ON but is reported as OFF. When activating `retrieve_state`, the bulb reports itself as ON. When setting transition to 0, the problem is gone.

So it seems that the turn off command is somehow ignored. Maybe this has something to do with the fact that brightness and color commands cannot be set together via Zigbee? See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/445#issuecomment-373649142 or https://github.com/basnijholt/adaptive-lighting/issues/49.

Any ideas?

Koenkk, 2021-01-05:
@schmittner can you provide the herdsman debug log of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

schmittner, 2021-01-06:
Sure. It does not happen every time, but I managed reproduce this state. What is happening from HA side (MQTT messages):
1. Manually turn lamp on
2. Automation listens on turn on events and sets the lamp brightness and color based on time of day ("adaptive lighting")
3. Manually turn lamp off

Note that I replaced the actual device ID with `0xaabbccddeeff0011`.

```
Zigbee2MQTT:debug 2021-01-06 07:51:23: Received MQTT message on 'zigbee2mqtt/ikea_lamp/set' with data '{"state": "ON"}'
Zigbee2MQTT:debug 2021-01-06 07:51:23: Publishing 'set' 'state' to 'ikea_lamp'
  zigbee-herdsman:controller:endpoint Command 0xaabbccddeeff0011/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +1s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xaabbccddeeff0011:46810/1 (0,0,1) +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":46810,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":202,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,134,1]}} +1s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,218,182,1,1,6,0,202,0,30,3,1,134,1,19] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,202,12] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,202,12] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,202] - 12 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":202} +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,218,182,0,238] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,218,182,0,238] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [218,182,0] - 238 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":46810,"relaycount":0,"relaylist":[]} +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,218,182,0,238] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,218,182,0,238] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [218,182,0] - 238 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":46810,"relaycount":0,"relaylist":[]} +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,6,0,218,182,1,1,0,141,0,236,183,165,0,0,5,8,134,11,1,0,218,182,29,53] +45ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,6,0,218,182,1,1,0,141,0,236,183,165,0,0,5,8,134,11,1,0,218,182,29,53] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,218,182,1,1,0,141,0,236,183,165,0,0,5,8,134,11,1,0,218,182,29] - 53 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":46810,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":141,"securityuse":0,"timestamp":10860524,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,134,11,1,0]}} +47ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":134,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":1,"statusCode":0}},"address":46810,"endpoint":1,"linkquality":141,"groupID":0}' +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
Zigbee2MQTT:info  2021-01-06 07:51:23: MQTT publish: topic 'zigbee2mqtt/ikea_lamp', payload '{"brightness":254,"color":{"x":0.649,"y":0.343},"linkquality":141,"state":"ON","update":{"state":"idle"},"update_available":false}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,218,182,1,1,0,141,0,240,184,165,0,0,7,8,15,10,0,0,16,1,218,182,29,190] +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,218,182,1,1,0,141,0,240,184,165,0,0,7,8,15,10,0,0,16,1,218,182,29,190] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,218,182,1,1,0,141,0,240,184,165,0,0,7,8,15,10,0,0,16,1,218,182,29] - 190 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":46810,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":141,"securityuse":0,"timestamp":10860784,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,15,10,0,0,16,1]}} +28ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":15,"manufacturerCode":null,"commandIden [... cut ...] :{"type":"Buffer","data":[8,137,11,0,0]}} +49ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":137,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":0,"statusCode":0}},"address":46810,"endpoint":1,"linkquality":144,"groupID":0}' +451ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
```

Koenkk, 2021-01-06:
If I understand correctly, as a result of `Zigbee2MQTT:debug 2021-01-06 07:51:23: Received MQTT message on 'zigbee2mqtt/ikea_lamp/set' with data '{"state": "OFF"}'` the bulb is NOT turned off? 

If that is the case this looks like a bug in the firmware. Seems it ignores this command when still in a transition.

schmittner, 2021-01-06:
> If I understand correctly, as a result of `Zigbee2MQTT:debug 2021-01-06 07:51:23: Received MQTT message on 'zigbee2mqtt/ikea_lamp/set' with data '{"state": "OFF"}'` the bulb is NOT turned off?

Correct.

> If that is the case this looks like a bug in the firmware. Seems it ignores this command when still in a transition.

That sounds like a reasonable explanation to me. Maybe this would be worth a note in the docs? In my personal setup, I'll also activate `retrieve_state` for those bulbs.

Koenkk, 2021-01-06:
Done

pro2call-nl, 2021-05-17:
Today i have bought an bulb. updated it to firmware version: 1.3.013 and it no longer ocours

baderas, 2021-05-18:
Strange, that is the same version that I used 12 Nov 2020 when I reported the problem.
Are you sure its gone and that you have 1.3.013 and not a newer one?