Koenkk/zigbee2mqtt #4968: Touchlink factory reset not working with Hue Play

attley86, 2020-11-13:
I'm not able to make the touchlink factory reset. Hue light is next to coordinator and all I get is this when I publish the "zigbee2mqtt/bridge/config/touchlink/factory_reset" with empty payload.

Its either the "BridgeLegacy" error or just "reset_failed" as an outcome:

My stick is using:
`'{"version":"1.13.1","commit":"2cb234c","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"debug","permit_join":true}'`

Log:
```
zigbee2mqtt:debug 2020-11-13 07:10:06: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset' with data ''
zigbee2mqtt:info  2020-11-13 07:10:06: Starting touchlink factory reset...
zigbee2mqtt:info  2020-11-13 07:10:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
zigbee2mqtt:error 2020-11-13 07:10:28: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SREQ '--> AF - interPanCtl - {"cmd":1,"data":[24]}' failed with status '186' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:249:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58))
zigbee2mqtt:debug 2020-11-13 07:12:25: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset' with data ''
zigbee2mqtt:info  2020-11-13 07:12:25: Starting touchlink factory reset...
zigbee2mqtt:info  2020-11-13 07:12:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
zigbee2mqtt:warn  2020-11-13 07:12:51: Failed to factory reset device through Touchlink
zigbee2mqtt:info  2020-11-13 07:12:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_failed","meta":{"status":"failed"}}'
```

joekerna, 2020-11-13:
I have the same issue with my Philips 7099930PH

```
info  2020-11-13 14:22:26: Starting touchlink factory reset...
info  2020-11-13 14:22:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"reset_started","meta":{"status":"started"},"type":"touchlink"}'
warn  2020-11-13 14:22:37: Failed to factory reset device through Touchlink
info  2020-11-13 14:22:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"reset_failed","meta":{"status":"failed"},"type":"touchlink"}'
debug 2020-11-13 14:26:01: Saving state to file /opt/zigbee2mqtt/data/state.json
```

sjorge, 2020-11-26:
Same issue with an older hue R14 candle... 

```
Zigbee2MQTT:info  2020-11-26 20:00:54: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/identify', payload '{"data":{"channel":11,"ieee_address":"0x0017880104259333"},"status":"ok"}'
Zigbee2MQTT:info  2020-11-26 20:00:59: Start Touchlink factory reset of '0x0017880104259333' on channel 11
Zigbee2MQTT:info  2020-11-26 20:01:00: MQTT publish: topic 'zigbee2mqtt/bedroom/lights/bulb3', payload '{"brightness":254,"color_temp":370,"linkquality":60,"state":"OFF"}'
(node:25660) UnhandledPromiseRejectionWarning: Error: Cannot execute command, in Inter-PAN mode
    at ZStackAdapter.checkInterpanLock (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:875:19)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:166:22)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:23:12)
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:165:44)
    at Queue.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:29:46)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:8:71
(Use `node --trace-warnings ...` to show where the warning was created)
(node:25660) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 30)
(node:25660) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Zigbee2MQTT:info  2020-11-26 20:01:00: MQTT publish: topic 'zigbee2mqtt/bedroom/lights/bulb2', payload '{"brightness":254,"color_temp":370,"linkquality":65,"state":"OFF"}'
Zigbee2MQTT:info  2020-11-26 20:01:00: MQTT publish: topic 'zigbee2mqtt/bedroom/lights/bulb1', payload '{"brightness":254,"color_temp":370,"linkquality":63,"state":"OFF"}'
Zigbee2MQTT:info  2020-11-26 20:01:00: MQTT publish: topic 'zigbee2mqtt/bedroom/lights/bulb4', payload '{"brightness":254,"color_temp":370,"linkquality":63,"state":"OFF"}'
```

Identify works fine, but reset does not.

MattWestb, 2020-11-27:
@sjorge Is TL reset working on other lights like your IKEA ones ?
If yes you need repower the HUE ones before they is opened for TL reset (with newer firmware).
If not working with other lights its some sone bug in the Z2M / NCP firmware side.

sjorge, 2020-11-27:
After unplug/replug of the coordinator it showed the hue bulb I wanted to reset and a nearby ikea bulb too. So i think it's a firmware/z2m issue with the coordinator.

MattWestb, 2020-11-27:
Good if the coordinator getting firmware lock you cant see any lights.

Little careful with IKEA lights, i was by mistake resetting one GU10 WS2.0 in my kitchen with deCONZ more then 6 meters away then not looking so carefully on the IEEE and must touchlinking it back to IKEA GW.

sjorge, 2020-11-27:
Yeah! The Hue needs to be super close but trådfri seems to show up ~2.5 meters away.

LEDs were normal on the coordinator when it was failing.

~ sjorge

closed 2021-01-04 by github-actions[bot]