Koenkk/zigbee2mqtt #4969: Group state is not updating in Home Assistant

pedrolamas, 2020-11-13:
## What happened
I have a couple of Philips lamps in a group (KidsCornerGroup) and an Ikea Tradfri E1743 button (KidsCornerSwitch) binded to it.

[image]

[image]

When I turn on the lights with the KidsCornerSwitch, Home Assistant does not show the group as on.

## What did you expect to happen

Home Assistant should show the group is on if I use the KidsCornerSwitch to turn it on.

## How to reproduce it (minimal and precise)

Home assistant relevant config:

```yaml
lights:
- platform: mqtt
  schema: json
  name: Kids Corner
  command_topic: "zigbee2mqtt/KidsCornerGroup/set"
  state_topic: "zigbee2mqtt/KidsCornerGroup"
  brightness: true
```

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: 20201026

pedrolamas, 2020-11-13:
Given [this](https://www.zigbee2mqtt.io/information/report.html#support), am I correct to say that the problem here is with my Philips lamps that don't support reporting back the change to z2m?

pedrolamas, 2020-11-13:
I see now this is due to Philips lamps...meh!

Closing the issue now.

pedrolamas, 2020-11-29:
Reopening this issue, as I've been thinking further on this matter...

From what I can see, the Ikea Tradfri E1743 does send a message when I press on it for on/off, so can't we just poll the lamps on the group for their state?

My understanding by looking at the [tests](https://github.com/Koenkk/zigbee2mqtt/blob/c0143cb40d9f139e03e32b23393f5846e86a941c/test/report.test.js#L253) is that this should be the case, but it doesn't seem to be working in my case.

pedrolamas, 2020-11-29:
@sjorge you seem to have written the original code for polling the Philips Hue lamps as they don't report changes, so I was wondering if you have any thoughts on this?

sjorge, 2020-11-29:
Depending on the remote, it should already be polled:
https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/report.js#L242 using the mapping from earlier https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/report.js#L63

Is it one of the remotes that is sold seperate? Or was it one that came bundled with a blind?
If the later, maybe we're missing some commands in the mapping.

MattWestb, 2020-11-29:
@pedrolamas The E1743 is sending on/off commands to its bounded groupe and if Z2M is listening on the groupe it can catching the status that the last group command and assuming the state the group suld having.
But the E1524/E1810 is not sending on/off its sending toggle to the group then its not so easy to knowing the groupe state without logic in Z2M. The normal in the light part of ZLL and ZB3 is the the state is reported from the lights not the controlling devices (Philips and some more hís doing the opposite). And Sjorge have working around the "HUE problem" :-)))

pedrolamas, 2020-11-29:
@MattWestb @sjorge thank you both for your comments.

I'm using an E1743 and I can see it sending the correct on/off commands (as shown on the log on the first post) so I would expect it to poll the lamps afterwards or just consider them on/off but that doesn't seem to be the case right now...

MattWestb, 2020-11-29:
And one more problem is if one or more lights is in more groups (that is possible) or unicast from the coordinator then the state of the command from the controlling device is wrong. The state of the group should being all lights is of = group off, and not all lights is off = on. 
And the state of one light is from on/off cluster 0x0006 = off and not wath light level its having because its being reported at it was having before being turned off and is being then its turning on agen ala LL and ZB3 light standard.

pedrolamas, 2020-11-29:
Ok, after some debugging I got some more info!

[This condition](https://github.com/Koenkk/zigbee2mqtt/blob/cb9699e12abffcfa97a7a4add40275d5a00827f8/lib/extension/report.js#L267) is false for my devices as `poll.manufacturerID` is 4107 while the `endpoint.getDevice().manufacturerID` is 4116.

I'm now working out the rest...

sjorge, 2020-11-29:
Did Philips switch manufacturerID... if that is the case the pollOnMessage probably needs updating

pedrolamas, 2020-11-29:
My thoughts exactly! Trying to determine that now, but one thing is for sure: those are Philips Hue lamps as I bought them myself quite a few years ago in a retailer (Currys Pc World) here in the UK! 😉

pedrolamas, 2020-11-29:
According to [this list](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin_private.h), 4116 (0x1014) is the manufacturer code for Atmel, and apparently they produced quite a few devices for Philips (example [here](https://blog.adafruit.com/2016/06/14/teardown-of-a-philips-hue-led-lightbulb-with-zigbee-and-atmega2564-avr-iot-iotuesday/))

There are a few database.db entries in other issues showing this (example, just search for "4116" on [this thread](https://github.com/Koenkk/zigbee2mqtt/issues/3647#issuecomment-643573369) and check the manufacturer name to confirm it is "Philips")

I guess it is safe to assume that we need to have the polling for 0x1014 (Atmel built) lamps!

Koenkk, 2020-11-30:
Fixed by #5139

PR #5139 (2020-11-30): Poll lamps with Philips, Atmel, Gledopto, Mueller-Licht manufacturer codes
This fixes #4969 by checking for both Philips and Atmel manufacturer codes (as we came to realize that Philips Hue lamps have components from both)
closed 2020-11-30 by Koenkk