Koenkk/zigbee2mqtt #4973: RTCGQ11LM reports occupancy constantly with new coordinator

raqua, 2020-11-13:
## What happened
I am running zigbee2mqtt in Docker for some time. I was using CC2531 until today. I have bought zzh! device and replaced my coordinator today, re-pairing all the devices, but keeping the same configuration.

## What did you expect to happen
All seems to work fine, except for my RTCGQ11LM motion sensor. I have "occupancy_timeout: 180" in my settings and since switching to new coordinator, it reports motion all the time there is some, so every second or so. I am concerned about battery life when it reports so often and my logs are flooded with those messages. 

## How to reproduce it (minimal and precise)
Have RTCGQ11LM paired to CC2531 with occupancy_timeout: 180 (probably not even that is needed as default should work too)
Check logs that it reports after specified timeout.
Switch to CC26X2R1 and check logs to see that motion is reported constantly.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: 20201026

raqua, 2020-11-13:
Configuration: https://pastebin.com/ft89h2Mp
Debug logs: https://pastebin.com/7Q3B5mEb
Debug logs stripped of color: https://pastebin.com/j3S6xGdX

raqua, 2020-11-13:
Occupancy: false is correctly reported after 180 seconds of no movement.

Koenkk, 2020-11-14:
AFAIK this only happens the first 5 minutes after pairing the device.

raqua, 2020-11-14:
Yes, it has resolved itself, although I think it took a few hours.. 
Thank you.

closed 2020-11-14 by raqua