Koenkk/zigbee2mqtt #4976: Problem with Maps (fronted) and Graphviz

kwetnico, 2020-11-14:
Information about the device  :+1: 
Raspberry PI4
29 devices Xiaomi and Heiman
Zigbee2mqtt master 1.16.1

Hello,

Thanks @Koenkk  for yesterday. #4971 

I redid my network to 0 because I just went from a CC2531 coordinator to Conbee2.
I was able to reintegrate all my devices
But when I want to generate a map with Fronted or Graphiz it does not put any link between my coordinator and my devices. (it works when you pair them and after that it disappears).
Is this normal? What to do ??

[image]
[image]

Another question since I switched to the Conbee2 I have the LQI value ton fronted, it is the signal if I say no stupidity, which is between 150 and 220 (before with the CC2531 I had between 30 and 90) c what is the minimum and maximum interval for this value?

Thanks

kwetnico, 2020-11-14:
Same problem with dev version to zigbee2mqtt
[image]

Koenkk, 2020-11-14:
This is normal for Xiaomi sensors, they sleep for too long and depending on what router used the router removes them from the child list. **This does not mean the sensor won't work or is disconnected from the network**

kwetnico, 2020-11-14:
Hello, thank you for your answer.  They are connected directly to my coordinator which is the CONBEE2 key. But I just added them and none have a visible link. They will create the link on their own in several days?

Koenkk, 2020-11-14:
No, but as I mentioned this is normal for Xiaomi battery powered devices.

kwetnico, 2020-11-14:
Yes I had seen but why so far I have only had this symptoms with a cc2531 coordinator.  Here is a graphiz photo from last week.  What can I do to find this visual
[image]

kwetnico, 2020-11-15:
> No, but as I mentioned this is normal for Xiaomi battery powered devices.

For me there is a problem with the feedback from the LQI which is implemented +1 (identical value for all devices)
and for the cartography which no longer works by indicating the links.
I repeat I don't have this with CC2531 key

kwetnico, 2020-11-18:
@Koenkk 
Several of us have the same problem with the CONBEE2 key that we don't have with the CC2531.
No map with the conbee2 coordinator and the information from the LQI is not good

Here : https://github.com/Koenkk/zigbee-herdsman/issues/72

Koenkk, 2020-11-19:
Can you provide the herdsman debug logging when running the network scan so @chrishae can take a look?

kwetnico, 2020-11-19:
> Can you provide the herdsman debug logging when running the network scan so @ChrisHae can take a look?

Here is a debug mode:
I launched then manually updated a device, its name: "Temperatue_Bureau", you will be able to see the LQIs which return figures which are not realistic.
And I launched a creation of the map under Fronted at the end.
Thanks

`pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.16.1 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2020-11-19 09:52:13: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2020-11-19 09:52:13: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-19.09-52-12' filename: log.txt
Zigbee2MQTT:debug 2020-11-19 09:52:13: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-11-16.18-03-06'
Zigbee2MQTT:info  2020-11-19 09:52:13: Starting Zigbee2MQTT version 1.16.1 (commit #6b32f30)
Zigbee2MQTT:info  2020-11-19 09:52:13: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-11-19 09:52:13: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'DeconzAdapter' +0ms
CREATED DECONZ ADAPTER
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0 +0ms
  zigbee-herdsman:deconz:driver Serialport opened +12ms
  zigbee-herdsman:controller:log Started with result 'resumed' +21ms
  zigbee-herdsman:controller:log Injected database: true, adapter: true +1ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1 +78ms
  zigbee-herdsman:deconz:frameParser MAC: 0x00212effff0546e4 +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 2 paramId: 7 +101ms
  zigbee-herdsman:deconz:frameParser NWKADDR: 0 +92ms
Zigbee2MQTT:info  2020-11-19 09:52:14: zigbee-herdsman started
  zigbee-herdsman:deconz:driver send read firmware version request from queue. seqNr: 3 +100ms
  zigbee-herdsman:deconz:frameParser read firmware version response - version: 0,7,102,38 +101ms
Zigbee2MQTT:info  2020-11-19 09:52:14: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"}'
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 4 paramId: 5 +101ms
  zigbee-herdsman:deconz:frameParser PANID: 1a62 +100ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 5 paramId: 11 +100ms
  zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd +101ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 6 paramId: 28 +100ms
  zigbee-herdsman:deconz:frameParser CHANNEL: 11 +100ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 7 paramId: 24 +100ms
  zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden +100ms
Zigbee2MQTT:debug 2020-11-19 09:52:15: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2020-11-19 09:52:15: Currently 29 devices are joined:
Zigbee2MQTT:info  2020-11-19 09:52:15: Telecommande (0x00158d00023e61c2): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Porte_Entree (0x00158d0004147308): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Presence_Salon (0x00158d00042cfc5f): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Fenetre_Salon (0x00158d000414728a): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Porte_Fenetre (0x00158d0003cb3435): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Temperature_Salon (0x00158d000411ea5c): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Fenetre_Cuisine (0x00158d00041472e4): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Temperature_Cuisine (0x00158d0004119451): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Capteur_Vibration_Garage (0x00158d0003d33b15): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: Presence_Garage (0x00158d0003554dfc): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2020-11-19 09:52:15: DetecteurFumee (0x00158d0004c07fa8): HS1SA - HEIMAN Smoke detector (EndDevice)
[... cut ...]
  zigbee-herdsman:deconz:driver query aps data confirm +190ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 83 +36ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x4d01 request id: 19 confirm status: 0 +229ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 61 +397ms
`

kwetnico, 2020-11-20:
> Can you provide the herdsman debug logging when running the network scan so @ChrisHae can take a look?

@Koenkk 
Hello I put the logs but no news from @ChrisHae  and even on his subject forum on the CONBEE2 key.

kwetnico, 2020-11-22:
Hello @Koenkk  and @ChrisHae I have just tested with the new firmware released on November 18 (deCONZ_ConBeeII_0x26670700.bin.GCF) and still the same problems with the ConbeeII key

Koenkk, 2020-11-22:
@kwetnico I understand your issue, but I don't think this can be solved as explained in https://github.com/Koenkk/zigbee2mqtt/issues/4976#issuecomment-727225308

kwetnico, 2020-11-22:
> @kwetnico I understand your issue, but I don't think this can be solved as explained in [#4976 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/4976#issuecomment-727225308)

I also understand your explanation but why I do not have this with the cc2531 key ...?

Koenkk, 2020-11-22:
The CC2531 doesn't cleanup the child table, it seems that the Conbee II does after devices didn't check-in for a long time (which is correct behaviour if devices sleep for too long).

kwetnico, 2020-11-23:
> The CC2531 doesn't cleanup the child table, it seems that the Conbee II does after devices didn't check-in for a long time (which is correct behaviour if devices sleep for too long).

Last question this explains the LQI figures which are really strange with the ConbeeII?
The LQI is the same for all network devices connected to the ConbeeII coordinator.
The number is incremented by +1 each time and loops from 1 to 255.

Koenkk, 2020-11-23:
That is a known issue: https://github.com/Koenkk/zigbee2mqtt/issues/4039

kwetnico, 2020-11-23:
> That is a known issue: #4039

Thanks

kwetnico, 2020-11-23:
> > That is a known issue: #4039
> > @Koenkk
> > What's good command for to have the good LQI ?
> > This command ?
> > mosquitto_pub -t zigbee2mqtt/bridge/networkmap -m graphviz
> 
> I don't see the LQI value..

closed 2020-11-23 by kwetnico