Koenkk/zigbee2mqtt #4979: Tuya TS0601 wall switches stopped working after herdsman update

barskykd, 2020-11-14:
## What happened

I had these two switches:

https://pastebin.com/m1Firb4F

Then I updated zigbee-herdsman-converters to latest master. 
And switches stopped working. "No converter available" errors in logs: 
https://pastebin.com/6gTAy8iC

I tried this simple fix https://pastebin.com/5zcytmDs
and there no more errors in logs https://pastebin.com/r63aNeNa
but still nothing is sent to mqtt.

## What did you expect to happen

Expected what switches would be still working.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.14.2
Adapter hardware: CC2531
Adapter firmware version:
```
{"version":"1.14.2","commit":"faaf3e4",
"coordinator":{"type":"zStack12","meta":{
    "transportrev":2,
    "product":0,
    "majorrel":2,
    "minorrel":6,
    "maintrel":3,
   "revision":20190608}},
"log_level":"debug","permit_join":true}
```

barskykd, 2020-11-14:
So this changes will make ***my*** switches working but will probably break it for someone else
https://pastebin.com/PEywgC1S

Koenkk, 2020-11-14:
Can you make a PR for this? We will look further there.

drzony, 2020-11-15:
You need the latest zigbee-herdsman in order for zigbee-herdsman-converters to work. Try getting the latest develop zigbee2mqtt

drzony, 2020-11-15:
@Koenkk This is caused by desync of zigbee-herdsman and zigbee-herdsman-converters after my rework

Koenkk, 2020-11-15:
@drzony thanks! 

@barskykd you indeed need to update zigbee2mqtt, zigbee-herdsman and zigbee-herdsman-converters. The best way to do is by switching to the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). You cannot mix-up different version of zigbee-herdsman/zigbee-herdsman-converters which each other as they are potentially not compatible with each other.

closed 2020-11-15 by Koenkk

barskykd, 2020-11-16:
Yes. That fixed it. Thanks