Koenkk/zigbee2mqtt #4980: Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt'

darroyod, 2020-11-14:
## What happened
Since the update from 1.15.0 to 1.16.1, the devices are not able to be discovered by Home Assistant. It seems that zigbee2mqtt is not publishing the payloads as Home Assitant expects.
The error that home assistant shows is this:

`Nov 14 13:35:09 mayordomo hass[2261]: 2020-11-14 13:35:09 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Salon.persianaSalon/availability'}], 'command_topic': 'zigbee2mqtt/Salon.persianaSalon/set', 'device': {'identifiers': ['zigbee2mqtt_0x00047400008cb2bc'], 'manufacturer': 'Legrand', 'model': 'Netatmo wired shutter switch (067776)', 'name': 'Salon.persianaSalon', 'sw_version': 'Zigbee2MQTT 1.16.1'}, 'json_attributes_topic': 'zigbee2mqtt/Salon.persianaSalon', 'name': 'Salon.persianaSalon_cover', 'position_topic': 'zigbee2mqtt/Salon.persianaSalon', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/Salon.persianaSalon/set', 'unique_id': '0x00047400008cb2bc_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},)
Nov 14 13:35:09 mayordomo hass[2261]: Traceback (most recent call last):
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/homeassistant/components/mqtt/cover.py", line 183, in async_discover
Nov 14 13:35:09 mayordomo hass[2261]:     config = PLATFORM_SCHEMA(discovery_payload)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/validators.py", line 208, in __call__
Nov 14 13:35:09 mayordomo hass[2261]:     return self._exec((Schema(val) for val in self.validators), v)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/validators.py", line 287, in _exec
Nov 14 13:35:09 mayordomo hass[2261]:     raise e if self.msg is None else AllInvalid(self.msg, path=path)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/validators.py", line 283, in _exec
Nov 14 13:35:09 mayordomo hass[2261]:     v = func(v)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__
Nov 14 13:35:09 mayordomo hass[2261]:     return self._compiled([], data)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
Nov 14 13:35:09 mayordomo hass[2261]:     return schema(data)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__
Nov 14 13:35:09 mayordomo hass[2261]:     return self._compiled([], data)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
Nov 14 13:35:09 mayordomo hass[2261]:     return base_validate(path, iteritems(data), out)
Nov 14 13:35:09 mayordomo hass[2261]:   File "/opt/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
Nov 14 13:35:09 mayordomo hass[2261]:     raise er.MultipleInvalid(errors)
Nov 14 13:35:09 mayordomo hass[2261]: voluptuous.error.MultipleInvalid: extra keys not allowed @ data['availability']
`

No error is logged from the zigbee2mqtt side.
After the downgrade from 1.16.1 to 1.15.0 and with no configuration changes, everything starts working again.

## What did you expect to happen

## How to reproduce it (minimal and precise)
Upgrade from 1.15.0 to 1.16.1 (latest as of today) and start zigbee2mqtt service.

## Debug info
Zigbee2MQTT version:
Adapter hardware: Electrolama zig-a-zig-ah! Rev A
Adapter firmware version: Z-Stack_3.x.0

Koenkk, 2020-11-14:
You are probably running a not suported HA version, minimum required version is 0.112 (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.16.0)

darroyod, 2020-11-20:
You were right: with latest HASS version (0.118.1 on 20/nov/2020) the problem is solved.

closed 2020-11-20 by darroyod

maxfarrior, 2021-01-13:
I had this same issue, I was running HA 0.110.0. 

I upgraded to 0.112.0 and that did _not_ solve the problem. I didn't want to step through each version, so I just upgraded to the latest HA (2021.1.1) and all my sensors started working again.

Just throwing that out there.