Koenkk/zigbee2mqtt #4982: TuYa TS0044 not support in 1.16 zigbee2mqtt

inhibitz, 2020-11-14:
can not pair the zigbee device. although it is officially on the support list. there are not even logs. what could be the problem?

martinUB, 2020-11-19:
Hey there,
same issue in my case. Can't get device to pair. I don't get any message from the switch in the home assistant zigbee2mqtt log.
If some further information is needed, just ask for it :) 
Looking forward for a solution.

martinUB, 2020-11-24:
Hey inhibitz.
In my case it was a device independent issue. Try unplugging your coordinator, then plug it in again and after that restart the zigbee2mqtt or in my case complete home assistant. I was able to pair the device then. Check if join(all) is allowed.
Cheers
Martin

closed 2021-01-01 by github-actions[bot]

tnijkes, 2021-01-09:
Same here.
Anyone got it working yet?