Koenkk/zigbee2mqtt #4983: Philips Hue Fair with Bluetooth and new model name

eberlemartin, 2020-11-14:
Hello,
I just got my new Philips Hue Fair ceiling lamp and it does not pair with Z2M. I found out that the entry in devices.js has "3043041P7" as modelname, my lamp somehow identifies itself as "3043041P6". I tried to make a PR with an updated PR but i did not know what to put as "zigbeeModel"...so im opening a request :)

### Log:
`Zigbee2MQTT:info  2020-11-14 21:28:09: Device '[**Removed**]' joined
Zigbee2MQTT:info  2020-11-14 21:28:09: Starting interview of '[**Removed**]'
Zigbee2MQTT:info  2020-11-14 21:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"[**Removed**]","ieee_address":"[**Removed**]"},"type":"device_joined"}'
Zigbee2MQTT:info  2020-11-14 21:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"[**Removed**]","ieee_address":"[**Removed**]","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2020-11-14 21:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"[**Removed**]"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-11-14 21:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"[**Removed**]"},"type":"pairing"}'
Zigbee2MQTT:debug 2020-11-14 21:28:09: Device '[**Removed**]' announced itself
Zigbee2MQTT:info  2020-11-14 21:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"[**Removed**]","ieee_address":"[**Removed**]"},"type":"device_announce"}'
Zigbee2MQTT:info  2020-11-14 21:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"[**Removed**]"},"type":"device_announced"}'
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"modelId":"4034031P6"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Philips"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200124"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-11-14 21:28:11: Received Zigbee message from '[**Removed**]', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1.65.9_hB3217DF4"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-11-14 21:28:11: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info  2020-11-14 21:28:12: Successfully interviewed '[**Removed**]', device has successfully been paired
Zigbee2MQTT:warn  2020-11-14 21:28:12: Device '[**Removed**]' with Zigbee model '4034031P6' and manufacturer name 'Philips' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2020-11-14 21:28:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"[**Removed**]","ieee_address":"[**Removed**]","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:info  2020-11-14 21:28:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"[**Removed**]","supported":false},"type":"pairing"}'`

## Information about the device + link
Philips Hue Fair with Bluetooth
https://www.philips-hue.com/de-at/p/hue-white-ambiance-deckenleuchte-fair/4034031P6

### Not supported Message
WARNING: Received message from unsupported device with Zigbee model '4034031P6' and manufacturer name 'Philips'

I wanted to add following to devices.js but did not know which zigbeeModel name to use:

`    {
        zigbeeModel: ['LTC002'],
        model: '4034031P6'
        vendor: 'Philips',
        description: "Hue Fair with Bluetooth',
        meta: {turnsOffAtBrightness1: true},
        extend: hue.light_onoff_brightness_colortemp,
        ota: ota.zigbeeOTA,
    },`

Can someone help me fix the entry or just correct it and put it in the devices.js?

Cheers

me

sjorge, 2020-11-14:
The zigbeeModel array should also be 4034031P6 I think.

eberlemartin, 2020-11-16:
Thanks for the quick answer. I created a PR in the herdsman repository and its already merged. 👍

closed 2020-11-16 by eberlemartin