Koenkk/zigbee2mqtt #4985: CC2531 Disconnects after some time

igiul667, 2020-11-14:
## What happened
CC2531 debbuger gets recognized as a serial device but then disappears after some time passes. It does that if it is just connected or if I try to run zigbee2mqtt
## What did you expect to happen
The CC2531 should stay connected as a serial device
## How to reproduce it (minimal and precise)
Don't know, it randomly stopped the zigbee2mqtt and it wouldn't start again

KiLLeRRaT, 2020-11-16:
Hi, is this just a recent issue, or a new install?

Sounds like something that's happening to me after a potential Windows update... I'm currently troubleshooting, and will add more info here if I find anything...

igiul667, 2020-11-16:
It happened randomly, no updates. Running on Ubuntu on intel nuc. I reflashed the adapter and solved the proble. Could my cc2531 be broken in a way that makes it erase it's flash contents? or could it be the firmware to have issues?

closed 2020-11-16 by igiul667