Koenkk/zigbee2mqtt #4989: Ability to query group state

stefanheule, 2020-11-15:
It would be nice to be able to query the state of a group. My home automation system uses this at startup (of the home automation system, not zigbee2mqtt) to learn the current state of the system. It seems like there is no way to get the state of groups any more, they only output their state when the state changes.

## Describe the solution you'd like
Explicitly querying the state of a group at `"zigbee2mqtt/<group_name>/get"` should send out the current state of that group.

## Describe alternatives you've considered
I could have my external system know the group to device mapping and learn the state by querying all the member devices. That's doable, but it seems like just re-implementing state tracking outside of zigbee2mqtt.

## Additional context
It used to be possible to query the group state by querying any of the members. This was recommended as a solution by @Koenkk in https://github.com/Koenkk/zigbee2mqtt/issues/764#issuecomment-658572559. However, I believe with the change for https://github.com/Koenkk/zigbee2mqtt/issues/3461 that stopped working.

stefanheule, 2020-12-16:
If a developer familiar with the code sends me a few pointers, I could try implementing this myself and sending a PR. What are some files/functions I likely need to touch?

Koenkk, 2020-12-17:
> My home automation system uses this at startup (of the home automation system, not zigbee2mqtt) to learn the current state of the system.

For this use case you can use the `retain` option. This is a native MQTT feature which will send the last message published to a topic upon subscribing to it. Read more about this option here: https://www.zigbee2mqtt.io/information/configuration.html#all-devices 

Example for `configuration.yaml`:

```yaml
groups:
  '1':
    friendly_name: group_1
    retain: true
```

stefanheule, 2020-12-17:
Thanks! But that requires resubscribing, which may not always be desirable. Is there a reason there is no option to query the state of a group just like you can query individual elements? If I were to send a PR that adds this functionality, would you accept that?

It seems like a very natural feature, but maybe I'm missing something.

Koenkk, 2020-12-18:
> Is there a reason there is no option to query the state of a group just like you can query individual elements? 

Because groups actually have no state (and the state of devices in it can vary in theory). 

I'm wondering if it makes more sense to add a command to republish all known states? E.g by publishing to `zigbee2mqtt/bridge/request/publish_states`?

stefanheule, 2020-12-18:
Okay, I guess that makes sense. Basically groups can be set, and it's equivalent semantically to setting each member of the group to the same state (though in terms of timing the group set seems to do all of them at once, while the iterative "do each member" is noticeably sequential).

A command to republish all states would actually be truly useful to me. Would it also republish group states? My guess is no, since as you point out there is no real state with groups. Either way, it would be useful :)

Koenkk, 2020-12-20:
> Would it also republish group states? My guess is no, since as you point out there is no real state with groups. Either way, it would be useful :)

This would be republish the states as available in the `data/state.json` file.

closed 2021-01-28 by github-actions[bot]