Koenkk/zigbee2mqtt #4991: Philips hue cannot be turned ON - error (0x10: MEM_ERROR)

waqaslam, 2020-11-15:
I'm having a zigbee2mqtt integration together with some nodered flows. Everything used to work perfectly fine for more than a year till I recently upgraded to version **1.16.1**

I'm using a CC2531 with firmware 20190608 and I only have total 3 devices attached to it. (philips hue, ikea motion sensor and ikea remote)

The flow is something like: when my ikea's motion sensor sets occupancy to true, I turn on the philips hue, and when occupancy is false then turn off the hue. Simple...

This is when occupancy is turned to **false**. All seems fine...

```
Nov 15 13:52:46 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 13:52:46: MQTT publish: topic 'zigbee2mqtt/ikea-motion-sensor', payload '{"battery":34,"linkquality":76,"occupancy":false,"requested_brightness_level":254,"requested_brightness_percent":100,"update_available":false}'
Nov 15 13:52:46 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 13:52:46: MQTT publish: topic 'zigbee2mqtt/philips-hue-corridor', payload '{"brightness":254,"color":{"x":0.4571,"y":0.4097},"color_mode":2,"color_temp":220,"linkquality":81,"state":"OFF"}'
```
and this is when occupancy is set to **true**. This is where the problem lies:

```
Nov 15 13:52:55 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 13:52:55: MQTT publish: topic 'zigbee2mqtt/ikea-motion-sensor', payload '{"battery":34,"linkquality":81,"occupancy":true,"requested_brightness_level":254,"requested_brightness_percent":100,"update_available":false}'
Nov 15 13:52:55 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:error 2020-11-15 13:52:55: Publish 'set' 'state' to 'philips-hue-corridor' failed: 'Error: Command 0x0017880102377224/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":11763,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":72,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,110,1]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))'
Nov 15 13:52:55 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 13:52:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'philips-hue-corridor' failed: 'Error: Command 0x0017880102377224/11 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (SREQ '--> AF - dataRequest - {\"dstaddr\":11763,\"destendpoint\":11,\"srcendpoint\":1,\"clusterid\":6,\"transid\":72,\"options\":0,\"radius\":30,\"len\":3,\"data\":{\"type\":\"Buffer\",\"data\":[1,110,1]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"philips-hue-corridor"},"type":"zigbee_publish_error"}'
```
and this is when occupancy is again timed out and switched back to **false**. All seems fine.

```
Nov 15 13:55:55 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 13:55:55: MQTT publish: topic 'zigbee2mqtt/ikea-motion-sensor', payload '{"battery":34,"linkquality":81,"occupancy":false,"requested_brightness_level":254,"requested_brightness_percent":100,"update_available":false}'
Nov 15 13:55:55 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 13:55:55: MQTT publish: topic 'zigbee2mqtt/philips-hue-corridor', payload '{"brightness":254,"color":{"x":0.4571,"y":0.4097},"color_mode":2,"color_temp":220,"linkquality":78,"state":"OFF"}'`
```
Btw, it does work fine when I manually inject the state to on/off for the bulb.

These are the logs when operated manually:
```
Nov 15 14:10:32 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 14:10:32: MQTT publish: topic 'zigbee2mqtt/philips-hue-corridor', payload '{"brightness":254,"color":{"x":0.4571,"y":0.4097},"color_mode":2,"color_temp":220,"linkquality":78,"state":"OFF"}'
Nov 15 14:10:37 ZigPi zigbee2mqtt[279]: Zigbee2MQTT:info  2020-11-15 14:10:37: MQTT publish: topic 'zigbee2mqtt/philips-hue-corridor', payload '{"brightness":254,"color":{"x":0.4571,"y":0.4097},"color_mode":2,"color_temp":220,"linkquality":78,"state":"ON"}'
```

Do you have any idea why turning the philips hue to ON gets failed in connection with ikea's motion sensor?

Koenkk, 2020-11-16:
Appears somehow your coordinator got corrupted, does reflashing fix it?

waqaslam, 2020-11-16:
I can try reflashing it.
Is there any newer firmware for CC2531 other than 20190608?

Koenkk, 2020-11-17:
No (default variant), but there is an alternative variant (source routing) which provides increased stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

closed 2020-12-25 by github-actions[bot]