Koenkk/zigbee2mqtt #4993: state "TOGGLE" ignores "transition"

tricoos, 2020-11-15:
## What happened
Sending the TOGGLE state to an IKEA Tradfri Lamp ignores any transition setting in the payload. Replacing TOGGLE by ON or OFF utilizes the specified transition. Probably a general issue, but only tested with Tradfri.

## What did you expect to happen
I expected the TOGGLE to use the specified transition. However, it uses some default transition of maybe ~0.5s regardless of the value entered.

## How to reproduce it (minimal and precise)
```
zigbee2mqtt/{deviceOrGroup}/set
{"state":"TOGGLE", "transition":0}
```
or
```
zigbee2mqtt/{deviceOrGroup}/set
{"state":"TOGGLE", "transition":2}
```
The transition value is always ignored.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: 1.8.0 (source routing)

tricoos, 2020-12-16:
I still see this annoying behavior every day. Will it be fixed - can it be fixed?

tricoos, 2020-12-16:
Maybe someone can try if this also happens with other lamps?

dmitrystarikov, 2020-12-16:
Aqara bulb (ZNLDP12LM), zigbee2mqtt v1.16.2, coordinator cc2652p (by egony)
Same issue

Koenkk, 2020-12-17:
In Zigbee the on/off/toggle commands actually cannot have a transition. Currently this is worked around by internally changing this to a brightness command. But this cannot be done for the toggle.

tricoos, 2020-12-17:
Well, why then is there a transition for toggle in the first place? If it would **not** transition I would understand it. But it does without being told so.

Koenkk, 2020-12-17:
@tricoos the transition you are seeing is the default IKEA on/off transition, this differs per vendor and depends on the firmware of the bulb.

tricoos, 2020-12-17:
Ah, I see. I actually thought that Zigbee2MQTT would handle that as it knows the status of each bulb, especially because a Zigbee device can only be connected to a single hub which means that there is no way that anyone except Zigbee2MQTT could change the bulb status.
From a user standpoint the transition looks like a bug because none was specified. What do you think about toggling in Zigbee2MQTT? Maybe with a different command to be backwards-compatible with the old one?
Or would that not be possible? The other way would be me having to manage the bulb status and implement toggling on the client side just parsing Zigbee2MQTT bulb status sent via MQTT and storing it in Redis for comparison (no. I'm not using HA ;)

MattWestb, 2020-12-17:
@tricoos The coordinator is only alone as long it don't have any other devices joined in the network.
All controlling devices (thru zigbee standard) can / is sending commands to other devices / groupes and if its one coordinator / router or one groupe its up to the device how its sending / addressing the command. Its by the zigbee standard and if using ZLL / ZB3 light and occupancy they is "living" without coordinator is on line (opposite to HA devices that normally must having one coordinator). The standard with light is the status is reported from the lights and the coordinator is listening to it and making the status of the device / groupe. 
Controlling devices is only sending commands to devices / groupes and don't reporting to the coordinator (ZLL and ZB3 standard). And its up to the coordinator catching all and making the status of devices and groupes with devices from that. Some devices is not reporting status (like some Philips HUE lights) then its normal pulling them for getting the status (no zigbee standard).

tricoos, 2020-12-17:
@MattWestb Thanks for the detailed explanation. 
So I guess I have to use the aforementioned workaround instead. Still looks like a bug to me (not from the technical standpoint, just from the user standpoint) - so maybe the toggle functionality (transition) should be explained in the documentation as not to confuse users like me.

closed 2021-01-24 by github-actions[bot]