Koenkk/zigbee2mqtt #4996: Fail start coordinator for 4 times on boot

ster1um, 2020-11-16:
On boot up of hass.io (docker container), the coordinator fail start 4 times and last go up.
It's normal this behavior?

LOG

> [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-11-16T12:17:25: PM2 log: Launching in no daemon mode
2020-11-16T12:17:26: PM2 log: App [npm:0] starting in -fork mode-
2020-11-16T12:17:26: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js
Zigbee2MQTT:info  2020-11-16 12:17:38: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-16.12-17-35' filename: log.txt
Zigbee2MQTT:info  2020-11-16 12:17:38: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
Zigbee2MQTT:info  2020-11-16 12:17:38: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-11-16 12:17:59: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-16 12:17:59: Failed to start zigbee
Zigbee2MQTT:error 2020-11-16 12:17:59: Exiting...
Zigbee2MQTT:error 2020-11-16 12:17:59: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.16.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-16T11_17_59_630Z-debug.log
2020-11-16T12:18:00: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-16T12:18:00: PM2 log: App [npm:0] starting in -fork mode-
2020-11-16T12:18:00: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js
Zigbee2MQTT:info  2020-11-16 12:18:10: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-16.12-18-07' filename: log.txt
Zigbee2MQTT:info  2020-11-16 12:18:10: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
Zigbee2MQTT:info  2020-11-16 12:18:10: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-11-16 12:19:01: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-16 12:19:01: Failed to start zigbee
Zigbee2MQTT:error 2020-11-16 12:19:01: Exiting...
Zigbee2MQTT:error 2020-11-16 12:19:01: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.16.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-16T11_19_01_745Z-debug.log
2020-11-16T12:19:02: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-16T12:19:02: PM2 log: App [npm:0] starting in -fork mode-
2020-11-16T12:19:02: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js
Zigbee2MQTT:info  2020-11-16 12:19:11: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-16.12-19-09' filename: log.txt
Zigbee2MQTT:info  2020-11-16 12:19:11: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
Zigbee2MQTT:info  2020-11-16 12:19:11: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-11-16 12:19:31: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-16 12:19:31: Failed to start zigbee
Zigbee2MQTT:error 2020-11-16 12:19:31: Exiting...
Zigbee2MQTT:error 2020-11-16 12:19:31: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.16.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-16T11_19_32_096Z-debug.log
2020-11-16T12:19:33: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-16T12:19:33: PM2 log: App [npm:0] starting in -fork mode-
2020-11-16T12:19:33: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js
Zigbee2MQTT:info  2020-11-16 12:19:42: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-16.12-19-40' filename: log.txt
Zigbee2MQTT:info  2020-11-16 12:19:43: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
Zigbee2MQTT:info  2020-11-16 12:19:43: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-11-16 12:20:03: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-16 12:20:03: Failed to start zigbee
Zigbee2MQTT:error 2020-11-16 12:20:03: Exiting...
Zigbee2MQTT:error 2020-11-16 12:20:03: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.16.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-16T11_20_03_515Z-debug.log
2020-11-16T12:20:05: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-16T12:20:05: PM2 log: App [npm:0] starting in -fork mode-
2020-11-16T12:20:05: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js
Zigbee2MQTT:info  2020-11-16 12:20:13: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-16.12-20-11' filename: log.txt
Zigbee2MQTT:info  2020-11-16 12:20:14: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
Zigbee2MQTT:info  2020-11-16 12:20:14: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-11-16 12:20:25: zigbee-herdsman started
Zigbee2MQTT:info  2020-11-16 12:20:25: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"}'
Zigbee2MQTT:info  2020-11-16 12:20:25: Currently 2 devices are joined:
>

Anjerlaan, 2020-11-16:
Do you have multiple USB devices connected to your system?
If so, the path to the serial (USB)port in configuration.yaml may be different (e.g.):
 
serial:
  port: /dev/ttyUSB0

Problem with USB devices is that after each (re)boot, the ports may have switched: USB0 becomes USB1 and vice versa.
This can be solved following: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

ster1um, 2020-11-23:
I have reflashed the usb dongle to 1.2 firmware, no problem at all.
The problem is the firmware 3.x!

closed 2020-11-23 by ster1um