Koenkk/zigbee2mqtt #4997: Aqara b1 hand_open option not onby default, json error.

Dennisd80, 2020-11-16:
## What happened:  Aqara b1curtain doesn't open by hand, therefor i've made and automation,(home assistant but with this automation it says: ERROR: Invalid JSON '{ 'options': { 'hand_open': true } }', skipping...

automation 
`- alias: handmatig gordijen
  trigger:
  - platform: homeassistant
    event: start
  action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/Gordijnen Jayden Zolder/set
      payload: '{ ''options'': { ''hand_open'': true } }'`

## What did you expect to happen That the curtain would open/close bij hand

## How to reproduce it (minimal and precise) , rerun script.

## Debug info
Zigbee2MQTT version: Latest dev  1.16.1-dev
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-11-16:
Instead of `''` use `"`.

Dennisd80, 2020-11-16:
Your the best

closed 2020-11-16 by Dennisd80