Koenkk/zigbee2mqtt #4999: Periodic temporary and permanent loss of devices.

sysardex, 2020-11-16:
## What happened

I have 56 zigbee devices.
Tradfri bulbs. More than 40 pcs
Xiaomi flood sensors
Xiaomi temperature sensors
Xiaomi buttons
Zigbee routers in every room
Issue 1
Periodically during use, one or more devices become unavailable for commands and status. The device is available on the network again if you restart the zigbee2mqtt server or restart the device itself. Logs error 2020-11-16 02:14:19: Failed to ping '0x000b ...

Issue 2
Periodically after restarting zigbee2mqtt it loses 1-2 devices. Most often these are tradfri bulbs. Logs error 2020-11-16 03:27:01: Entity '0x086b ....' is unknown. The device appears only if you add it to zigbee2mqtt again. Restoring the old version of the zigbee2mqtt database does not help. The coordinator loses the device permanently.

## What did you expect to happen

## How to reproduce it (minimal and precise)

Nothing, Common use.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531 as coordinator, CC2530 as routers.
Adapter firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20200211,"transportrev":2},"type":"zStack30x"}'

igosoft, 2020-11-16:
I'm facing very similar issue since a week

ademobile, 2020-11-17:
I am also having issues with devices (seems to be end points) just losing network connectivity. This problem has started in the last few weeks since at least 1.16.0, however it may have started since1.15.0

Currently using 1.16.1
CC1352P_2 with Z-Stack_3.x.0 20201026 firmware
59 devices

sinegovsky-ivan, 2020-11-17:
Confirm - i have the same issue since 2-3 weeks (from 1.15.0).
I have to re pair devices once at 2-3 days.

Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531 as router, Aqara Wall Switch (with neutrals - works as router) - 12 devices and 20 another different devices.

In 99% of cases - aqara devices going offline. All other devices works fine.

MattWestb, 2020-11-17:
@sinegovsky-ivan Some Aqara / Xiaomi wall switches and outlets is not routers an acting as no sleeping end devices and cat routing traffic in the network. Only checking that is working as router and if its not doing that you need more routers for building the mesh and your CC-2531 is very near its limit (12+20=32 devices) and getting / having problems :-(

sinegovsky-ivan, 2020-11-17:
@MattWestb i'm sorry. my mistake. i checked and i have 2538+ 2592.

as i see on the map almost all of my switches have the childs:
[image]

MattWestb, 2020-11-17:
Great then you should not having performance problem with the coordinator !!!
Only trying getting LQi being enough so the mesh can working and not only forming one start that is not tolerant for interference and routers going down.

Koenkk, 2020-11-18:
All: Xiaomi battery powered devices leaving is a known issue which unfortunately cannot be solved from Zigbee2MQTT. Read more about what you can do here: https://www.zigbee2mqtt.io/devices/WXKG12LM.html#troubleshooting-device-stops-sending-messagesdisconnects-from-network

@sysardex do you have the debug log from starting Z2M till the error?

> Issue 2
Periodically after restarting zigbee2mqtt it loses 1-2 devices. Most often these are tradfri bulbs. Logs error 2020-11-16 03:27:01: Entity '0x086b ....' is unknown. The device appears only if you join it to zigbee2mqtt again. Restoring the old version of the zigbee2mqtt database does not help. The coordinator loses the device permanently.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

igosoft, 2020-11-18:
@Koenkk I use the same setup since a year and above issues appeared from probably 1.15 😀. Restarting zigbee2mqtt in HA always solves the problem

sysardex, 2020-11-18:
> To enable debug logging set in `configuration.yaml`:
> 
> ```yaml
> advanced:
>   log_level: debug
> ```

`error 2020-11-18 16:43:00: Publish 'set' 'state' to '0x000b3cfffef85603' failed: 'Error: Command 0x000b3cfffef85603/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
debug 2020-11-18 16:43:00: Error: Command 0x000b3cfffef85603/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))
info  2020-11-18 16:43:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000b3cfffef85603' failed: 'Error: Command 0x000b3cfffef85603/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"0x000b3cfffef85603"},"type":"zigbee_publish_error"}'
error 2020-11-18 16:43:00: Publish 'set' 'state' to '0x086bd7fffe3b6261' failed: 'Error: Command 0x086bd7fffe3b6261/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
debug 2020-11-18 16:43:00: Error: Command 0x086bd7fffe3b6261/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))
info  2020-11-18 16:43:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x086bd7fffe3b6261' failed: 'Error: Command 0x086bd7fffe3b6261/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"0x086bd7fffe3b6261"},"type":"zigbee_publish_error"}'
`

sysardex, 2020-11-18:
Can changing channel help me? Can channel 11 be occupied by other devices?

Koenkk, 2020-11-18:
@igosoft can you provide your debug log when sending commands fail, and what adapter/firmware version are you using?

@sysardex with what adapter/firmware is this?

fwelvering, 2020-11-18:
I have this issue with ikea tradfri dimmer. 
After a few hours It stops working. When pressing a button a few times i see an announce in the log. Afterparty that It will work again

sysardex, 2020-11-18:
> @sysardex with what adapter/firmware is this?

V3 20200211

igosoft, 2020-11-18:
@Koenkk Here the latest log. Why it receives messages ? --> "Received MQTT message on 'homeassistant/ "

`debug 2020-11-18 16:15:47: Loaded state from file /share/zigbee2mqtt/state.json
info  2020-11-18 16:15:47: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-18.16-15-44' filename: log.txt
debug 2020-11-18 16:15:47: Removing old log directory '/share/zigbee2mqtt/log/2020-11-12.19-30-57'
info  2020-11-18 16:15:47: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
info  2020-11-18 16:15:47: Starting zigbee-herdsman...
debug 2020-11-18 16:15:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2020-11-18 16:15:50: zigbee-herdsman started
info  2020-11-18 16:15:50: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
debug 2020-11-18 16:15:50: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
info  2020-11-18 16:15:50: Currently 0 devices are joined:
info  2020-11-18 16:15:50: Zigbee: disabling joining new devices.
info  2020-11-18 16:15:50: Started frontend on port 8099
info  2020-11-18 16:15:50: Connecting to MQTT server at mqtt://homeassistant:1883
info  2020-11-18 16:15:50: Connected to MQTT server
info  2020-11-18 16:15:50: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-11-18 16:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"1.16.1","config":{"advanced":{"adapter_concurrent":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/share/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00158d00018b51d6":{"friendly_name":"lazienka-prysznic-ruch","occupancy_timeout":300},"0x00158d0001d91ed5":{"friendly_name":"korytarz-ruch","occupancy_timeout":120},"0x00158d000200b2c2":{"friendly_name":"ogrod-zimowy-drzwi"},"0x00158d0002132f54":{"friendly_name":"lazienka-prysznic-przycisk"},"0x00158d000247cc05":{"friendly_name":"lazienka-prysznic-temp"},"0x00158d00024802a0":{"friendly_name":"salon-temp","humidity_precision":1,"pressure_precision":1,"temperature_precision":1},"0x00158d0004514848":{"friendly_name":"lazienka-wanna-ruch","occupancy_timeout":300},"0x842e14fffe5e6ab5":{"friendly_name":"ikea-range-extender"}},"experimental":{"new_api":true,"output":"json"},"external_converters":[],"frontend":{"port":8099},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","include_device_information":false,"server":"mqtt://homeassistant:1883","user":"mqtt"},"passlist":[],"permit_join":false,"queue":{},"serial":{"disable_led":false,"port":"/dev/ttyACM0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extended_pan_id":"0xdddddddddddddddd","pan_id":6754},"permit_join":false,"version":"1.16.1"}'
info  2020-11-18 16:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]}},"110":{"bindings":[],"clusters":{"input":[],"output":[]}},"12":{"bindings":[],"clusters":{"input":[],"output":[]}},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]}},"2":{"bindings":[],"clusters":{"input":[],"output":[]}},"242":{"bindings":[],"clusters":{"input":[],"output":[]}},"3":{"bindings":[],"clusters":{"input":[],"output":[]}},"4":{"bindings":[],"clusters":{"input":[],"output":[]}},"47":{"bindings":[],"clusters":{"input":[],"output":[]}},"5":{"bindings":[],"clusters":{"input":[],"output":[]}},"6":{"bindings":[],"clusters":{"input":[],"output":[]}},"8":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Coordinator","ieee_address":"0x00124b000be67171","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"}]'
info  2020-11-18 16:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/groups', payload '[]'
[... cut ...]
debug 2020-11-18 19:45:47: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-11-18 19:50:47: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-11-18 19:55:47: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-11-18 20:00:47: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-11-18 20:05:47: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-11-18 20:10:47: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-11-18 20:15:47: Saving state to file /share/zigbee2mqtt/state.json
`

baderas, 2020-11-19:
I had the same problem with my CC2531 and afterwards with a CC2591.
(long before 1.15.0).
Solution: Using slaesh’s CC2652RB stick. Since then, nothing got lost anymore.
Around 25+ Devices, CC2531/CC2591 began to have those kinds of problems.

sysardex, 2020-11-19:
Now the random device timeout again. Debug did not provide any further information. The device has been unresponsive for over an hour. But if I restart z2m now then the device will be online again.

`
error 2020-11-19 12:03:56: Failed to ping '0x000b57fffe51ea59'
debug 2020-11-19 12:06:06: Failed to ping '0x000b57fffe51ea59'
debug 2020-11-19 12:08:17: Failed to ping '0x000b57fffe51ea59'
...
debug 2020-11-19 13:37:07: Failed to ping '0x000b57fffe51ea59'
debug 2020-11-19 13:39:17: Failed to ping '0x000b57fffe51ea59'
`

Koenkk, 2020-11-19:
@sysardex I suggest to disable the availability feature first and see if the problems are fixed after that.

@igosoft your logs says: `Currently 0 devices are joined:` , so it's not strange that you don't receive messages from devices.

brhahlen, 2020-11-19:
@Koenkk I've seen the suggestion to disable the availability feature before, but _where_ exactly can I do that? Either I'm missing something obvious in the config, or I'm not seeing it... Thanks!

sinegovsky-ivan, 2020-11-19:
@brhahlen just remove or set to '0' parameter `availability_timeout` from section `advanced`

sinegovsky-ivan, 2020-11-19:
@Koenkk my issue was fixed after i enabled 2 features at config:
```
availability_timeout: 60
report: true
```

depen84, 2020-11-19:
I got 4 Ikea routers and they are all affected, types below. I replaced my CC2531 adapter on Z2M v1.14.2, with a zzh  adapter (fw: 20200925) on Z2M v1.16.1. Before it was rock solid, now it´s very unreliable :-(

@sinegovsky-ivan your suggestion didn't work for me.

My affected Ikea routers:
LED1545G12 fw:2.3.050, TRADFRI bulb E27 WS opal 980lm 
L1529 fw:1.2.224, TRADFRI Floalt LED light panel
E1746 fw:2.2.005, TRADFRI signal repeater

From the logs:
```
Nov 19 19:51:13 zigbee2mqtt npm[3621]: Zigbee2MQTT:info  2020-11-19 19:51:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'ikea_floalt_01' failed: 'Error: Command 0x000d6ffffe6bcc10/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"ikea_floalt_01"},"type":"zigbee_publish_error"}'
```

MattWestb, 2020-11-19:
@defcon84 You have no connection to the device because of very bad signal or interference.
Check WiFi channels around with mobile wifi scan app and also if you using your microwave in the near its blocking all radio communications then its doing heating.

defcon84, 2020-11-19:
@MattWestb I believe you meant to tag @depen84.. Almost the same :)

MattWestb, 2020-11-19:
More than right @defcon84 !!

Sorry for the wrong addressing it was not intended for the Netherlands (I think) ;-(

defcon84, 2020-11-19:
@MattWestb no worries and you are correct, sir. :)

igosoft, 2020-11-19:
Hello Koenkk

Indeed, there weren't any devices connected at that time. However it was
consequence of 3 times a day restarts of zigbee2mqtt after a week. I was
restarting zigbee2mqtt due to reported/observed lack of stability

 I've paired again two sensors and they were successfuly discovered and
connected. I will observe those for a day or two and let you know

depen84, 2020-11-20:
@MattWestb It might actually be a bad antenna. I paired up everything a couple of days ago and all worked just fine for almost a day. Then it just went sideways. All My routers were just gone and then I found this thread. When I read your comment I recall that I made the antenna point up instead of down just to make it blend in better. Today I made it point down again and now my zigbee just works. I'll have to leave it like this to know for sure and I'll report back in a couple of days.

MattWestb, 2020-11-20:
@depen84 It can also being bad souldrings on the PCB that is moving with temperature, humidity and angle of the PCB and the antenna. Some have reporting that TI modules (with PCB and external antennas) it having better radio performance with the internal PCB antennas then the external one but then its something wrong with the module.
And with un luck you can getting one reflection of the radio waves that is canceling all out and you getting no connection at all (= mother nature is a BI...).

You is doing it right that leaving it in that position for some days. After that you can trying other poisons but normally its the best 90° from the earth plane but its very depending of the antenna and the surrounding.

If like looking on cracy antennas search for "coke antenna", its looks not real but is working well if making it right :-)))

PS: All IKEA devices except the control outlet (have one large condensator very near the side of the antenna inside the case) have very good radio performance compared with the old TI devices (CC-253X chips) and is normally working good as routers in zigbee 3 network.

brhahlen, 2020-11-21:
I have one Tradfri bulb that disconnected and will not reconnect again, for some reason. All other devices, even in the same room, still work without problem.
I am using a Electrolama zzh! CC2652R, and it paired the first time properly.
I'm on the latest Z2M Dev Docker (auto updating every night), and the latest stable firmware.

sysardex, 2020-11-22:
> @sysardex I suggest to disable the availability feature first and see if the problems are fixed after that.

At first glance, the problem appears less frequently. But it still exists. Can I try to change the channel?

Zigbee2MQTT:error 2020-11-22 20:11:06: Publish 'set' 'state' to '0x000b57fffe8c1d62' failed: 'Error: Command 0x000b57fffe8c1d62/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 56471 - 1 - 193 - 6 - 11 after 10000ms)'
Zigbee2MQTT:debug 2020-11-22 20:11:06: Error: Command 0x000b57fffe8c1d62/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 56471 - 1 - 193 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Zigbee2MQTT:info  2020-11-22 20:11:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000b57fffe8c1d62' failed: 'Error: Command 0x000b57fffe8c1d62/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 56471 - 1 - 193 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x000b57fffe8c1d62"},"type":"zigbee_publish_error"}'

baldfox, 2020-11-23:
> @Koenkk my issue was fixed after i enabled 2 features at config:
> 
> ```
> availability_timeout: 60
> report: true
> ```

I've added this too, will report back if it fixes my issues.

vlavrynovych, 2020-11-27:
> I've added this too, will report back if it fixes my issues.
Hi @baldfox , any updates on this? Very interesting if this config helps.

baldfox, 2020-11-28:
> > I've added this too, will report back if it fixes my issues.
> > Hi @baldfox , any updates on this? Very interesting if this config helps.

Unfortunately, it's not made any difference. The dropouts are happening more frequently now. I did upgrade HA to 0.118.4 as I saw there was a fix for something MQTT related but it didn't do anything. My zigbee devices drop to unavailable. I restart the add-on, they come back immediately.

igosoft, 2020-11-28:
@baldfox I confirm the same behaviour with my side

igosoft, 2020-11-28:
I just restarted HA and that also solved my problem (unit it will appear again). So far I was restarting Z2M

baldfox, 2020-11-29:
I've been restarting HA too but it always returns. Definitely an issue. I think for me it started around 1.16.0 on z2m.

sysardex, 2020-11-29:
Another loss of device after restart z2m.
```
Zigbee2MQTT:debug 2020-11-30 02:16:14: Received MQTT message on 'zigbee2mqtt/0x086bd7fffe5690e2/set' with data '{"state": "ON"}'
Zigbee2MQTT:info  2020-11-30 02:16:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x086bd7fffe5690e2"},"type":"entity_not_found"}'
Zigbee2MQTT:error 2020-11-30 02:16:14: Entity '0x086bd7fffe5690e2' is unknown
```

baldfox, 2020-11-30:
OK something else am noticing. It might be related or not, but whenever I need to restart my Z2m add-on, i also have one mqtt (tasmota 9.1) device that also switches on. It happens every single time despite me clearing retained, and going through powerretain, switch retain etc etc.... here's an example of how frequently restarts are needed or happening - 
[image]

igosoft, 2020-12-01:
@baldfox @sysardex  I finally solved my problem which is  most probably very similar to the issues you described. Here's what I did to resolve it:
- I replaced a battery in one of endpoint- temperature sensor 
- I set wifi chanel far from chanel 11 used by zigbee2mqtt (eg 6 or 2 )
- I replaced WiFi router to d-link DIR-605 (cheap and old hw)
- I reduced WiFi router signal power from 100% to 35% 

It works fine since 4 days. I didn't do anything else, no software updates or configuration changes. I could be very funny coincidence that it also happened  for you at the same time (month)

Most probably my previous router changed WiFi chanel automatically to something near to 11 and WiFi signal was too strong even if I moved the router far from raspberry pi and on top of that some sensors with weak batteries weren't able to handle WiFi interferences. Please notice that I wasn't able to pair one sensor with battery level 91%. I had to replace battery

sysardex, 2020-12-01:
More errors and complete failure of all devices

```
Zigbee2MQTT:error 2020-12-01 21:07:32: Publish 'set' 'state' to '0x000b57fffe89b6d3' failed: 'Error: Command 0x000b57fffe89b6d3/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 51311 - 1 - 126 - 6 - 11 after 10000ms)'
Zigbee2MQTT:debug 2020-12-01 21:07:32: Error: Command 0x000b57fffe89b6d3/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 51311 - 1 - 126 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Zigbee2MQTT:info  2020-12-01 21:07:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000b57fffe89b6d3' failed: 'Error: Command 0x000b57fffe89b6d3/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 51311 - 1 - 126 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x000b57fffe89b6d3"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:info  2020-12-01 21:07:32: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe9e1cf7', payload '{"brightness":254,"linkquality":0,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:error 2020-12-01 21:07:33: Publish 'set' 'state' to '0x000b57fffe51ebbb' failed: 'Error: Command 0x000b57fffe51ebbb/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 31750 - 1 - 127 - 6 - 11 after 10000ms)'
Zigbee2MQTT:debug 2020-12-01 21:07:33: Error: Command 0x000b57fffe51ebbb/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 31750 - 1 - 127 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Zigbee2MQTT:info  2020-12-01 21:07:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000b57fffe51ebbb' failed: 'Error: Command 0x000b57fffe51ebbb/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 31750 - 1 - 127 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x000b57fffe51ebbb"},"type":"zigbee_publish_error"}'
```

@igosoft Maybe then it is better to take z2m to another channel?

igosoft, 2020-12-01:
@sysardex yes I gave a try but my z2m complains when I save changed channel. I think you will get the best results if you can combine weaker WiFi signal with different WiFi chanel than z2m setup :) Here is my z2m error which appears when I change the chanel number. However I can change the other setting with no errors :)

 "Failed to save addon configuration, not a valid value for dictionary value @ data['options']. Got {'data_path': '/share/zigbee2mqtt', 'external_converters': [], 'devices': 'devices.yaml', 'groups': 'groups.yaml', 'homeassistant': True, 'permit_join': False, 'mqtt': {'base_topic': 'zigbee2mqtt', 'server': 'mqtt://homeassistant:1883', 'user': 'mqtt', ......"

baldfox, 2020-12-02:
@igosoft Thanks for the update. Am using Pfsense with Unifi and nothing has changed for over a year in terms of wifi channels etc. My wifi is channel 11 and my zigbee is 11. I moved zigbee to 15, but have now lost all devices. I guess I need to restart the whole server / zigbee usb stick? As you can see zigbee 11 and wifi 11 is not an overlap per se, also strange things have been working normally by and large for the last 18mths - 2yrs. So I don't think it's wifi interference.
[image]

baldfox, 2020-12-02:
I've completely lost my zigbee devices now having changed the network channel on z2m. Any ideas how to bring it back?

fwelvering, 2020-12-02:
In my experiance, After a channel change you have to re-join all devices

igosoft, 2020-12-02:
@baldfox Im not WiFi expert but if you compare frequencies of WiFi Chanel and ZigBee channel than you still might have overlapping frequencies. Please notice that I change also WiFi signal power. I will play with it more to figure it out where is the root cause.

I'm sorry that you lost the devices. I confirm that you have to pair them again - no other changes are required to my knowledge.  

My current setup:
ZigBee Chanel 11
WiFi Chanel 7
WiFi signal strength : 35%
Believe me I didn't do any other changes.

Maybe you can also check other WiFi router around you by running WiFiAnalyzer application and compare frequencies of your ZigBee with other WiFi networks frequencies

https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

baldfox, 2020-12-02:
Thanks for coming back to me guys @fwelvering @igosoft - it's my fault i should have done more homework before switching the channel. I would never have bothered otherwise. I was only trying it just on the off chance. There's no other wifi signals around me, to run interference and nothings changed in 18mths except the actual software. I'll try and resurrect the sensors when I can be bothered. Will they retain their existing default names do you think or you have to re-input all the details again? I only ask as I have loads of node-red stuff linked.

igosoft, 2020-12-02:
@baldfox When I lost all of zigbee devices too and performed pairing again than I did not have change anything in HA, node-red or Z2M to make it work again

baldfox, 2020-12-03:
> @baldfox When I lost all of zigbee devices too and performed pairing again than I did not have change anything in HA, node-red or Z2M to make it work again

I tried to re-add a device, but it kept telling me the name was in use. I just restored instead. Less hassle. Will see what happens with the latest release of z2m... 1.16.2 is just out

sysardex, 2020-12-25:
Nothing has changed for me. I need to reboot z2m and devices multiple times daily. Once every 2-3 days I need to do a pair of lost devices again. Very often, due to timeouts, not all lamps in the chandelier turn off. It is very uncomfortable.

timdonovanuk, 2021-01-03:
I am having the same issue across multiple devices/brands with a cc1352p-2:

```
2020-12-30T23:47:05+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-30 23:47:05: Failed to ping 'garage_strip_light_2'
2020-12-30T23:53:56+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-30 23:53:56: Failed to ping 'bathroom_2_ceiling_light_4'
2020-12-31T00:06:00+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 00:06:00: Failed to ping 'garage_strip_light_2'
2020-12-31T00:17:41+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 00:17:41: Failed to ping 'bedroom_1_ceiling_light_2'
2020-12-31T00:34:26+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 00:34:26: Failed to ping 'bathroom_2_ceiling_light_2'
2020-12-31T00:52:49+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 00:52:49: Failed to ping 'garage_strip_light_2'
2020-12-31T00:52:54+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 00:52:54: Failed to ping 'socket_10'
2020-12-31T00:55:56+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 00:55:56: Failed to ping 'garage_strip_light_1'
2020-12-31T01:00:41+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 01:00:41: Failed to ping 'bedroom_1_ceiling_light_1'
2020-12-31T01:12:44+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 01:12:44: Failed to ping 'garage_strip_light_2'
2020-12-31T01:21:02+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 01:21:02: Failed to ping 'socket_10'
2020-12-31T01:21:36+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 01:21:36: Failed to ping 'lower_hallway_light'
2020-12-31T01:29:08+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 01:29:08: Failed to ping 'garage_ceiling_light_1'
2020-12-31T02:06:31+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 02:06:31: Failed to ping 'garage_strip_light_2'
2020-12-31T02:14:41+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 02:14:41: Failed to ping 'bedroom_1_ceiling_light_1'
2020-12-31T02:48:21+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 02:48:21: Failed to ping 'garage_strip_light_2'
2020-12-31T02:48:27+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 02:48:27: Failed to ping 'garage_strip_light_1'
2020-12-31T03:06:27+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:06:27: Failed to ping 'garage_strip_light_1'
2020-12-31T03:20:01+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:20:01: Failed to ping 'kitchen_ceiling_light_1'
2020-12-31T03:20:38+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:20:38: Failed to ping 'bedroom_1_lamp_1'
2020-12-31T03:21:33+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:21:33: Failed to ping 'bathroom_2_ceiling_light_1'
2020-12-31T03:22:02+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:22:02: Failed to ping 'socket_5'
2020-12-31T03:30:27+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:30:27: Failed to ping 'socket_10'
2020-12-31T03:30:36+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:30:36: Failed to ping 'lower_hallway_light'
2020-12-31T03:41:01+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 03:41:01: Failed to ping 'kitchen_ceiling_light_2'
2020-12-31T04:06:36+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 04:06:36: Failed to ping 'lower_hallway_light'
2020-12-31T04:51:02+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 04:51:02: Failed to ping 'kitchen_ceiling_light_1'
2020-12-31T05:01:48+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:01:48: Failed to ping 'garage_strip_light_2'
2020-12-31T05:11:25+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:11:25: Failed to ping 'socket_10'
2020-12-31T05:11:36+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:11:36: Failed to ping 'lower_hallway_light'
2020-12-31T05:11:45+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:11:45: Failed to ping 'garage_strip_light_2'
2020-12-31T05:18:25+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:18:25: Failed to ping 'bathroom_2_ceiling_light_3'
2020-12-31T05:18:28+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:18:28: Failed to ping 'socket_10'
2020-12-31T05:37:49+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:37:49: Failed to ping 'garage_strip_light_1'
2020-12-31T05:57:35+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 05:57:35: Failed to ping 'garage_strip_light_2'
2020-12-31T06:00:33+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 06:00:33: Failed to ping 'garage_strip_light_2'
2020-12-31T06:19:44+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 06:19:44: Failed to ping 'garage_strip_light_1'
2020-12-31T06:20:03+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 06:20:03: Failed to ping 'socket_10'
2020-12-31T06:27:37+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2020-12-31 06:27:37: Failed to ping 'lower_hallway_light'
[... cut ...]
2021-01-03T07:01:59+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2021-01-03 07:01:59: Failed to ping 'garage_strip_light_2'
2021-01-03T07:07:32+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2021-01-03 07:07:32: Failed to ping 'bedroom_1_lamp_1'
2021-01-03T07:46:45+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2021-01-03 07:46:45: Failed to ping 'entrance_light'
2021-01-03T07:53:48+00:00 docker zigbee2mqtt[368]: [31mZigbee2MQTT:error[39m 2021-01-03 07:53:48: Failed to ping 'bedroom_1_lamp_1'
```

walthowd, 2021-01-07:
Just so everyone is aware, there is a low level stack bug in the SiLabs EFR32 used by IKEA devices that causes them to crash. 

When they do crash, they still respond to multicast/group commands but no longer to direct NWK/APS queries and they also seem to stop routing for any connected end device children. 

This bug is fixed in Emberznet 6.7 and higher from SiLabs -- but a majority of IKEA devices are running earlier 6.0 to 6.5 releases that are all affected by this bug. 

The bug occurs when a bulb is processed a ZDO parent announcement and other traffic arrives. Seems to be triggering more often on large networks. 

[image]

MattWestb, 2021-01-07:
The bug validated by triggering it (with ZHA custom component) in the deCONZ github furum of the ZHA deves for around 2 months ago :-)

baldfox, 2021-01-09:
> Nothing has changed for me. I need to reboot z2m and devices multiple times daily. Once every 2-3 days I need to do a pair of lost devices again. Very often, due to timeouts, not all lamps in the chandelier turn off. It is very uncomfortable. I tried changing the frequency several times.

So I upgraded everything to the latest version of HA (2020.12.7) and OS (5.10) and the problem still remains. I cannot for the life of me figure this out. If I restart the z2m add-on the aqara temp sensors (x2) come on immediately. I replaced the battery in one despite it showing 37% left, but it made no difference. Anyone have any success yet? I'm really loathed to change the channel and rejoin everything again, when nothing has changed from a z2m channel/wifi power/wifi channel or positioning perspective.

timdonovanuk, 2021-01-12:
>  If I restart the z2m add-on the aqara temp sensors (x2) come on immediately. 

Are you 100% certain of this? I thought the same thing for a while, but HA was simply getting the last MQTT message from the sensors, stored in the MQTT queue. In HA it looked like the sensors went "unavailable", I rebooted z2m, and they became "available" again, but they weren't actually available - I could tell because they always just showed the last temps they reported before vanishing.

baldfox, 2021-01-12:
> Are you 100% certain of this? I thought the same thing for a while, but HA was simply getting the last MQTT message from the sensors, stored in the MQTT queue. In HA it looked like the sensors went [...]

I have them set up in node-red (HA add-on) and as soon as I restart z2m and ask alexa or google home what the temperature is on either of them, they come back with the correct response, When they're unavailable, The command in NR falls over as they can't access HA for the reading. Now could the temperatures be frozen and report old temps, there's a slim chance it could be I suppose, but it's difficult for me to tell as I have no warning when or for how long they go unavailable for.

raf1111, 2021-01-22:
Zigbee2mqtt
Current version: 1.17.0.1
Coordinator LAUNCHXL-CC26X2R1. with the newest firmware
64 Devices

For some reason the Phillips lamps doesn't work after the restart from my home server.
Seems there disconnected  

Also 4 wall switches 230V are disconnected 

the end nodes work correctly 

[image]

raf1111, 2021-01-22:
Before the Restart 
[image]

raf1111, 2021-01-22:
Is there a solution to quickly reconnect everything ?

sysardex, 2021-01-27:
Today, after rebooting HA and Z2M, one more paired device was lost. It is still present in devices.yaml, but it is not in the list of paired devices. The latest information about this device in the logs:
debug 2021-01-27 10:47:13: Received MQTT message on 'zigbee2mqtt/0x086bd7fffe03dbd9/availability' with data 'online'

igosoft, 2021-01-28:
I also did some tests last weekend. I shutdown my rp3 for few hours (>4).
However when the system started, none of paired devices were found. I had
to pair everything again 🥺

[3 more replies]

closed 2021-05-12 by github-actions[bot]