Koenkk/zigbee2mqtt #5000: Add network_key, no more pairing

klorydryk, 2020-11-17:
Zigbee2MQTT version 1.16.1
Bridge CC2531 revision 20190223

My configuration was without network_key, but was working well for more than one year. I decided to add the network_key. Then I've been unable to add any device.
I removed them in the list and tried to re-pair them, but they never appears in the journal. I stopped zigbee2mqtt then removed the usb bridge, put it back, start zigbee2mqtt (multiple times).

I reinstalled zigbee2mqtt from the ground: same result, but after a lot of attempts, I've ben able to pair ONE device, which appeared in the device configuration list, but failed to interview... then was forever silent.

The last log : https://pastebin.com/0dzuezCp

Thanks !

Koenkk, 2020-11-18:
Make sure to turn of your complete old network first (e.g. remove all bulbs from power)

klorydryk, 2020-11-19:
I did that, then stopped the service, remove the usb bridge, re-install the usb bridge, restart the service, tried to add a device.
Same result.
I'll try to remove and re-install zigbee2mqtt with all devices off.

klorydryk, 2020-11-19:
Still the same : https://pastebin.com/mYWj7C7P

I try to add different devices (Osram Smart+, Ikea Almhult...), nothing move on the log, like it is seeing nothing.

Koenkk, 2020-11-19:
@klorydryk also check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

klorydryk, 2020-11-22:
After reading it, I decided to flash my bridge with a recent firmware (I took [this one](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190619.zip)). As you can see [here](https://pastebin.com/hccePVWA) the result is the same.
And nothing's still pairing. :(

Koenkk, 2020-11-23:
Would you be able to sniff the traffic when putting the device into pairing mode? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

klorydryk, 2020-11-23:
I switched my key for another one, working at a friend's place. It worked. I switched again, for mine, it worked.
I don't understand, sorry for this :s

closed 2020-11-23 by klorydryk