Koenkk/zigbee2mqtt #5002: Netvox Z809A pairs and communicates, but gets configuration error: UNSUPPORTED_ATTRIBUTE

upiw, 2020-11-17:
# Bug Report
## What happened

- Netvox Z809A Device joined 
- Successfully interviewed and the device was successfully been paired
- Device is supported, identified as: Netvox Power socket with power consumption monitoring (Z809A)

```text
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Power socket with power consumption monitoring","friendly_name":"0x00137a000000bfc1","model":"Z809A","supported":true,"vendor":"Netvox"},"type":"pairing"}'
```

But then I got the following configuration failure _Attribute Not Supported_:
```text
Zigbee2MQTT:error 2020-11-17 10:29:44: Failed to configure '0x00137a000000bfc1', attempt 1 (Error: ConfigureReporting 0x00137a000000bfc1/1 haElectricalMeasurement([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:386:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```

## What did you expect to happen
When I toggle to power switch button on the Z809A I get the following 
```json
{"linkquality":26,"state":"OFF"}
```
 but I really need the additional power monitoring attributes as specified in the _node_modules/zigbee-herdsman-converters/devices.js_ file, i.e. rmsVoltage, rmsCurrent, activePower, and powerFactor.

```json
 {
        zigbeeModel: ['Z809AE3R'],
        model: 'Z809A',
        vendor: 'Netvox',
        description: 'Power socket with power consumption monitoring',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.on_off, fz.electrical_measurement_power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await configureReporting.onOff(endpoint);
            await configureReporting.rmsVoltage(endpoint);
            await configureReporting.rmsCurrent(endpoint);
            await configureReporting.activePower(endpoint);
            await configureReporting.powerFactor(endpoint);
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    }

```

## Debug Info
Zigbee2MQTT version: 1.16.1 (commit #6b32f30)
Adapter hardware: Conbee 2
```text 
MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"6b32f30","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.1"}'
```
Adapter firmware version: Unknown

Koenkk, 2020-11-18:
It looks that your specific device doesn't not support power metering (?). Are you able to confirm it does when using this switch with another bridge (in case you have it)?

upiw, 2020-11-18:
Hi there. I have a C++ file that we used in a working application that definitely did report various power stats including immediate usage, accumulated since reset, and voltage.
I wonder if this would be of any help?  I’ll include it.  Even if you could give me some pointers and I could have a go at helping out somehow? I’m not a C++ coder but I do know ES6.

The functions relating to Netvox should contain very useful information that the dev team found differed from the Netvox documentation.

Kind regards, and thanks so much for you help.  I have 20 of these devices and it would be absolutely super to get them working with your great zigbee2mqtt project.

Upi (Peter)

Koenkk, 2020-11-19:
If you have it working with a different bridge try to sniff the traffic of this while pairing it until you receive power reports. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

upiw, 2020-11-20:
Thanks for you reply Koen,

Wow - that’s what it takes eh?  Hmmm.

OK - can I try one last thing.  I’ve included a JSON capture of the response from the application sent to the browser for a paired device.

Is there enough info in this to go forward?

Regards,

Upi (Peter)

Koenkk, 2020-11-21:
Maybe I'm missing it, but I don't see a json capture?

upiw, 2020-11-22:
Thanks for you reply Koen,  Triying again to send this JSON capture

Wow - that’s what it takes eh?  Hmmm.

OK - can I try one last thing.  I’ve included a JSON capture of the response from the application sent to the browser for a paired device.

Is there enough info in this to go forward?

Regards,

Upi (Peter)

closed 2020-12-30 by github-actions[bot]

upiw, 2021-01-21:
Hi Koen,

I’ve purchased an zzh CC2652R multi protocol RF stick.
When that arrives, can I uses that as the hardware to sniff the Zigbee traffic?

Regards,

Upi

Koenkk, 2021-01-21:
There is no sniffer firmware available AFAIK, only CC2531 for now: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

You can ask TI to provide a sniffer firmware here: https://e2e.ti.com/

upiw, 2021-01-31:
Thanks for the sniffer info. For me that will be the last resort.

I’ve managed to muck around with the devices NinjaBlocks and Netvox and I’m now getting the following coming up in the debug logs.
This is a message from one of my Z809A wall plugs (I modified the Netvox device entry)

debug 2021-01-31 14:45:06: Received Zigbee message from '0x78a0510000479372', type 'attributeReport', cluster 'seMetering', data '{"57344":0,"57345":244,"57346":0,"57347":609}' from endpoint 1 with groupID null

This shows that the device is really capable of reporting power, voltage, current, and power since factory reset.

What I am now missing is how to get these values being converted and published from the zigbee2mqtt bridge into an MQTT topic?

I luckily happend to have one Ninja Block wall socket  and I can see ‘power’ being part of the payload in MQTT for that device:
{"linkquality":170,"power":300,"state":"OFF"}

it has a diferent debug log content:
Jan 31 15:17:03 raspberrypi npm[25826]: Zigbee2MQTT:debug 2021-01-31 15:17:03: Received Zigbee message from '0x00137a000001a4e4', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":0}' from endpoint 1 with groupID null

Can you please help to point me to some information that will allow me to work out what to do?

I love your project.  I also think that Zigbee is a great protocol.  So rock solid.

Thanks for you efforts.

Cheers Upi

Koenkk, 2021-01-31:
- Can you share the data/database.db entry of this device?
-  If `Received Zigbee message from '0x78a0510000479372', type 'attributeReport', cluster 'seMetering', data '{"57344":0,"57345":244,"57346":0,"57347":609}'` is indeed a power report it not in a standard way. Can you try figuring out what each value means here?

upiw, 2021-02-01:
```json
{"id":12,"type":"Router","ieeeAddr":"0x78a0510000479372","nwkAddr":63275,"manufId":4255,"manufName":"iiNet Labs","powerSource":"Mains (single phase)","modelId":"iiNet Smart plug","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6,21,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"iiNet Smart plug","manufacturerName":"iiNet Labs","powerSource":1,"zclVersion":3,"appVersion":20,"stackVersion":47,"hwVersion":12,"dateCode":"20130826"}},"genOnOff":{"attributes":{"onOff":1}},"seMetering":{"attributes":{"57344":0,"57345":247,"57346":0,"57347":710}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05dec4","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff05dec4","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff05dec4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":20,"stackVersion":47,"hwVersion":12,"dateCode":"20130826","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1612181724550}
```

Thing is that I seem to be getting two debug entries for the same device: (yes the device id is different but they are the same type of device)

```json
{"level":"debug","message":"Received Zigbee message from '0x78a0510000479383', type 'attributeReport', cluster 'seMetering', data '{\"57344\":0,\"57345\":245,\"57346\":0,\"57347\":606}' from endpoint 1 with groupID null"}
```
AND

```json
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/0x78a0510000479383', payload   \"energy\":606,\"linkquality\":16,\"power\":0,\"state\":\"OFF\"}'"} {"energy":606,"linkquality":16,"power":0,"state":"OFF"}
```

Sorry - but I don't really know how I managed to get that topic publish working.  I previously got something similar by hacking the _fromZigbee.js:metering_ property and using the _ 57346_ like the following and that has probably made it work :

```javascript
 if (msg.data.hasOwnProperty('57346')) {
                let power = msg.data['57346'];
                if (factor != null) {
                    power = (power * factor) * 1000; // kWh to Watt                                                                                                               
                }
                payload.power = precisionRound(power, 2);
            }
```

This is what my device change looks like. I've hijacked the Netvox one for my iiNet smart plug

``` // Netvox                                                                                                                                                                     
 {
        zigbeeModel: ['iiNet Smart plug'],
        model: 'Z809A',
        vendor: 'iiNet',
	description: 'Power socket with power consumption monitoring',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await reporting.onOff(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.rmsCurrent(endpoint, {min: 1, change: 1});
            await reporting.activePower(endpoint, {min: 1, change: 1});
            await reporting.powerFactor(endpoint, {min: 1, change: 1});
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    },
```

Clearly I am kind-of getting somewhere but I really have little idea about how I really should be doing it.
Also - I don't have any idea how to make a new device configuration or what should be in the _configure_ function for my device; where these _await reporting.rmsVoltage_ type functions are defined and what is appropriate for my device?

There is a _fromZigbee. haElectricalMeasurement_ but it didn't seem to work until I used the modified  _fromZigbee: [fz.metering]

Any help would be most gratefully appreciated.

Cheers Upi

Koenkk, 2021-02-01:
Your device does not support the `haElectricalMeasurement` cluster (`2820` is not in the inClusterList), the `seMetering` cluster is (`1794`). The `seMetering` cluster only supports `energy` and `power`, `voltage` and `current` are not supported by this cluster.

Can you apply the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2188/files and check if you get power and energy reports?

upiw, 2021-05-16:
Hi Koenk,

Its been a while but I’ve worked out (at least one way how) to see the power / energy / voltage etc. from this power adaptor.
I’m not to sure how to formally go from here to a git pull request.

I had to add the following to the metering JSON in  fromZigbee.js

    metering: {
        /**                                                                                                               
         * When using this converter also add the following to the configure method of the device:                        
         * await readMeteringPowerConverterAttributes(endpoint);                                                          
         */
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const payload = {};
            const multiplier = msg.endpoint.getClusterAttributeValue('seMetering', 'multiplier');
            const divisor = msg.endpoint.getClusterAttributeValue('seMetering', 'divisor');
            const factor = multiplier && divisor ? multiplier / divisor : null;

            if (msg.data.hasOwnProperty('instantaneousDemand')) {
                let power = msg.data['instantaneousDemand'];
                if (factor != null) {
                    power = (power * factor) * 1000; // kWh to Watt                                                       
                }
                payload.power = precisionRound(power, 2);
            }
            if (msg.data.hasOwnProperty('57346')) {
                let power = msg.data['57346'];
                if (factor != null) {
                    power = (power * factor) * 1000; // kWh to Watt                                                       
                }
                payload.power = precisionRound(power, 2);
            }
            if (msg.data.hasOwnProperty('57347')) {
                let energy = msg.data['57347'];
                payload.energy = energy;
            }
            if (msg.data.hasOwnProperty('57345')) {
                let voltage = msg.data['57345'];
                payload.voltage = voltage;
            }

            return payload;

        },
    },

This started getting me those values.  I’ve no idea where this code is really meant to go. I’m happy to do the work if you can tell me what I need to do to do this properly/formally?

Also I added this to devices.js

  // Netvox                                                                                                 
    {
        zigbeeModel: ['iiNet Smart plug'],
        model: 'Z809A',
        vendor: 'iiNet',
        description: 'Power socket with power consumption monitoring',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await reporting.onOff(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.rmsCurrent(endpoint, {min: 1, change: 1});
            await reporting.activePower(endpoint, {min: 1, change: 1});
            await reporting.powerFactor(endpoint, {min: 1, change: 1});

        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    },

I think it needed (at least ) the fz.metering fromZigbee array element.  Not sure if the other stuff does anything

I’m happy to do some work on this but I don’t really understand how.  

You can see my debugging output for such a Netvox iiNet (badged) device:

May 16 21:10:45 rPiZb npm[1349]: Zigbee2MQTT:info  2021-05-16 21:10:45: MQTT publish: topic 'zigbee2mqtt/0x78a0510000479371', payload '{"energy":4862,"linkquality":229,"power":76,"state":"ON","voltage":242}'

That is what I was wanting to see. 

Thanks again ,

Cheers Upi

Koenkk, 2021-05-16:
@upiw can you try with the following changes: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2585/files  ? (updated to the latest dev branch since the repository changed a bit, https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

upiw, 2021-05-17:
Thanks Koen - I really appreciate all you have done and are doing to make your platform so helpful to people like me.

I added another condition to the ’fromZigbee.js’.  I needed a condition for curren’ which is reportable by the device in attribute 57344

 if (msg.data.hasOwnProperty('57344')) payload.current = msg.data['57344’];

It didn’t seem to work properly just adding  the fromZigbee attribute to the netvox.js device in your PR.  I actually have a ninja socket adapter but I haven’t tested it with that device file.

I had to add an iiNet.js device to get it to recognise my iiNet branded netvox. (See below)

I don’t know how a device is mapped to a <device>.js file.  Perhaps if it is just the model then we could make this device file more generic to Z809A but it didn’t work for me untill I added a separate device file for it.

const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const reporting = require('../lib/reporting');
const e = exposes.presets;

module.exports = [
    {
        zigbeeModel: ['iiNet Smart plug'],
        model: 'Z809A',
        vendor: 'iiNet',
        description: 'Power socket with current, power, total energy, and voltage monitoring',
        fromZigbee: [fz.on_off, fz.Z809A_metering],
        toZigbee: [tz.on_off],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await reporting.onOff(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint);
        },
        exposes: [e.switch(), e.power(), e.energy()],
    },
];

When I get a chance to fetch my Ninja socket adapter, I can try that out with the pure ’netvox.js’ device.  It might work with the ’ninja_blocks.js’ device.

Thanks again,

Cheers Upi

Koenkk, 2021-05-17:
@upiw If I understand correctly, with this definition your device works as expected?