Koenkk/zigbee2mqtt #5003: Help wanted to add device support for YPHIX 50208689 and 50208690

KobesM, 2020-11-17:
## Information about the device + link
**YPHIX 50208689**
ZigBee Wireless dimmer switch for 1 zone
[https://www.leddirect.nl/amfile/file/download/file/35/product/213/](https://www.leddirect.nl/amfile/file/download/file/35/product/213/)

**YPHIX 50208690**
ZigBee Wireless dimmer switch for 2 zones
[https://www.leddirect.nl/amfile/file/download/file/57/product/224/](https://www.leddirect.nl/amfile/file/download/file/57/product/224/)

## data/database.db entry of the device
..

## My progress
I started to try and figure out how to add the device to devices.js following the instructions: [https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)

I am running the  hassio-zigbee2mqtt addon on hassio.

I downloaded the devices.js and placed it in /share/zigbee2mqtt/ and added the following content:
```js
// YPHIX 50208689
    {
        zigbeeModel: ['ZGRC-KEY-007'],
        model: '50208689',
        vendor: 'YPHIX',
        description: 'Zigbee 2 button wall switch',
        supports: 'on/off, brightness',
        fromZigbee: [],
        toZigbee: [],
    },
```

the log now displays the device:
```
Zigbee2MQTT:debug 2020-11-17 22:09:56: Device '0xccccccfffe8a0bc9' announced itself
Zigbee2MQTT:info  2020-11-17 22:09:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xccccccfffe8a0bc9","ieee_address":"0xccccccfffe8a0bc9"},"type":"device_announce"}'
Zigbee2MQTT:info  2020-11-17 22:09:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xccccccfffe8a0bc9"},"type":"device_announced"}'
Zigbee2MQTT:debug 2020-11-17 22:09:59: Received Zigbee message from '0xccccccfffe8a0bc9', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-17 22:09:59: No converter available for '50208689' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":100}'
Zigbee2MQTT:debug 2020-11-17 22:09:59: Received Zigbee message from '0xccccccfffe8a0bc9', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-17 22:09:59: No converter available for '50208689' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":100}'
Zigbee2MQTT:debug 2020-11-17 22:09:59: Received Zigbee message from '0xccccccfffe8a0bc9', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-17 22:09:59: No converter available for '50208689' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":100}'
Zigbee2MQTT:debug 2020-11-17 22:09:59: Received Zigbee message from '0xccccccfffe8a0bc9', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-17 22:09:59: No converter available for '50208689' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":100}'
Zigbee2MQTT:debug 2020-11-17 22:09:59: Received Zigbee message from '0xccccccfffe8a0bc9', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-17 22:09:59: No converter available for '50208689' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":100}'
Zigbee2MQTT:debug 2020-11-17 22:09:59: Received Zigbee message from '0xccccccfffe8a0bc9', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-17 22:09:59: No converter available for '50208689' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":100}'
```

But i can't seem to figure out how to continue implementing the device from the documentation. Can anyone help me out?

McGiverGim, 2020-11-18:
I've never done it, but seeing the documentation,  you need to look at the point 3: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#3-adding-converters-for-your-device

More or less you need to:
- Look at the file: node_modules/zigbee-herdsman-converters/converters/fromZigbee.js
- Check if there is a converter for battery there with the cluster and data of your log, I think this generic will be enough:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/86192c48f1b6307edb8bcd771ee117544767fa3d/converters/fromZigbee.js#L671-L707

Now you only need to add it to to your code:
`fromZigbee: [fz.temperature],`

KobesM, 2020-12-15:
Decided not to pursue this

closed 2020-12-15 by KobesM