Koenkk/zigbee2mqtt #5004: zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP

lorek123, 2020-11-18:
## What happened
I have some issues with setting correct temperature on GS361A-H04, but I going through logs showed me a messages:
```
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #1041 with data {"type":"Buffer","data":[0]}
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #1043 with data {"type":"Buffer","data":[1]}
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #1043 with data {"type":"Buffer","data":[1]}
```
this maybe a wrong repo to create a issue report, please let me know and I will put it in a correct place if needed.
## What did you expect to happen
If this can help me debug why is thermostat is not working properly then a more descriptive message.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version:  20190608

Rossez, 2020-11-18:
I have simmilar random error:

`zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #366 with data {"type":"Buffer","data":[0]}`

Device: Moes branded Tuya thermostatic valve - model number HY368
Zigbee2MQTT version: 1.16.0
Adapter hardware: Electrolama zig-a-zig-ah!
Adapter firmware version: CC26X2R1_20201026

insipiens, 2020-11-27:
thats the low battery DP, have you upgraded to the latest version of Z2M?

Rossez, 2020-11-27:
I am on  1.16.1. I suppose it will be resolved in next Z2M stable release? Thank you.

stoinov, 2020-12-17:
Want to add my random DPs with Moes TRV. Zigbee2MQTT version 1.16.2 and zigbee-herdsman (0.13.37)
```
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":4,"transid":126,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}}
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":4,"transid":139,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
```
I can see that dp13 [does exist](https://github.com/Koenkk/zigbee-herdsman-converters/blob/d40d3ac027c4413988c29e4f2a9b5923a396e2bf/converters/common.js#L202) but it seems specific for etopThermostat, yet is still in the Moes section.

Kalypox, 2020-12-19:
@stoinov 
Same here, i opened issue:
https://github.com/Koenkk/zigbee2mqtt/issues/5347

Your valve is still working ? Mine is dead !

shirou93, 2021-01-04:
1.17.0 z2m
Still errors

`zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":222,"transid":160,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":222,"transid":173,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":222,"transid":233,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":223,"transid":12,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":223,"transid":40,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":223,"transid":53,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":223,"transid":71,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":223,"transid":84,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":223,"transid":95,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}} zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data`

agavrin, 2021-01-10:
I have the same issue:

```
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":27,"transid":76,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}},
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":27,"transid":89,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}},
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":27,"transid":99,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}},
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":27,"transid":112,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}},

```

sensei73, 2021-01-18:
Same here, seems to prevent updating 'local_temperature_calibration'.

```
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data
```

aka0xxx6, 2021-01-31:
Hi there,
I also got this issue.
```
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":13,"transid":27,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":13,"transid":144,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}}
```

Device: Moes branded https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html
Zigbee2Mqtt version: 1.17.0 (commit #07cdc9d) inside docker

RubenKelevra, 2021-02-24:
I have 4 thermostats and only one on one it seems to show these warnings...

@sensei73 wrote

> Same here, seems to prevent to set any value.

All thermostats work fine for me. So I cannot confirm this.

RubenKelevra, 2021-02-24:
@sjorge noticed this on the commit:

> Should this go under a different manufacturer header? Or is etop and alias for moes?

Source: https://github.com/Koenkk/zigbee-herdsman-converters/commit/5c5d861f9ae490869d3f2b0fa725d8718b9ea4e6#r44059975

@drzony do you have an idea what DP 13 and 115 means? Can we ignore them safely?

RubenKelevra, 2021-02-24:
I created a PR to remove this value, which seems to be used nowhere anyway.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/2290

stoinov, 2021-03-27:
@RubenKelevra and @drzony any update on those DPs?

drzony, 2021-03-27:
@stoinov I don't have any of those TRVs, so I cannot verify.

RubenKelevra, 2021-03-27:
> @RubenKelevra and @drzony any update on those DPs?

They still appear in my logs and don't know how to go forward from here.

The function of those units is definitely not hindered, though.

Btw:
The stale bot is absolutely annoying and closes regularly tickets which are not fixed.

drzony, 2021-03-27:
@RubenKelevra You can try to figure out what those DPs do. https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html can help you reverse engineer the DP. You can maybe compare it to something that is displayed on the valve. Or you can sniff the traffic while using a Tuya gateway.
The warnings are only there to tell someone that wants to do the reverse engineering what needs to be done, they do not impact already implemented functionality.

sjorge, 2021-03-27:
Perhaps these should me logged as debug and not warning. That would probably mean nobody will see them unless actual developers debugging devices. 

That might decrease the chance of someone with a device and the knowledge to solve it, solving it though.

RubenKelevra, 2021-03-27:
Haven't done this yet, not sure I'll find ever time to get my hands dirty with that 🤔

There's nothing on the displays which would indicate that we're missing anything from the data we currently get.

And I don't have a vendor gateway.

ronniebee, 2021-04-05:
following, same issue here...

drzony, 2021-04-05:
@sjorge That's why it is a warning, if someone able to do the work sees it, then he can fix this. It's a warning that the device is not fully implemented. I don't have those valves, so I cannot really do anything about that, unless somebody sniffs the traffic with those DPs while communicating with Tuya gateway and/or is able to pinpoint when those DPs are sent (after a button click? periodically?). Seems like they are some diagnostic data.

drzony, 2021-04-05:
115 is bool, so maybe child lock (somebody could enable it on valve and check if it changed to 1)
13 is bitmap, so maybe error reporting or some other state info (again try clicking the valve and verify if something changes in the messages)

shirou93, 2021-04-05:
@drzony Unrecognized DP #115  appear when i click windows detecion switch.

drzony, 2021-04-05:
@shirou93 Great, does it change? i.e. Does:
`DP #115 with data {"status":13,"transid":27,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}`
change to
`DP #115 with data {"status":13,"transid":27,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[1]}}`
?

shirou93, 2021-04-05:
@drzony [image]

Data 1 enable
Data 0 disable

drzony, 2021-04-05:
@shirou93 I'll make the patch, but there already is window detection on this thermostat, can you verify which exact thermostat do you own?

shirou93, 2021-04-05:
@drzony this thermostat show as https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#tuya-ts0601_thermostat in zigbee2mqtt device panel.

drzony, 2021-04-05:
@shirou93 I meant which physical device do you have. Do you have a link? A picture? Brand name? I think that the person that implemented this thermostat bundled together several different brands which share some of the data points, but not all of them.

drzony, 2021-04-05:
@shirou93 can you share database.db entry for this device?

shirou93, 2021-04-05:
@drzony 
{"id":20,"type":"EndDevice","ieeeAddr":"0xbc33acfffe5d3449","nwkAddr":29939,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"G��'\u0011","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1617655616225}

drzony, 2021-04-05:
@shirou93 Do you have a link where you bought it? Does it look like the one in the picture here https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#tuya-ts0601_thermostat? What is the model number on the packaging?

shirou93, 2021-04-05:
@drzony  link https://a.aliexpress.com/_vOIW4N
I remember blue package no white as users photos on reviews.

I can check exactly model on package tomorrow.

shirou93, 2021-04-06:
@drzony no model info on package :([image]

drzony, 2021-04-08:
@shirou93 We'll need someone with Siterwell thermostat as it seems that the person adding Siterwell merged Moes and Siterwell ones although they differ in DPs

lorek123, 2021-04-08:
Do you need the thermostat I mentioned in the issue? I can help, but I don't have a tuya gateway.

drzony, 2021-04-08:
We need someone with a TRV that looks like this https://www.zigbee2mqtt.io/devices/GS361A-H04.html . Tuya gateway is not required, but some knowledge on how to use changed devices.js and access to console on the machine with z2m is required.
Also we will need someone with https://www.zigbee2mqtt.io/devices/D3-DPWK-TY.html

lorek123, 2021-04-08:
I have the mentioned thermostat, z2m in docker in the latest version. I only changed the coordinator to slaesh one in the meantime

drzony, 2021-04-08:
@lorek123 
OK, so here is what needs to be done:
1. Make sure you are using latest dev docker
2. Add `fz.tuya_data_point_dump` to fromZigbee line in `devices.js` for `Siterwell` TRV
3. Try some buttons on the TRV
4. `tuya.dump.txt` file should appear in your `data` directory
5. Do `tail -f tuya.dump.txt`, you should start seeing some data when you press something on TRV
6. Make a new text file and copy the data with comments e.g.

```
# Setting the temperature from Home Assistant
1616503500073 0x123456789012345 00 3a 66 04 00 02
1616503511566 0x123456789012345 00 3b 66 04 00 00
# Pressing the child lock
1616503500073 0x123456789012345 00 3a 66 04 00 02
1616503511566 0x123456789012345 00 3b 66 04 00 00
# Setting temperature from the valve
```
and so on. If possible cover the whole functionality of the valve.

@shirou93 If you could do the same with `TS0601_thermostat` (in `devices.js`) that would be awesome

closed 2021-05-16 by github-actions[bot]

shirou93, 2021-05-16:
issue still exsist.

hnykda, 2021-08-02:
@drzony Got redirected from https://github.com/Koenkk/zigbee2mqtt/discussions/5748#discussioncomment-1119126 that you would appreciate some help in testing. I have these TRVs. It seems I got actually three different (sub?)types (manufacturer has different weird suffix) even though I bought them in a [batch of 7 pieces](https://www.aliexpress.com/item/1005001895935932.html). They look like this (have a rotary button that can be pushed, two-digit simple display, I suspect it might be [sasswell but who knows](https://expo.tuya.com/product/712066)): 
[image]
[image]

They show up like this in z2m:
[image]

[image]

I am running z2m as a hassio addon, but I can manage docker and stuff, so I guess I should be able to do the necessary testing.

drzony, 2021-08-02:
Those are Saswell TRVs, they are fully implemented. I got hold of manufacturer documentation for them and implemented it accordingly. As for your problem with current temperature: make sure you set mode to heat and turn off away mode. If that does not help, try to re-pair them with z2m (AFAIR enable join in z2m, set TRV to off via dial, then press and hold the dial). They have a bit of lag with reporting the temperature (about 1-2 sec), so keep that in mind and do it slowly :)
Also I have "_TZE200_c88teujp" ones, so they might have changed something in the meantime.

BigValen, 2021-11-22:
I have z2m 1.22.0 on NixOS 21.05. Seems it has problems pushing updates to the TS0601_thermostat (HomeAssistant keeps resending updates to temperature settings that aren't being applied). I wonder if this is related to DP#13 being unreadable.

```
Nov 22 10:49:58 nas zigbee2mqtt[1022385]: Zigbee2MQTT:info  2021-11-22 10:49:58: MQTT publish: topic 'zigbee2mqtt/Attic Rad', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCKED","comfort_temperature":20,"current_heating_setpoint":19,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":113,"local_temperature":15,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":100,"preset":"manual","system_mode":"manual","update":{"state":"idle"},"update_available":true,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":18},{"hour":11,"minute":31,"temperature":23},{"hour":12,"minute":31,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'
Nov 22 10:49:59 nas zigbee2mqtt[1022385]: zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":0,"transid":149,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}}

```