Koenkk/zigbee2mqtt #5005: Command execution delayed (NO issue of MQTT)

iAcki, 2020-11-18:
Hello zigbee2mqtt-team,

I know this question has been asked a number of times, but none of the previous solutions work for me, so I'll ask myself now. I use zigbee2mqtt on my Raspberry Pi3 + together with a Conbee2 stick. I currently use it to control a Philips Hue bulb and a motion sensor via Node-Red. In Node-Red I have a debug node for both the "Mqtt-Out" node and "Mqtt-In" node. When the motion sensor detects movement, an Mqtt command is issued and sent to the lamp. This command is sent within a few milliseconds after the motion was detected and received by the "Mqtt-In" node instantly (maybe 50ms - I'm listing to the whole Zigbee2MQTT - Topic), but the lightbulb still needs 1 - 2 seconds to change it's state (after I noticed "mqtt-in command").

### But the really wired
If I use an inject node to trigger the flow manually, the light turns on within a few milliseconds, if the same flow is triggered via the motion sensor, the lamp takes several seconds. The lamp, the motion sensor and the Pi 3+ are only about 2 meters apart. At the moment I'm not using a USB extension cable, but that shouldn't be the reason, shouldn‘t it?

Does somebody has any idea?

Thanks, Christian 

*** my flow ***
[flows.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/5559046/flows.json.txt)

*** a screenshot of the mqtt messages ***
[image]

P.:S.: Hopefully my attachments are helpful.

Koenkk, 2020-11-18:
Looks like a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/4884#issuecomment-728903121

iAcki, 2020-11-18:
Hi Koenkk,

thank you for your quick response. I‘ve read the issue and it seems to be the same issue  I have. I‘ll stay tuned for an update in the linked issue. 

Thanks, 
Christian

Koenkk, 2020-11-18:
Good, I expect there will be a solution in the coming days, please keep an eye out there. Closing this.

closed 2020-11-18 by Koenkk