Koenkk/zigbee2mqtt #5006: CC2530: displayed wrong firmware build in zigbee2mqtt-GUI

Maik-100, 2020-11-18:
Hi,
I opened an issue because in the GUI of Zigbee2MQTT is displayed the wrong firmeware build date.

see: nurikk/z2m-frontend#140

Please check source code of firmware of CC2530.

Koenkk, 2020-11-18:
I think the firmware build date was not updated, @ptvoinfo is it indeed set to 2018-09-26?

ptvoinfo, 2020-11-18:
@Koenkk yes, I forgot to update the date. We can ignore this. I'll update it on the next version.

Koenkk, 2020-11-18:
@ptvoinfo thanks, closing this because the shown firmware data is correct

closed 2020-11-18 by Koenkk

Maik-100, 2020-11-18:
Thanks, but this issue is present on the router firmware for the CC2531, 
too.

So please update and check the firmware data before you publish a new 
firmware version, it is very easy to check.
This helps to reduce misunderstandings and improves quality of the firmware.