Koenkk/zigbee2mqtt #5007: Adapter crashing when generating network map

JaspervRijbroek, 2020-11-18:
## What happened
When generating a network map (any) the stick crashes.
When the network map is generating I see in the log a message that the request has been accepted, and almost immediately after the adapter disconnects.

```
hass_zigbee2mqtt | Zigbee2MQTT:info  2020-11-18 15:49:17: Starting network scan (includeRoutes 'false')
hass_zigbee2mqtt | Zigbee2MQTT:error 2020-11-18 15:49:18: Adapter disconnected, stopping
```

When checking out dmesg I can find the following message:

```
[355082.784582] usb 2-1.4: device descriptor read/64, error -110
```

After a couple of seconds the adapter power-toggles and resumes operation as usual, but no map.

I don't specifically know what happened as in the past this function worked without issues (no updates have been done)

## What did you expect to happen
The map to be generated

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-11-18:
Can you try with the following firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing , this should provide more stability.

JaspervRijbroek, 2020-11-19:
@Koenkk I'll try this firmware, might take a while before I can flash it ;)

JaspervRijbroek, 2020-11-28:
@Koenkk I have tried this firmware and the network map can be generated again :D, thanks a lot for your support!

closed 2020-11-28 by JaspervRijbroek