Koenkk/zigbee2mqtt #5009: Binary sensors invert option

definitio, 2020-11-18:
Same as #3569.
Workaround with `payload_on` and `payload_off` breaks other sensors (linkquality, battery) after HA 0.118 update.
Schema validation error:
```
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'availability': [{'topic': 'zigbee/bridge/state'}, {'topic': 'zigbee/device/availability'}], 'device': {'identifiers': ['zigbee2mqtt_0x0000000000000001'], 'manufacturer': 'Xiaomi', 'model': 'MiJia door & window contact sensor (MCCGQ01LM)', 'name': 'device', 'sw_version': 'Zigbee2MQTT 1.16.1'}, 'device_class': 'battery', 'json_attributes_topic': 'zigbee/device', 'name': 'device_battery', 'payload_off': False, 'payload_on': True, 'state_topic': 'zigbee/device', 'unique_id': '0x0000000000000001_battery_zigbee', 'unit_of_measurement': '%', 'value_template': '{{ value_json.battery }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 83, in async_discover_sensor config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['payload_off'] 
```

Koenkk, 2020-11-19:
You should not put this in `device_options` as that will change it for all devices (also e.g. bulbs) but define it per device.

CodeOptimist, 2022-05-18:
Not sure if I'm dumb or what but I've been trying for hours and can't seem to get this to work.

```
  '0x00158d0007e3223e':
    friendly_name: Lid
    homeassistant:
      contact:
        payload_on: false
        payload_off: true
```

jmealo, 2023-12-17:
I'm having the same issue... the states are not inverted.

dasb00ter, 2024-02-10:
Is this still open?
`    homeassistant:
      contact:
        payload_on: false
        payload_off: true  `
same as above behaviour

CodeOptimist, 2024-02-11:
Thanks for commenting! I tried that (again), deleted device, restarted zigbee2mqtt, still doesn't work. 🤷‍♂️

CodeOptimist, 2024-02-11:
You know what? It looks like the received messages are actually correct:
(I opened it then closed it, and this shows that, ordered chronologically.)
```
Received 9:46:20 PM
QoS: 0
Payload: battery: 100
contact: true
device_temperature: 25
linkquality: 145
power_outage_count: 3
temperature: 26
voltage: 3045

Received 9:46:23 PM
QoS: 0
Payload: battery: 100
contact: false
device_temperature: 25
linkquality: 153
power_outage_count: 3
temperature: 26
voltage: 3045
```

But even though the last message has `contact: false`, the sensor is still wrong, says `Open` and `Was opened`
[image]

I'm going to remove the payload config now and re-observe.

CodeOptimist, 2024-02-11:
Oh my god I fixed it... how incredibly stupid of a scenario.
These Aqara sensors are inverted by default, to invert them it's actually:
```
        payload_on: false
        payload_off: true
```

CodeOptimist, 2024-05-03:
@Koenkk This can be closed. 😉

closed 2024-05-04 by Koenkk

jonlanceley, 2024-05-30:
Thanks @CodeOptimist i didnt realise that either until i read your post.

To summarise what worked for me with an Aqara door & window contact sensor is edit this file:

/homeassistant/zigbee2mqtt/configuration.yaml

devices:
  '0x00158d0007e3223e':
    friendly_name: something
    homeassistant:
      contact:
        payload_on: true
        payload_off: false

Restart zigbee2mqtt addon.

I didnt need to delete the device, and didnt restart homeassistant.

Before I made this change i did have it already configured to show as an occupancy sensor. I was doing this for a pressure mat underneath a matress.

KalleTangring, 2024-07-27:
Did this to invert a IKEA PARASOLL Door/Window Sensor
Worked perfekt.
Thank you!

----
/homeassistant/zigbee2mqtt/configuration.yaml

devices:
'0x00158d0007e3223e':
friendly_name: something
homeassistant:
contact:
payload_on: true
payload_off: false

Restart zigbee2mqtt addon.
----

zeus85, 2024-11-26:
> Thanks @CodeOptimist i didnt realise that either until i read your post.
>
> To summarise what worked for me with an Aqara door & window contact sensor is edit this file: [...]

I tried it, but without success. In this way or opposite, sensor appearing in HA as "opened". And in MQTT INFO there's always the same for binary_sensor:
payload_off: true
payload_on: false