Koenkk/zigbee2mqtt #5012: [BUG] Cannot read property 'bind' of undefined

kevincaradant, 2020-11-19:
Hi

I don't understand why, but I have 9 switches like this one. (ref: WXKG06LM)
And just for one switch, when I use it, I got this following issue in Z2M:
```
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-19 12:31:54: Failed to configure 'switch_couloir_livingroom', attempt 3 (TypeError: Cannot read property 'bind' of undefined
zigbee2mqtt    |     at bind (/app/node_modules/zigbee-herdsman-converters/devices.js:48:24)
zigbee2mqtt    |     at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:584:19)
zigbee2mqtt    |     at Configure.configure (/app/lib/extension/configure.js:143:45)
zigbee2mqtt    |     at Configure.onZigbeeEvent (/app/lib/extension/configure.js:125:18)
zigbee2mqtt    |     at Controller.callExtensionMethod (/app/lib/controller.js:378:44))
```

What does than means ? 

It's weird because the behavior expected is good. On / Off lights. But everytime I click on it I got this issue.

Zigbee2MQTT version: 1.16.1
Adapter hardware: CC1352P-2
Adapter firmware version: {'meta': {'maintrel': 1, 'majorrel': 2, 'minorrel': 7, 'product': 1, 'revision': 20201026, 'transportrev': 2}, 'type': 'zStack3x0'}

Koenkk, 2020-11-19:
Can you share the data/database.db entry of this device?

kevincaradant, 2020-11-19:
Yes of course, this my my file:

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001ca1b535","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0001a1f256","nwkAddr":37528,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65281":{"1":3085,"3":23,"4":5032,"5":39,"6":[0,5],"10":64542,"11":0,"100":0},"modelId":"lumi.sensor_motion.aq2","appVersion":3}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[],"meta":{}}},"appVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1605819511173}
{"id":3,"type":"Router","ieeeAddr":"0x00124b001ed01121","nwkAddr":35752,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-B-001ZS","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-B-001ZS","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.5"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":229}},"lightingColorCtrl":{"attributes":{"currentX":29687,"currentY":28311,"colorTemperature":375}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.5","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1605820024900}
{"id":4,"type":"Router","ieeeAddr":"0x00124b001ef6a291","nwkAddr":29522,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-B-001Z","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-B-001Z","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.2"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":132}},"lightingColorCtrl":{"attributes":{"currentX":29687,"currentY":28311,"colorTemperature":375}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.2","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1605820024906}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d0001e43e4c","nwkAddr":29847,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,65535,1030,1024,1280,1,3],"outClusterList":[0,25],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":3}},"genBasic":{"attributes":{"65281":{"1":3025,"3":24,"4":17320,"5":70,"6":[0,1],"10":58098,"11":0,"100":0},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20170627","swBuildId":"3000-0001","modelId":"lumi.sensor_motion.aq2"}},"ssIasZone":{"attributes":{}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[],"meta":{}}},"appVersion":5,"stackVersion":2,"hwVersion":1,"dateCode":"20170627","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1605818131910}
{"id":6,"type":"Router","ieeeAddr":"0x00124b001d400dd3","nwkAddr":65357,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-B-001Z","epList":[11,13],"endpoints":{"11":{"profId [... cut ...] ":{"currentX":29687,"currentY":28311,"colorTemperature":375}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"3.0.0","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1605820025466}
```

Is it a way to clean it a little bit, I think 7 devices are wronf because I have only 51 devices but I saw 58 lines :)

kevincaradant, 2020-11-19:
In database, I don't see the "friendly_name"

This is my configuration file to help you to match with the good one.

```
'0x00158d0004833080':
    friendly_name: switch_couloir_livingroom
    retain: true
    legacy: false
```

kevincaradant, 2020-11-19:
Oh that's weird but why some devices like this one here are twice in the database file :

```
{"id":17,"ieeeAddr":"0x00158d0004872f34","nwkAddr":56013,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{}}

AND

{"id":18,"type":"EndDevice","ieeeAddr":"0x00158d0004872f34","nwkAddr":63563,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,65535,1026,1027,1029],"outClusterList":[0,4,65535],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2158}},"msRelativeHumidity":{"attributes":{"measuredValue":4912}},"msPressureMeasurement":{"attributes":{"16":10028,"20":-1,"measuredValue":1019,"scale":-1,"scaledValue":10191}},"genBasic":{"attributes":{"65281":{"1":2995,"4":17320,"5":35,"6":[0,1],"10":58098,"100":2149,"101":4912,"102":101882},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":30,"dateCode":"20191205","swBuildId":"3000-0001","modelId":"lumi.weather"}}},"binds":[],"meta":{}}},"appVersion":5,"stackVersion":2,"hwVersion":30,"dateCode":"20191205","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1605818151167}
```

Koenkk, 2020-11-20:
It seems the device someow did not pair correctly, can you try removing `0x00158d0004833080` from your network with force remove and repair? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

kevincaradant, 2020-11-20:
Thanks 

For the moment after a "force removed" and a re-paired the device. It's seems Okay :)

However, it was not all the time that I got this issue, ( it was like two clicks on five during the day for example).

I'm closing this issue but if this issue appears again I will re-open it again ;)

closed 2020-11-20 by kevincaradant