Koenkk/zigbee2mqtt #5015: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms

ChipWolf, 2020-11-19:
## What happened

This issue has happened with two CC2531s, I ordered the second because I assumed the first was defective.
After a device is flashed, I'm able to associate with all of my ZigBee devices.. shortly after this error appears on the first run, the device stops functioning, including on subsequent runs

Error from the first run
```
(node:17) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise
which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
```

Error from the subsequent runs
```
?[31mZigbee2MQTT:error?[39m 2020-11-19 17:39:52: Error while starting zigbee-herdsman
?[31mZigbee2MQTT:error?[39m 2020-11-19 17:39:52: Failed to start zigbee
?[31mZigbee2MQTT:error?[39m 2020-11-19 17:39:52: Exiting...
?[31mZigbee2MQTT:error?[39m 2020-11-19 17:39:52: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
```

## How to reproduce it (minimal and precise)

Using the following config, run zigbee2mqtt, wait for all the devices to join
```
      advanced:
        log_level: debug
        network_key: [<redacted>]
      devices:
        "0x00158d000533c498":
          friendly_name: Nursery Environment
        "0x680ae2fffef91e07":
          friendly_name: En Suite Motion
        "0xccccccfffe9dcaf9":
          friendly_name: Bedroom Lamp
        "0xccccccfffea9121b":
          friendly_name: En Suite Spot Left
        "0xccccccfffeb7e9f9":
          friendly_name: En Suite Rail Center
        "0xccccccfffeb8bf1a":
          friendly_name: En Suite Rail Near
        "0xccccccfffebac0e8":
          friendly_name: Kitchen Rail Near Outer
        "0xccccccfffee4f302":
          friendly_name: Kitchen Rail Far Outer
        "0xccccccfffee4f36c":
          friendly_name: En Suite Spot Right
        "0xec1bbdfffe2de199":
          friendly_name: Kitchen Rail Far Inner
        "0xec1bbdfffe2f4999":
          friendly_name: En Suite Rail Far
        "0xec1bbdfffe432007":
          friendly_name: Kitchen Rail Near Inner
        "0xec1bbdfffe87fcd9":
          friendly_name: En Suite Remote
        "0xec1bbdfffe8c83e7":
          friendly_name: Kitchen Remote
        "0xec1bbdfffe9eb6b1":
          friendly_name: Bedroom Remote
      groups:
        "1":
          devices:
            - "0xec1bbdfffe432007"
            - "0xec1bbdfffe2de199"
            - "0xccccccfffee4f302"
            - "0xccccccfffebac0e8"
          friendly_name: Kitchen
          retain: false
        "2":
          devices:
            - "0xccccccfffee4f36c"
            - "0xccccccfffea9121b"
          friendly_name: En Suite
          retain: false
      homeassistant: true
      mqtt:
        server: "mqtt://vernemq"
```

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: latest

Koenkk, 2020-11-19:
Looks like your CC2531 crashed, reflashing your CC2531 will likely fix it, to prevent this from happening:
- Make sure to **not** use the availability feature (disabled by default): https://www.zigbee2mqtt.io/information/availability.html
- Try switching to the source routing firmware, this provides more stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
- If this doesn't help you have to upgrade to a more powerful coordinator: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

ChipWolf, 2020-11-19:
Looks like stopping zigbee2mqtt, unplugging the adapter, plugging it back in, and starting zigbee2mqtt in that order fixes it briefly.

I believe it's on source routing firmware already, but I'll be sure to flash it with that one next time.

I only have a handful of devices, so it'd be surprised if I needed a more powerful one, but I'll consider it. Do you have a specific recommendation @Koenkk?

kevincaradant, 2020-11-19:
Hi

I think he will recommand to you the ZZH that he have for his own production setup ;)

https://electrolama.com/projects/zig-a-zig-ah/

ChipWolf, 2020-11-20:
Looks like flashing source routing worked. I was previously flashing default for anyone reading this in future.

closed 2020-11-20 by ChipWolf

ChipWolf, 2020-11-20:
Ta @kevincaradant I'll pick up a couple when they're back in stock

ChipWolf, 2020-11-20:
I'm unable to join any more than 5 devices with source routing, same issue as #2448

kevincaradant, 2020-11-20:
Hoping what I m going to say could help you but If you installed the source routing firmware, it's normal to have only 5 devices on the co-ordinator because you can only have 5 direct childrens like that. 

If you want more than 5 devices keeling the source firmware routing, I think you need at least one router on your production devices. That could be a bulb, a plug ect.

Look the readme of the @Koenkk github with firmware. This is a screenshot from my phone but you can see direct children 5 :)

[image]

ChipWolf, 2020-11-20:
TIL @kevincaradant. Thanks again
I'll investigate how to set this up, is there a good Slack/Discord for this kind of support?

Koenkk, 2020-11-20:
@ChipWolf , @kevincaradant is right, to fix this reflash your CC2531 again with the source routing FW and first pair your routers.

ChipWolf, 2020-11-20:
@Koenkk I paired some of the TRADFRI bulbs first which I'm guessing work as routers, is there something special I need to do in order for the 'non-routers' to pair to the routers?

ChipWolf, 2020-11-20:
Worked perfectly, lots of bulb removing and replacing later and it looks like the environment is sorted.
Thanks for your help!

Zeddy-, 2021-02-02:
Just wanted to say that i have the same thing happening with my Slaesh's CC2652RB using the latest Firmware (Flashed yesterday).

The network was running fine for two hours, then i noticed that a device that i thought was linked suddenly did this:
info  2021-02-02 02:26:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"eTRV0100-4","ieee_address":"0x14b457fffed5539a"},"type":"device_announce"}'
info  2021-02-02 02:26:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"eTRV0100-4"},"type":"device_announced"}'

Which after a while triggered this it seems:
info  2021-02-02 02:29:31: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices' *****
info  2021-02-02 02:29:31: MQTT publish: topic 'zigbee2mqtt/bridge/log' *****

And that's when it stops working.

flatsiedatsie, 2021-12-20:
I just saw this error when trying to pair an IKEA lightbulb to Zigbee2MQTT running on a raspberry pi with a Sonoss Zigbee 3.0 Plus USB stick.

Tomk211, 2022-01-24:
I saw it in the same configuration (Sonoff Zigbee 3.0 Plus USB stick) but with trying to pair IKEA Tradfri remote controller....

ChipWolf, 2022-01-25:
@flatsiedatsie @Tomk211 did you both flash the source routing FW?

Tomk211, 2022-01-25:
Im using this one: CC1352P2_CC2652P_launchpad_coordinator_20211217.hex

ChipWolf, 2022-01-25:
@Tomk211 Ah I'm not personally familiar with that device, perhaps open a new ticket

flatsiedatsie, 2022-01-25:
@ChipWolf I don't remember, sorry. I have both, so it could be either.

jb123567, 2022-01-25:
I'm also experiencing this issue with a ZZH using the latest coordinator firmware (20211217). A restart of the Zigbee2MQTT add on temporarily sorts the issue, but it occurs again within a few hours.

Tomk211, 2022-01-26:
In my case pairing could be started now successfully surprisingly without changing or restarting something...

ChipWolf, 2022-01-27:
@Koenkk can we lock this conversation plz with the view to create a new issue if people are still facing the same problem