Koenkk/zigbee2mqtt #5017: Silvercrest strip plug

MiguelAngelLV, 2020-11-19:
## Information about the device + link
..

https://www.lidl.si/sl/p/product-recommendation/razdelilec-z-usb-vticnicami/p59289

## data/database.db entry of the device
..

Actually is detect as Tuya TS011F and only can control the first plug istead of three but it show three endpoints

```
0xbc33acfffe69942c
-- Endpoint 1
---- Output clusters
------- genOta
------- genTime
---- Input clusters
-------- genBasic
-------- genIdentify
-------- genGroups
-------- genScenes
-------- genOnOff
-- Endpoint 2
----- Input clusters
------- genIdentify
------- genGroups
------- genScenes
------- genOnOff
-- Endpoint 3
----- Input clusters
------- genIdentify
------- genGroups
------- genScenes
------- genOnOff
-- Endpoint 242
---- Output clusters
------- greenPower

```

sjorge, 2020-11-19:
Can  you share the database entry for the device?

MiguelAngelLV, 2020-11-20:
````json
{
   "id":6,
   "type":"Router",
   "ieeeAddr":"0xbc33acfffe69942c",
   "nwkAddr":38491,
   "manufId":4098,
   "manufName":"_TZ3000_1obwwnmq",
   "powerSource":"Mains (single phase)",
   "modelId":"TS011F",
   "epList":[
      1,
      2,
      3,
      242
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":266,
         "inClusterList":[
            0,
            3,
            4,
            5,
            6
         ],
         "outClusterList":[
            25,
            10
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"TS011F",
                  "manufacturerName":"_TZ3000_1obwwnmq",
                  "powerSource":1,
                  "zclVersion":3,
                  "appVersion":69,
                  "stackVersion":0,
                  "hwVersion":1,
                  "dateCode":""
               }
            }
         },
         "binds":[
            
         ],
         "meta":{
            
         }
      },
      "2":{
         "profId":260,
         "epId":2,
         "devId":266,
         "inClusterList":[
            3,
            4,
            5,
            6
         ],
         "outClusterList":[
            
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "meta":{
            
         }
      },
      "3":{
         "profId":260,
         "epId":3,
         "devId":266,
         "inClusterList":[
            3,
            4,
            5,
            6
         ],
         "outClusterList":[
            
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "meta":{
            
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[
            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":69,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1605805143334
}
````

sjorge, 2020-11-20:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1506

You want to update the above entry so the fingerprint key reads

```
fingerprint: [
  {modelID: 'TS011F', manufacturerName: '_TZ3000_wzauvbcs'},
  {modelID: 'TS011F', manufacturerName: '_TZ3000_1obwwnmq'},
],
```

sjorge, 2020-11-20:
Let me know if that works, I can do a PR, if it does.

sehraf, 2020-11-20:
For the record: Lidl seems to have multiple Zigbee devices https://www.lidl.de/de/smart-home/s7382185. The gateway is branded Silvercrest as well.
(Though not available in Germany yet)

sjorge, 2020-11-20:
Yeah a bunch showed in on the zigbee alliance website, all seem to be tuya based it rebrands.

So it should hopefully be just adding the fingerprints to existing devices

~ sjorge

MiguelAngelLV, 2020-11-21:
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1506
> 
> You want to update the above entry so the fingerprint key reads
> 
> ```
> fingerprint: [
>   {modelID: 'TS011F', manufacturerName: '_TZ3000_wzauvbcs'},
>   {modelID: 'TS011F', manufacturerName: '_TZ3000_1obwwnmq'},
> ],
> ```

I use Home Assistant Addon... How I can change these line?

MiguelAngelLV, 2020-11-21:
Um... It seems that we already have a fix for this...

https://github.com/Koenkk/zigbee-herdsman-converters/commit/94b6b98e168e6bf39a977dcce0957bc56d2a75d5

closed 2020-11-21 by MiguelAngelLV