Koenkk/zigbee2mqtt #5018: Zigbee2Mqtt fails with conbee II

oliviermichaelis, 2020-11-19:
## What happened
Zigbee2mqtt did not get into an operational state using the conbee II stick. The web dashboard was in a loading/initialising state. The permit_join button did not work, I was unable to pair any devices. 
## What did you expect to happen
Be able to use Zigbee2mqtt with the conbee II.
## How to reproduce it (minimal and precise)
Started docker container with the following config:
```
homeassistant: true
permit_join: false
mqtt:
  server: "mqtt://mosquitto.default.svc.cluster.local:1883"
serial:
  adapter: deconz
  port: "/dev/ttyACM0"
frontend:
   port: 8080
experimental:
  new_api: true
advanced:
  log_level: debug
```
and with the `DEBUG=zigbee-hersman*` environment variable

## Debug info
Zigbee2MQTT version:  `latest-dev` (49f37ac)
Adapter hardware: Conbee II
Adapter firmware version: `deCONZ_ConBeeII_0x26670700.bin.GCF`
Hardware: Raspberry Pi 3b+

Log is found at: https://pastebin.com/gXaFaBN1
I'm happy to provide any requested logs/configs/information and willing to test any preliminary fixes. Thanks!

taugusti, 2020-11-20:
Hi,
your problem is that you need to find the right port.
  port: "/dev/ttyACM0"
is for the CC2531 device.

In Hass:
go to supervisor->system->(3 vertical dots)->hardware
[image]
in the serial section you will find something with "dresden" in the text - that is your port identifier:
[image]

in my case:
port: "/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1997293-if00"

LouiS22, 2020-11-20:
Also you don't need the " "-s.

Mine is the following below.

serial:
  port: /dev/ttyACM1
  adapter: deconz

oliviermichaelis, 2020-11-20:
Hi @taugusti, 
Thanks for your suggestion, however, I'm fairly certain that the serial device port is the correct one as specified in this comment: https://github.com/Koenkk/zigbee-herdsman/issues/72#issuecomment-593945839. I've double checked it in the linux `/sys/` tree. Nevertheless I've tried it out with `/dev/serial/by-id` without success.

Hi @LouiS22,
As you pointed out it is not necessary to use double quotes, however I think it's perfectly valid yaml according to the specification :)

I've since tried it out on an AMD64 machine with success - Zigbee2mqtt started up within seconds and I was able to use the web overlay and pair a device. I suspect that it has something to do with the ARM64 architecture, the operating system or the hardware. Do you have any pointers?

oliviermichaelis, 2020-11-20:
I've endlessly tried to get it to work on different hardware (Raspberry Pi 3/4), tried out all possible combinations of firmware and zigbee2mqtt versions (different dev versions) only to get it working with a USB extension cable... With the extension cable, the logs looked completely different. Even within a couple of centimeters it wasn't able to pair. Seems like the Conbee II can not be next to power cables, metal or electronic devices :)

closed 2020-11-20 by oliviermichaelis