Koenkk/zigbee2mqtt #5019: How to force Z2M to send again the last state known by Z2M ?

kevincaradant, 2020-11-19:
Hi

I don't know if it's an issue or just me who don't understand how is it working but today I had a power down in my building (3 seconds) and then the power come back.

However, all lights in my appartment was ON and stay like that :(. I don't want it.

I setup my NUC to POWER ON after failure power, so 3 minutes later, Z2M and HA runs again, I thought that the lights was going down because it was the last state in memory for all my lights but nothing happens and everythiung stay like that until I click on each Domotized switch.

Do you have any idea why ? is it linked to Z2M ? Home Assistant ? 

I'm lost about htis subject.

Thank you

MiguelAngelLV, 2020-11-20:
You can configure these devices with "retain" flag

kevincaradant, 2020-11-20:
Hi

I forgot to say every devices has already 'retain: true'

But it seems to does not work :(

kevincaradant, 2020-11-20:
@MiguelAngelLV , do you know how is it working ? Should I write or do something else than other to write in `configuration.yaml` the attribute `retain: true` ?

This is my configuration.yaml:

```
homeassistant: true
permit_join: true
frontend:
  port: 9090
experimental:
  new_api: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://xxxxxxxxxxx'
  user: xxxxxx
  password: xxxxxxxx
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 6755
  channel: 25
  baudrate: 115200
  rtscts: false
  availability_timeout: 60
devices:
  '0x00158d0002136633':
    friendly_name: switch_mini_kitchen
    retain: true
    legacy: false
  '0x00158d0002135b0b':
    friendly_name: switch_mini_table_basse_livingroom
    retain: true
    legacy: false
  '0x00158d0001e4524c':
    friendly_name: open_close_window_livingroom
    retain: true
  '0x00158d000201b485':
    friendly_name: open_close_kitchen
    retain: true
  '0x00158d0001e45264':
    friendly_name: open_close_door_livingroom
    retain: true
  '0x00158d0001e43e4c':
    friendly_name: motion_toilet
    retain: true
    occupancy_timeout: 600
  '0x00158d0001a1f256':
    friendly_name: motion_desktop
    retain: true
    occupancy_timeout: 1800
  '0x00158d000222e35e':
    friendly_name: motion_kitchen
    retain: true
    occupancy_timeout: 600
  '0x00158d0001b1448d':
    friendly_name: open_close_desktop
    retain: true
  '0x00158d0001a1f2c3':
    friendly_name: motion_bedroom
    retain: true
    occupancy_timeout: 600
  '0x00158d0001bc23c0':
    friendly_name: water_leak_bathroom
    retain: true
  '0x00124b001eadcc46':
    friendly_name: gledopto_kitchen
    retain: true
  '0x00124b001d4344fd':
    friendly_name: gledopto_1_bathroom
    retain: true
```

Is it from HA or Z2M that I should manage the auto send of the last state ? 

Thank you

sjorge, 2020-11-20:
z2m does not on it's own, apply the previous state after a device has been power cycled.

It depends on the bulb itself what the default state is, for most this will be on at 100% brightness so that traditional light switches still work.

Some brands, like Philips Hue allow you to configure the state after a power cycle: https://www.zigbee2mqtt.io/devices/929002335001.html#power-on-behavior but most brands do not.

kevincaradant, 2020-11-20:
After writing this message, I realized it was for sure HA that he should take in consideration the MQTT message from Z2M :)

So I think it's a possible HA issue in my case. I didn't found any new events captured by HA after Z2M send them during the restart.

But after to think about it, Z2M just send as payload "state: OFF" and for "HA" the state was already "OFF" before the power down for HA it self. I think HA doesn't understand that when Z2M send the "state: OFF" state, HA should apply this state on my devices.

So, is it a wrong or missing information from Z2M which should specify a  special action or something else to prevent HA or an other domoticz system that is the retain message and just a state information message ?

You see what I mean, not sure to be clear ;)

insipiens, 2020-11-27:
If using Mosquitto broker - have you got persistent=true in your mqtt broker configuration? this way the current state is saved to a file periodically. Without this the current state of devices is only held in memory - so it is lost if the machine crashes.

kevincaradant, 2020-11-27:
Yes I'm using mosquito and since the begin I already persistance to true.

This is my configuration file 

[image]

That seems to be good as configuration ?

insipiens, 2020-11-27:
Yes, indeed. I checked some of my bulbs (esphome) and I realised that they were set to start full on in such an event as you described - a power cut or an outage of my HA environment.

closed 2021-01-04 by github-actions[bot]

kevincaradant, 2021-01-04:
Not sure this issue should be closed