Koenkk/zigbee2mqtt #5026: Issue about battery info for centralite 3400-D

kevincaradant, 2020-11-19:
Hi

I don't understand why the battery's value of my 3400D is not showed in Z2M and also in HA (by consequence) 
Log:
```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-20 00:31:34: MQTT publish: topic 'zigbee2mqtt/centralite', payload '{"action":"","linkquality":123,"temperature":22.62,"voltage":3100}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-20 00:31:34: MQTT publish: topic 'zigbee2mqtt/centralite/action', payload 'disarm'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-20 00:31:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"centralite","ieee_address":"0x000d6f000d8a5646"},"type":"device_announce"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-20 00:31:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"centralite"},"type":"device_announced"}'
```

In HA:
[image]

In Z2M:
[image]

Thank you 

Z2M Version: 1.16.1
Coordinator: CC1352P-2
Version Coordinator: 20201026

kevincaradant, 2020-12-20:
Issue still existing

Koenkk, 2020-12-20:
1. Can you try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to check logs if succesful  if not try to wakeup the device right before sending it). Battery right take up to 24 hours to report.
1. If that doesn't help try updating the devices.js definition to below and repeat step 1.

```js
{
    zigbeeModel: ['3400-D', '3400'],
    model: '3400-D',
    vendor: 'CentraLite',
    description: '3-Series security keypad',
    meta: {configureKey: 1, commandArmIncludeTransaction: true},
    fromZigbee: [fz.command_arm, fz.temperature, fz.battery],
    exposes: [e.battery(), e.temperature(), e.action([
        'disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'invalid_code', 'emergency',
    ])],
    toZigbee: [tz.arm_mode],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const clusters = ['msTemperatureMeasurement', 'genPowerCfg', 'ssIasZone', 'ssIasAce'];
        await bind(endpoint, coordinatorEndpoint, clusters);
        await configureReporting.temperature(endpoint);
        await configureReporting.batteryVoltage(endpoint);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
    onEvent: async (type, data, device) => {
        if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' &&
            globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
            const payload = {
                panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
            };
            await device.getEndpoint(1).commandResponse(
                'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
            );
        }
    },
},
```

kevincaradant, 2020-12-21:
Hi

Ok I will try to do this (step 1) but not ure to understand what should I do. 

In your UI on Z2M, do you have implemented this action ? 

Or should I try to send it since mqttExplorer for example ? 

Could you give me the complete "syntax" of the message ? 
Or simply confirm that I should execute this:
```
topic: `zigbee2mqtt/bridge/configure`
payload: 'centralite'
```

[image]

Koenkk, 2020-12-21:
From the frontend click the "gears" icon:

[image]

kevincaradant, 2020-12-23:
Step 1 didn't work.

I tried today step 2 but I don't see any difference

the first time I used the centralite I got this error in logs

```
ERROR: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot set property 'action_transaction' of undefined at Object.convert (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:816:40) at /app/lib/extension/receive.js:159:41 at Array.forEach (<anonymous>) at Receive.onZigbeeEvent (/app/lib/extension/receive.js:158:20) at Controller.callExtensionMethod (/app/lib/controller.js:381:44))
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"centralite","ieee_address":"0x000d6f000d8a5"},"type":"device_announce"}'
```

Have an idea why your code in your previous answer in this issue generate this error ?

Koenkk, 2020-12-24:
I think the code is not compatible anymore (did a lot of refactoring lately). First make sure you are on the latest dev and use the config below. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

```js
{
    zigbeeModel: ['3400-D', '3400'],
    model: '3400-D',
    vendor: 'CentraLite',
    description: '3-Series security keypad',
    meta: {configureKey: 1},
    fromZigbee: [fz.command_arm_with_transaction, fz.temperature, fz.battery],
    exposes: [e.battery(), e.temperature(), e.action([
        'disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'invalid_code', 'emergency'])],
    toZigbee: [tz.arm_mode],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const clusters = ['msTemperatureMeasurement', 'genPowerCfg', 'ssIasZone', 'ssIasAce'];
        await reporting.bind(endpoint, coordinatorEndpoint, clusters);
        await reporting.temperature(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    onEvent: async (type, data, device) => {
        if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' &&
            globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
            const payload = {
                panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
            };
            await device.getEndpoint(1).commandResponse(
                'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
            );
        }
    },
},
```

If it doesn't work provide debug logging of the error + some lines prior to that.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

kevincaradant, 2020-12-24:
I did a `docker-compose pull` before to start my docker container

I got this log (same issue)

```
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:07: Received Zigbee message from 'centralite', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:07: No converter available for '3400-D' with cluster 'ssIasAce' and type 'commandGetPanelStatus' and data '{}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:07: Received Zigbee message from 'centralite', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:07: No converter available for '3400-D' with cluster 'ssIasAce' and type 'commandGetPanelStatus' and data '{}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:09: Received Zigbee message from 'router_pc_desktop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,417000],"currentSummReceived":[0,0],"instantaneousDemand":672}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-24 21:04:09: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":108,"power":672,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:12: Received Zigbee message from 'centralite', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code":"123456","zoneid":0}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:error 2020-12-24 21:04:12: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot set property 'action_transaction' of undefined
zigbee2mqtt    |     at Object.convert (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:816:40)
zigbee2mqtt    |     at /app/lib/extension/receive.js:159:41
zigbee2mqtt    |     at Array.forEach (<anonymous>)
zigbee2mqtt    |     at Receive.onZigbeeEvent (/app/lib/extension/receive.js:158:20)
zigbee2mqtt    |     at Controller.callExtensionMethod (/app/lib/controller.js:381:44))
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:16: Device 'centralite' announced itself
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-24 21:04:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"centralite","ieee_address":"0x000d6f000d8a5646"},"type":"device_announce"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-24 21:04:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"centralite"},"type":"device_announced"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:17: Received Zigbee message from 'router_bathroom', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1000],"currentSummReceived":[0,0],"instantaneousDemand":2}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-24 21:04:17: MQTT publish: topic 'zigbee2mqtt/router_bathroom', payload '{"linkquality":93,"power":2,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-12-24 21:04:19: Received Zigbee message from 'router_pc_desktop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,417000],"currentSummReceived":[0,0],"instantaneousDemand":670}' from endpoint 1 with groupID 0
```

Of course, I edited the devices.js file with the previous code for my device before to test it :)

Koenkk, 2020-12-25:
Found the bug, fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

kevincaradant, 2020-12-25:
Nice! I will test it tonight. Merry Christmas @Koenkk :)

kevincaradant, 2020-12-25:
Are you sure it's actually fixed ? 

```
Zigbee2MQTT:debug 2020-12-25 20:30:30: No converter available for '3400-D' with cluster 'ssIasAce' and type 'commandGetPanelStatus' and data '{}'
Zigbee2MQTT:debug 2020-12-25 20:30:30: Received Zigbee message from 'centralite', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-25 20:30:30: No converter available for '3400-D' with cluster 'ssIasAce' and type 'commandGetPanelStatus' and data '{}'
Zigbee2MQTT:debug 2020-12-25 20:30:30: Received Zigbee message from 'router_pc_desktop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,422000],"currentSummReceived":[0,0],"instantaneousDemand":640}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-25 20:30:30: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"energy":4220,"linkquality":105,"power":6400,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-25 20:30:34: Received Zigbee message from 'centralite', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-25 20:30:34: No converter available for '3400-D' with cluster 'ssIasAce' and type 'commandGetPanelStatus' and data '{}'
Zigbee2MQTT:debug 2020-12-25 20:30:35: Received Zigbee message from 'centralite', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-25 20:30:35: No converter available for '3400-D' with cluster 'ssIasAce' and type 'commandGetPanelStatus' and data '{}'
```
I don't see the battery level and I can see the consumption in watt is always at 4000w instead of 400w so may I miss something ? 

I did a `dowker-compose pull` => `docker-compose down` => `docker-compose up -d` so I think I didn't miss nothing and of course I'm running on `latest-dev` tag

Koenkk, 2020-12-27:
> Are you sure it's actually fixed ? I don't see any error but I can see a no converter avaialable

Your issue was about the `commandArm` message, I dont' see it in these logs: `Zigbee2MQTT:debug 2020-12-24 21:04:12: Received Zigbee message from 'centralite', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code":"123456","zoneid":0}' from endpoint 1 with groupID 0`.

> I don't see the battery level

Did you do a reconfigure via the Zigbee2MQTT? (you should get a green popup in the bottom if succesfull). If it doesn't work I don't know anything else we can do to get the battery working

>  4000w instead of 400w about bitron video (plug AC) may I miss something ?

What plug model?

> And no action visible in Z2M UI, is it normal ?

If homeassistant: true is is your config, yes

kevincaradant, 2020-12-27:
> Did you do a reconfigure via the Zigbee2MQTT? (you should get a green popup in the bottom if succesfull). If it doesn't work I don't know anything else we can do to get the battery working

Yes already done with green popup :/

> What plug model?
Nevermind, we are already talking about it on the other issue ;)

>If homeassistant: true is is your config, yes
Alright so it's normal

kevincaradant, 2020-12-27:
[image]

How can I see it here at 2900 but in the other view I got N/A ? :/
[image]

Koenkk, 2020-12-28:
Ah I see, it reports voltage but we are talking about battery here. We have an option to convert voltage to a battery percentage. Added it know, you should get battery on the next voltage report.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

kevincaradant, 2020-12-28:
Alright, I just upgraded but I didn't see yet the 'battery' field 

`zigbee2mqtt    | Zigbee2MQTT:info  2020-12-28 22:41:06: MQTT publish: topic 'zigbee2mqtt/centralite', payload '{"action":"","linkquality":69,"temperature":20.6,"voltage":2900}'`

closed 2020-12-28 by kevincaradant