Koenkk/zigbee2mqtt #5028: Coordinator messages seems not to be handled properly in the availability extensio

muddy79, 2020-11-20:
## What happened
After enabling the availability option in configuration, I see a lot of occurrences of the same error in the log file (and in the front-end): `Failed to call 'Availability' 'onZigbeeEvent' (TypeError: Cannot read property 'friendlyName' of null`
The log file is here: https://pastebin.com/JF2FNphC

## What did you expect to happen
No error messages :) availability.js module should probably ignore messages from the coordinator (I do not know the code in deep, this is what I suspect...).

## How to reproduce it (minimal and precise)
1. Create zigbee group of lights (it's group #7 in the logged case)
2. Enable availability_timeout in the configuration (I have set it ti 300 s)
3. Call brightness change with transition
4. Coordinator generates zigbee message (seen in the log) with NWK source addr 0x0000 
5. It is generally ignored by the receiver in z2m (there is a log line "Ignoring message from coordinator")
6. But the method onZigbeeEvent from availiability.js is called anyway, there are no deviceSettings for coordinator while checking if it's not on blocklist / it's on the pass list.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: Conbee II
Adapter firmware version: latest (upgraded on Nov 2)

muddy79, 2020-11-20:
Workaround: it's enough to add the coordinator address to the configuration.yaml as a device, but I'm not sure if it does not break something else...
`  '0x0021____________':
    friendly_name: conbee_2`

Koenkk, 2020-11-21:
Should be fixed in latest dev (you can remove it from `configuration.yaml`).

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-29 by github-actions[bot]