Koenkk/zigbee2mqtt #5029: Coordinator Fails When Adding Supported Device

Pashovski, 2020-11-20:
## What happened
I am trying to add a supported device (Gledopto GL-B-008z), but it is saying it is unsupported.
https://paste.ubuntu.com/p/qHFqNwKMH3

I tried this morning, but now when I put the bulb into pairing mode, the coordinator fails, and I have to unplug it/ plug it back in. https://paste.ubuntu.com/p/ZDS22HHmWS/

## What did you expect to happen
The Device to pair
## How to reproduce it (minimal and precise)
Allow Join, start Pairing.
## Debug info
Zigbee2MQTT version:  1.16.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-11-21:
Seems that your coordinator crashes, can you check with the source routing firmware? (provides more stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

Pashovski, 2020-11-21:
> Seems that your coordinator crashes, can you check with the source routing firmware? (provides more stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

Absolutetly. I will do that Does this require setting a new pan_id and rep-pairing all the devices?

Koenkk, 2020-11-21:
No

Pashovski, 2020-11-21:
I flashed the source routing firmware. It is now showing firmware version 20190619. However, when I try to turn a light off I get a "no network route" error

https://paste.ubuntu.com/p/ygpn66pzqZ/

Pashovski, 2020-11-21:
That fixed it! Thank you so much!

closed 2020-11-21 by Pashovski