Koenkk/zigbee2mqtt #5031: Can't pair Smoke sensor Heiman HS1SA-E with cc2531

niko886, 2020-11-20:
## What happened

Hello there! I'm playing with 7 sensors and z2m, and eventually can't pair one of them Heiman HS1SA-E to cc2531. It doesn't produce any logs at all even with debug logs enabled. The last line of logs is like this:

```
Zigbee2MQTT:info  2020-11-20 18:44:06: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"7a774ef","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b000ffb829a","panID":6754},"permit_join":true,"version":"1.16.1-dev"}'
``` 

Full log sample: https://pastebin.com/Y1qespSc

I've read the FAQ, tried the extension cable and some random versions of z2m on docker (koenkk/zigbee2mqtt:1.14.0, 1.14.2, 1.14.3, latest, latest-dev, ...).

I've also tired two firmwares (Z-Stack_Home_1.2, Z-Stack_3.0.x) and tried to re-flash the device couple of times.

I've also found [some evidences](https://github.com/Koenkk/zigbee2mqtt/issues/3773) of that device working on zStack12 & 1.14.0 version.

There is actually "weird" instructions in [the manual](https://www.zigbee2mqtt.io/devices/HS1SA.html)

>Press the side button of the device with a paper clip for more than 2 seconds. The main button led will flash. Then make sure the device is awake during pairing phase or the configuration may fail. To ensure device is awake press the main button every 2 seconds until configuration is done in Zigbee2MQTT logs.

But in my case the device is starting to blink with green light rapidly and no any logs in z2m at all. When I press main button during pairing process, the siren is just going off.

## What did you expect to happen

Message in logs which would indicate that device is trying to start pairing process.

## Debug info

Adapter hardware: CC2531
Adapter firmware version: 

Z-Stack_Home_1.2
```
Zigbee2MQTT:info  2020-11-20 19:07:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"8d01a20","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.1-dev"}'
```

Z-Stack_3.0.x
```
Zigbee2MQTT:info  2020-11-20 18:41:35: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"8d01a20","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b000ffb829a","panID":6754},"permit_join":true,"version":"1.16.1-dev"}
```

Koenkk, 2020-11-21:
Make sure to also check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

niko886, 2020-11-21:
> Make sure to also check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

Hi! Yep, I've read & tried it thoroughly before asking. Although I might have missed something huge. I tried 6 other devices (not smoke sensors), all of them were pairing fine.

niko886, 2020-12-02:
Eventually I got another stick (CC2538), hooked it up to the same setup and everything worked correctly. I thought it was the problem with particular stick, but I gave it to my friend and he successfully verified that the same sensor&stick worked for him (his setup was on Pi).

So, I don't know where was the problem. Just as a suggestion: if you have the same issue you should try another, more reliable device. Thank you @Koenkk. I'll close it for now.

closed 2020-12-02 by niko886