Koenkk/zigbee2mqtt #5032: Immax 07008L - flash every X minutes

Bary, 2020-11-20:
# Bug Report
## What happened
My supported device https://www.zigbee2mqtt.io/devices/07008L.html is ON with any colour or completelly OFF. Every +- 13 minutes flash white LEDs and then return to previous colour or to previous OFF state.
**Description**
Neo SMART LED strip RGB + CCT, color, dimmable, Zigbee 3.0
**Firmware build date**
Jul 9 2018
**Firmware version**
1000-0001
**Vendor**
Immax
**Model**
07008L

## What did you expect to happen
no flashing of white LEDS
## How to reproduce it (minimal and precise)
this is ma configuration.yaml
`light:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/LEDpasek"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    xy: true
    hs: false
    schema: "json"
    command_topic: "zigbee2mqtt/LEDpasek/set"
    brightness_scale: 254
    effect: true`

on device lists I see light.mqtt_json_light (only for reading) - I am not sure that this light has something to do with LEDpasek light and light.ledpasek_light

This is in logs (maybe it is not the problem)
Zigbee2MQTT:debug 2020-11-20 20:13:07: Received Zigbee message from 'LEDpasek', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":257,"imageType":33069,"manufacturerCode":4509}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-20 20:13:15: Saving state to file /share/zigbee2mqtt/state.json

cloud icon after clicking to it writes Device 'LEDpasek' does not support OTA updates (that is correct)
## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version:

Bary, 2020-11-26:
Any news here? It is annoying. Is it possible to turn it off somehow in settings of zigbee2mqtt? thx BTW: can you add here "problem" label?

Bary, 2020-12-07:
with new version 1.16.2. seems to be ok.

closed 2020-12-07 by Bary

AuHau, 2021-02-12:
Hey,
I have the same problem now with the same device, @Bary is it still solved for you?

rotdrop, 2021-09-14:
I have a similar issue with 07004D/07005L bulbs

https://www.zigbee2mqtt.io/devices/07004D_07005L.html#immax-07004d%252F07005l

It is like they are playing sometimes an "effect" like perhaps "blink" or "channel_change" (but not quite like this).

kepi, 2022-02-09:
Same problem with three 07004D/07005L. I even tried to change the adapter to more recent device but didn't help.

This seems same as issue as reported in #4561 And from all information there it unfortunately looks like problem with bulbs. If anyone has any idea how to resolve this, would be great to hear.