Koenkk/zigbee2mqtt #5035: Best router hardware available out there

bkupidura, 2020-11-21:
In my apartment i have isssue with zigbee range. I decided to add dedicated router(s) to my network.
I have always power on devices (ex. wall switches), but i prefer to build core of my network on dedicated routers (connected to UPS for hight availability).

Currently i use ConbeeII as cordinator + 2x CC2530+CC2591 as routers.

Im happy with Conbee range, but CC* is another story ;)
I know that there is CC2652RB, but afaik there is no router firmware.
I tried to force second Conbee stick to join network as router, but i failed :)

My question is, what is best hardware available out there with router firmware?

MattWestb, 2020-11-21:
Way not adding some good zigbee 3 control outlets (plugs, but not old OSRAM then they have old TI CC-2530 inside and has not zigbee 3 stack) or some zigbee 3 bulbs ? 
Also IKEAs Signal repeater is powered with 5V so working with one power bank and is "floating around" without function for the moment but is routing the network traffic with zigbee 3 .
All "normal" router devices is routers in the network and is helping building the mesh if they is good behaving and have good radio / antennas inside.
The importing is shushing right ones so not getting problems with sleeping end devices like Xiaomi sensors. And zigbee 3 is more robust then rerouting the mesh then getting problems with interference or blocked / failing routers.

Look on the Zigbee alliance for zigbee 3 certified devices and avoiding "chinese Zigbee 3" devices that is not certified !!

Rasp/CornBee is very much overkill for being used as one "normal router" and is not bringing any advatech in the network but they have very good radios / antennas but is expensive compared consumer devices that is doing the same role in the network.

bkupidura, 2020-11-21:
To be clear, i know i can use bulbs or outlets. But from my experience they are not extending range much (i tested ikea bulbs, both gu10 and e27 - i was not happy with them). 
I own bunch (~60) end devices (most of them made by aqara/xiaomi). And they like to stick to one router.

Regarding rasp/conbee/CC2652RB price, im not looking for cheapest router. Im looking for best available.

Anybody tested IKEA repeater? How you would compare that to conbee or CC2530+CC2591 range?

MattWestb, 2020-11-21:
To finding anything that is worth then CC-253X with or without AP its not easy to day.
RaspCornBee have good processor / radio and very good ceramic antenna, its not easy finding something better if getting it working well in routing mode. (I think the router funktion is broken for some years in deCONZ and not being fixed if i remember right).
If not getting Rasp/CornBee working in routing mode then the next best I think is second gen TI NCPs with AP and the same problem finding one good working router firmware for it.

Then you is not using zigbee 3 devices (perhaps some new Xiaomi) i think Xiaomi router devices can being worth one tyre then they have the same zigbee stack and have the same setting for working together. I dont knowing if Zigate have making one router firmware but they is using Xiaomi devices as coordinator also the GW https://github.com/zigpy/zigpy-zigate/issues/59 so it can being worth look at.

Ikeas devices have all the same NCP and most the same antenna config (PCB antenna).
The hardware and radios is around or better then Rasp/CoenBee but the antena its OK but not as good as the Rasp/CoenBee ones (and its normal not optimale placed inside the devices as on the control outlet).
The antenna is always one problem then its inside one device. I was by mistake touchlink resetting one IKEA GU10 WS in my kitchen 6 meter away with deCONZ, that should not being possible with so low power that is being used then touch linking devices.

My experience of old Xiaomi sensors (not zigbee 3 ones) dont like pairing with Zigbee 3 routers that is working in real Zigbee 3 mode (not falling back to HA mode) but can moving to them and working well after having getting the network key (normal Xiaomi problem).

Shred99, 2020-11-23:
One of the best upgrades I've made to my network was to install a router with a decent aerial.  I have several Aqara door sensors monitoring outside gates that are 20 - 30m from the house.  I use a CC2530 board (no amplifier) with an RF Solutions 9dBi external antenna.  Beware that the antenna is 380mm long (massive) and it's not cheap at AU$15.  Note also that some CC2530 boards use an RP-SMA connector, so make sure you get the correct aerial for your board.

Unlike the cheap and nasty Chinese aerials with dubious specifications, this is a good quality device with some serious gain.  It is _not_ suitable if your house has multiple floors, since it will radiate very little signal up or down.

RF Solutions 9dBi 2.4GHz antenna.  Bought from RS-Online: https://au.rs-online.com/web/p/products/7934373/

Data sheet: https://docs.rs-online.com/f7f6/0900766b8129e509.pdf

MattWestb, 2020-11-23:
@Shred99 Sounds great !!
And without AP chip :-)))
On the paper the the CC-2530 with AP and internal antenna have better RF performance but as mother nature also speak chinese (murphy's law) its not very true in the real world !!

I think some (or moste) chinese CC-253X is having very bad quality components or is the manufacturing process  wrong or simply the design false made (PCB made from bad photos ?!?). If the PCB part is OK its also need one god antenna and is its crap one very good chip / AP cant helping as with all radio applications.
More users have reporting getting better signal with internal antennas then external ones then its something very wrong with the module.

If finding one good working combination pleas charing it so other user can making the same creasion that is working in real.

One more bad example is the very nice OSRAM outdoor (and indor) plug that is having one CC-2530 chip but have very short range and antenna that looks not working at all plus serious bugs in the zigbee stack that is corrupting package > children is not being pulled / acked and timing out (large Xiaomi sensor problem). I like the plug and its not easy finding good outdoor ones but i have not buying one. Perhaps i do it and putting one IKEA modul from one of there control outlet for getting it working good.

bkupidura, 2020-11-24:
So i can confirm @Shred99 words ;) I bought same 9dBi antena and i see better range 👍 I also plan to buy shorter version 5dBi (ANT-MDIP5-SMA), as long one will not fit for second router :)

I also tested some random chinese antenas (+3/6/9dBi), they work very randomly. 

To check antena gain i check RSSI reported by z2m (change from ~65-70 with oryginal 3dBi antena to ~80-85 with RF Solutions 9dBi).

markheloking, 2020-12-31:
I used to use a whole bunch of CC2531's (both as coordinator and as routers), and managed to get some issues when I started covering more area and connecting up to 10ish devices. I've now switched to a CC2538 coordinator (from [ali](https://nl.aliexpress.com/item/4001285481883.html)) and the range difference this brings me is honestly MASSIVE. I can't find a router firmware for it with a quick search atm though, so it seems that might not be out there yet. Am still using some CC2531's as routers in my network, but they're not doing much honestly, with the CC2538 coordinator absolutely outblasting them. I'm basically using them to cover deadspots.

closed 2021-02-09 by github-actions[bot]