Koenkk/zigbee2mqtt #5037: Xiaomi Aqara Opple WXCJKG13LM only button_X_single

leranp, 2020-11-21:
## What happened
The remote only gives button_X_single , and hold for buttons 3-6
The same problem like in this issue:
https://github.com/Koenkk/zigbee2mqtt/issues/2884

## What did you expect to happen
Triple, double click for all the buttons
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version:

niqu1982, 2020-11-21:
Try to push single, dual and hold action if no reset to default, pair, wekup and send configure from Z2M

leranp, 2020-11-21:
> Try to push single, dual and hold action then restart HA if no reset to default, pair, wekup and send configure from Z2M

Thanks, repair worked

closed 2020-11-21 by leranp

kiclaw, 2020-11-29:
> 
> 
> Try to push single, dual and hold action then restart HA if no reset to default, pair, wekup and send configure from Z2M

Could you please specify this operation? I can't get it. I need to restart while button pressed?

niqu1982, 2020-11-29:
after press all buttons and actions try restart HA