Koenkk/zigbee2mqtt #5040: "Set last_seen to epoch" logs constantly, eventually fills up disk

johntdyer, 2020-11-21:
## What happened

Logs are filled w/ "Set last_seen to epoch" 

```log
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.611810942Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.611827092Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.611835216Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.611842745Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.611850176Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.612085499Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.612097372Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.612102482Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.612107514Z"}
{"log":"\u001b[32mZigbee2MQTT:info \u001b[39m 2020-11-21 13:14:13: Set last_seen to epoch\n","stream":"stdout","time":"2020-11-21T18:14:13.612112551Z"}
```

Config

```yaml
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto:1883'
  client_id: zigbee2mqtt
  user: local-user
  password: xxxxx
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018ED3F35-if00
  disable_led: false
advanced:
  log_level: info
  last_seen: ISO_8601
  elapsed: true
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist:
    - '0x0017880110477f81'
    - '0x286d97000105a98a'
    - '0x0017880110472094'
    - '0x286d9700010511f8'
    - '0x0017880110478693'
    - '0x0017880110477e52'
    - '0xe20db9fffe0a748d'
    - '0xe20db9fffe0a6f8c'
    - '0xe20db9fffe0a8541'
    - '0xe20db9fffe0a4d67'
    - '0x0017880103f9d5e5'
    - '0x286d97000106f4e3'
    - '0x286d9700010cf748'
    - '0x24fd5b000107afb0'
    - '0x24fd5b000107afb0'
    - '0x286d97000109ebf4'
    - '0x286d97000109ebf4'
    - '0x286d9700010c49ad'
    - '0x0017880104b18162'
    - '0x0017880104b18162'
    - '0x0017880104b1840d'
    - '0x0017880104b1840d'
    - '0x00124b0014d987b3'
    - '0x00124b0014d95e5f'
device_options: {}
blocklist: []
passlist:
  - '0x0017880110477f81'
  - '0x286d97000105a98a'
  - '0x0017880110472094'
  - '0x286d9700010511f8'
  - '0x0017880110478693'
  - '0x0017880110477e52'
  - '0xe20db9fffe0a748d'
  - '0xe20db9fffe0a6f8c'
  - '0xe20db9fffe0a8541'
  - '0xe20db9fffe0a4d67'
  - '0x0017880103f9d5e5'
  - '0x286d97000106f4e3'
  - '0x286d9700010cf748'
  - '0x24fd5b000107afb0'
  - '0x24fd5b000107afb0'
  - '0x286d97000109ebf4'
  - '0x286d97000109ebf4'
  - '0x286d9700010c49ad'
  - '0x0017880104b18162'
  - '0x0017880104b18162'
  - '0x0017880104b1840d'
  - '0x0017880104b1840d'
  - '0x00124b0014d987b3'
  - '0x00124b0014d95e5f'
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false

```
currently the file is 20gb of these messages

## What did you expect to happen

 this to not log at info level. 

## How to reproduce it (minimal and precise)

happens constantly 

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version: latest

Koenkk, 2020-11-22:
This is only logged once Zigbee2MQTT receives a message on the following MQTT topic: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfiglast_seen .

You should check why this message is send so often (it's not something triggered by Zigbee2MQTT itself)

closed 2020-12-23 by Koenkk

mikkel75, 2021-01-04:
I also get these constantly - just did a clean check-out yesterday.
Running z2m with Domoticz and z2m-plugin...

Any suggestions?

tlc76, 2021-03-15:
Same issue here...
I managed to run zigbee2mqtt:latest official docker image without problems until I added the TZ parameter to the docker-compose.yml file.
From that moment on, at each and every startup the mqtt broker and the zigbee2mqtt log are flooded by hundreds of "Set last_seen to epoch" messages.

```
pi@pi4:/opt/zigbee2mqtt/data/log $ cat zigbee2mqtt.log
info  2021-03-15 22:45:52: Logging to console and directory: 'data/log' filename: zigbee2mqtt.log
info  2021-03-15 22:45:52: Starting Zigbee2MQTT version 1.18.1 (commit #a43d1f3)
info  2021-03-15 22:45:52: Starting zigbee-herdsman (0.13.71)
info  2021-03-15 22:45:57: zigbee-herdsman started
info  2021-03-15 22:45:57: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
info  2021-03-15 22:45:57: Currently 0 devices are joined:
warn  2021-03-15 22:45:57: `permit_join` set to  `true` in configuration.yaml.
warn  2021-03-15 22:45:57: Allowing new devices to join.
warn  2021-03-15 22:45:57: Set `permit_join` to `false` once you joined all devices.
info  2021-03-15 22:45:57: Zigbee: allowing new devices to join.
info  2021-03-15 22:45:58: Connecting to MQTT server at mqtt://localhost:1883
info  2021-03-15 22:45:58: Connected to MQTT server
info  2021-03-15 22:45:58: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"a43d1f3","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.18.1"}'
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:58: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
info  2021-03-15 22:45:59: Set last_seen to epoch
[... 61 lines cut ...]

Here is my docker-compose.yml zigbee2mqtt section:

```
  zigbee2mqtt:
    depends_on:
      - mosquitto
    image: "koenkk/zigbee2mqtt:latest"
    container_name: "zigbee2mqtt"
    restart: always
    network_mode: host
    volumes:
      - "/opt/zigbee2mqtt/data:/app/data"
      - "/run/udev:/run/udev:ro"
    devices:
      - "/dev/ttyAMA0:/dev/ttyACM0"
    environment:
      - TZ=Europe/Bucharest
    user: "${CURRENT_UID}:${CURRENT_GID}"
    privileged: true
```

Is there any solution for this, that would allow one to keep log_level on info or debug ?

thanks

tlc76, 2021-03-22:
@johntdyer, can you please reopen this issue?
I am getting the same behavior...
thanks

Koenkk, 2021-03-22:
@tlc76 this cannot be fixed from z2m, see https://github.com/Koenkk/zigbee2mqtt/issues/5040#issuecomment-731827770

mikkel75, 2021-03-23:
> This is only logged once Zigbee2MQTT receives a message on the following MQTT topic: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfiglast_seen . [...]

@Koenkk Do you mean "zigbee2mqtt/bridge/request/config/last_seen" ?

Koenkk, 2021-03-23:
Can you provide your debug log?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

tlc76, 2021-03-29:
> @tlc76 this cannot be fixed from z2m, see [#5040 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/5040#issuecomment-731827770)

Can this be somehow caused by the fact that z2m is running in a docker container?
I've never seen this behavior in a regular installation on a physical machine.
I have 2 separate z2m instances running in 2 different locations, one running in docker on a RPi 4b, the other one running natively on a RPi 3b+.

thanks

Koenkk, 2021-03-30:
@tlc76 I don't see how, if you could provide the debug log of this you will probably see that z2m receives requests to set the last seen.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

dennisgerding, 2021-05-16:
I got this issue too and found a solution :).
In the end I found out the message was caused by Zigbee2MqttAssistant because I had the ``AutosetLastSeen`` setting set to ``true``. The default value is ``false`` and after changing the setting to ``false`` the message was gone. I also didnt had any last seen issues when I switched the setting to ``false``.