Koenkk/zigbee2mqtt #5041: What is the correct nodeenv version for virtualenv?

mr2c12, 2020-11-21:
Today I updated my existing virtualenv version of z2m as I used to do.
Got z2m 16.1 (used to be on 14.1) but now I am unable to start z2m with the error

> Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
> was compiled against a different Node.js version using
> Nov 27 17:52:25 raspberrypi npm[23771]: NODE_MODULE_VERSION <can't remember the number>. This version of Node.js requires
> Nov 27 17:52:25 raspberrypi npm[23771]: NODE_MODULE_VERSION <can't remember the number>. Please try re-compiling or re-installing

Sorry for <can't remember the number> part, but I don't have the log for this error.

(previously reported at: https://github.com/Koenkk/zigbee2mqtt/issues/2425)

I tried to start from scratch and rebuild a new virtual env, following:
https://www.zigbee2mqtt.io/information/virtual_environment.html
as usual. But I ran into the very same error.

I think the problem is somehow related to the line:
`nodeenv -p -n 10.15.1`
Therefore I created a new virtualenv, but slightly changed the line above in:
`nodeenv -p`

The nodeenv has been created with node version 15.

I am still getting the error but now is:
Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 88. This version of Node.js requires
NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

I think it just a matter to set the proper nodeenv version (and update the doc).
Are you able to tell me which version to use to fix this error?

mr2c12, 2020-11-21:
Gotcha!

Look at: https://nodejs.org/en/download/releases/

NODE_MODULE_VERSION 72 requires Node 12.

When creating the nodeenv, you should use:
`nodeenv -p -n 12.12.0`

For z2m > 14.1

This: https://www.zigbee2mqtt.io/information/virtual_environment.html
should be updated :)

Koenkk, 2020-11-22:
After switching the node version you should reinstall the dependencies as stated here: `NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
the module (for instance, using npm rebuild or npm install).`

mr2c12, 2020-11-22:
Ok, in my case I just start from scratch, creating a new nodeenv in the new virtualenv.
At that point, everything was ok as usual.

Just to point out that setting the nodeenv as per the online guide won't work for z2m > 14.1.

Thanks!

closed 2020-11-22 by mr2c12