Koenkk/zigbee2mqtt #5043: [BUG] many errors (random) when calling action on devices

kevincaradant, 2020-11-21:
Hi

I saw that sometimes (today)  that I got some bulbs or others devices which doesn't do nothing during few seconds or sometimes they didn't change at all of state.

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:43:27: MQTT publish: topic 'zigbee2mqtt/switch_spots_kitchen', payload '{"linkquality":120,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-21 20:43:27: Publish 'set' 'state' to 'gledopto_kitchen' failed: 'Error: Command 0x00124b001eadcc46/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:43:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_kitchen' failed: 'Error: Command 0x00124b001eadcc46/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"gledopto_kitchen"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-21 20:43:27: Publish 'set' 'state' to 'gledopto_1_couloir_livingroom' failed: 'Error: Command 0x00124b00218120a8/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:43:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_1_couloir_livingroom' failed: 'Error: Command 0x00124b00218120a8/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"gledopto_1_couloir_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-21 20:43:27: Publish 'set' 'state' to 'gledopto_2_livingroom' failed: 'Error: Command 0x00124b0021812816/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:43:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_2_livingroom' failed: 'Error: Command 0x00124b0021812816/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"gledopto_2_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:43:27: MQTT publish: topic 'zigbee2mqtt/gledopto_1_hotte_kitchen', payload '{"brightness":242,"color":{"x":0.42,"y":0.411},"color_temp":375,"linkquality":60,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-21 20:43:27: Publish 'set' 'state' to 'gledopto_2_couloir_livingroom' failed: 'Error: Command 0x00124b00218120ac/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:43:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_2_couloir_livingroom' failed: 'Error: Command 0x00124b00218120ac/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"gledopto_2_couloir_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:43:27: MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":242,"color":{"x":0.527,"y":0.447},"color_temp":375,"linkquality":69,"state":"OFF"}'
```

AND 1 minutes later I got this:

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:17: MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":242,"color":{"x":0.527,"y":0.447},"color_temp":375,"linkquality":42,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-21 20:45:23: Publish 'set' 'state' to 'controller_leds_tv_livingroom' failed: 'Error: Command 0x00124b001fb4d55c/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 58014 - 11 - 156 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'controller_leds_tv_livingroom' failed: 'Error: Command 0x00124b001fb4d55c/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 58014 - 11 - 156 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"controller_leds_tv_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:23: MQTT publish: topic 'zigbee2mqtt/controller_leds_tv_livingroom', payload '{"brightness":90,"color":{"x":0.323,"y":0.329},"color_temp":407,"linkquality":120,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:23: MQTT publish: topic 'zigbee2mqtt/controller_leds_tv_livingroom', payload '{"brightness":90,"color":{"x":0.323,"y":0.329},"color_temp":407,"linkquality":120,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:23: MQTT publish: topic 'zigbee2mqtt/controller_leds_tv_livingroom', payload '{"brightness":90,"color":{"x":0.323,"y":0.329},"color_temp":407,"linkquality":123,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:23: MQTT publish: topic 'zigbee2mqtt/controller_leds_tv_livingroom', payload '{"brightness":90,"color":{"x":0.323,"y":0.329},"color_temp":407,"linkquality":123,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:23: MQTT publish: topic 'zigbee2mqtt/controller_leds_tv_livingroom', payload '{"brightness":90,"color":{"x":0.323,"y":0.329},"color_temp":407,"linkquality":123,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-21 20:45:23: Publish 'set' 'state' to 'controller_leds_kitchen' failed: 'Error: Command 0x00124b001f7a6063/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 64510 - 11 - 181 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:45:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'controller_leds_kitchen' failed: 'Error: Command 0x00124b001f7a6063/11 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 64510 - 11 - 181 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"controller_leds_kitchen"},"type":"zigbee_publish_error"}'
[... cut ...]
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:47:37: MQTT publish: topic 'zigbee2mqtt/gledopto_3_livingroom', payload '{"brightness":242,"color":{"x":0.323,"y":0.329},"color_temp":375,"linkquality":84,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-21 20:47:37: MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir_livingroom', payload '{"brightness":242,"color":{"x":0.323,"y":0.329},"color_temp":375,"linkquality":69,"state":"ON"}'
```

Why so many errors on different devices every time. It's not always the same devices.

What does than mean ? 

Thank you 

Z2M Version: 1.16.1
Coordinator: CC1352P-2
Version Coordinator: 20201026

kevincaradant, 2020-11-21:
This is my configuration.yaml if that could help. I don't know what happens, but it's a huge problem for me with my production installation. 

```
homeassistant: true
permit_join: true
frontend:
  port: 9090
experimental:
  new_api: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: mqtt
  password: mqtt
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 6755
  channel: 25
  baudrate: 115200
  rtscts: false
  availability_timeout: 60
```

I have 51 devices for the moment.

Koenkk, 2020-11-22:
Try the following 2 things:
- Switch to the new firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin (reported to greatly increase performance)
- Remove `availability_timeout: 60` from your configuration.yaml

kevincaradant, 2020-11-22:
Hi

I already read somewhere that `availability_timout: 60` do some bad things sometimes.
I think I added it (at the begin) to have the LQI signal (it's me in the other issue about Gledopto devices signal  ;))

I removed it for the moment as you suggest

I was waiting a stable version of your firmware in the master and also because I scared to have re-paired ALLLLLL my devices like yesterday just because I removed `availability_timout` and change cahnnel from `25` to `20` :/
Could you confirm to me if I upgrade the firmware, I will don't have to do anything about Z2M in the aim to have the same behavior that before ? :)

Any idea what happens when you read these logs ? Because all these devices had around of 130 as LQI. So no problem about the range ;)

kevincaradant, 2020-11-22:
Hop just right now it hapens again. I made many tries this morning without any issues and here it happens again:

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:40:14: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":84,"power":85.6,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:40:16: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":81,"power":85.6,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:40:24: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":84,"power":84.5,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:40:34: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":81,"power":69.9,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:40:44: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":84,"power":70.4,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:40:54: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":87,"power":70.1,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:14: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":81,"power":70.8,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:18: MQTT publish: topic 'zigbee2mqtt/controller_leds_tv_livingroom', payload '{"brightness":51,"color":{"x":0.485209769380024,"y":0.414585969195845},"color_temp":415,"linkquality":60,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:18: MQTT publish: topic 'zigbee2mqtt/gledopto_5_livingroom', payload '{"brightness":127,"color":{"x":0.462330945883173,"y":0.411139519777208},"color_temp":374,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 21:41:18: Publish 'set' 'state' to 'gledopto_3_livingroom' failed: 'Error: Command 0x00124b0021812873/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_3_livingroom' failed: 'Error: Command 0x00124b0021812873/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"gledopto_3_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 21:41:18: Publish 'set' 'state' to 'gledopto_1_couloir_livingroom' failed: 'Error: Command 0x00124b00218120a8/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_1_couloir_livingroom' failed: 'Error: Command 0x00124b00218120a8/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"gledopto_1_couloir_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 21:41:18: Publish 'set' 'state' to 'gledopto_1_hotte_kitchen' failed: 'Error: Command 0x00124b001d433487/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_1_hotte_kitchen' failed: 'Error: Command 0x00124b001d433487/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"gledopto_1_hotte_kitchen"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:18: MQTT publish: topic 'zigbee2mqtt/gledopto_kitchen', payload '{"brightness":241,"color":{"x":0.461502802724214,"y":0.41096206644152},"color_temp":373,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:41:20: MQTT publish: topic 'zigbee2mqtt/controller_leds_kitchen', payload '{"brightness":51,"color":{"x":0.49374523092787,"y":0.415134422282633},"color_temp":430,"linkquality":117,"state":"OFF"}'
```

kevincaradant, 2020-11-22:
Last thing, sometimes some devices starts to sens a lot of messages without reason.
Example here, a plug AC names `router_pc_desktop` which send it's state every 10 seconds since two hours. Why ? 

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:45:54: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":105,"power":70.6,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:04: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":105,"power":68.7,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:14: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":69,"power":69.4,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:24: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":66,"power":71.4,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:33: MQTT publish: topic 'zigbee2mqtt/thermometer_livingroom', payload '{"battery":91,"humidity":40.07,"linkquality":138,"pressure":1010.4,"temperature":20.67,"voltage":2985}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:33: MQTT publish: topic 'zigbee2mqtt/thermometer_livingroom', payload '{"battery":91,"humidity":40.41,"linkquality":144,"pressure":1010.4,"temperature":20.67,"voltage":2985}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:33: MQTT publish: topic 'zigbee2mqtt/thermometer_livingroom', payload '{"battery":91,"humidity":40.41,"linkquality":141,"pressure":1010,"temperature":20.67,"voltage":2985}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:34: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":66,"power":71.2,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:44: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":75,"power":71.5,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:51: MQTT publish: topic 'zigbee2mqtt/switch_door_bedroom', payload '{"battery":74,"linkquality":111,"voltage":2955}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:46:54: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":75,"power":69.6,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:47:04: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":66,"power":69.9,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:47:14: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":69,"power":70.6,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:47:24: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":69,"power":69.6,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:47:34: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":69,"power":69.3,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:47:35: MQTT publish: topic 'zigbee2mqtt/switch_mini_desktop', payload '{"battery":100,"linkquality":75,"voltage":3075}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:47:44: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":66,"power":70.8,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 21:47:54: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":69,"power":69.7,"state":"ON"}'
```

kevincaradant, 2020-11-22:
this is a weird example where a bulb `gledopto_2_hotte_kitchen` took 5 seconds before to switch off like others:
There is no error but this line really took 9 seconds before to be displayed:

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:39: MQTT publish: topic 'zigbee2mqtt/gledopto_5_livingroom', payload '{"brightness":127,"color":{"x":0.462330945883173,"y":0.411139519777208},"color_temp":374,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:39: MQTT publish: topic 'zigbee2mqtt/gledopto_2_livingroom', payload '{"brightness":127,"color":{"x":0.462330945883173,"y":0.411139519777208},"color_temp":374,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:39: MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir_livingroom', payload '{"brightness":127,"color":{"x":0.462330945883173,"y":0.411139519777208},"color_temp":374,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:39: MQTT publish: topic 'zigbee2mqtt/gledopto_1_hotte_kitchen', payload '{"brightness":229,"color":{"x":0.463162860122743,"y":0.411314131332149},"color_temp":375,"linkquality":42,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:39: MQTT publish: topic 'zigbee2mqtt/spots_kitchen', payload '{"linkquality":108,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:39: MQTT publish: topic 'zigbee2mqtt/router_bathroom', payload '{"linkquality":78,"power":0.2,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:44: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":99,"power":69.6,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:49: MQTT publish: topic 'zigbee2mqtt/router_bathroom', payload '{"linkquality":78,"power":0,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:50: MQTT publish: topic 'zigbee2mqtt/gledopto_2_hotte_kitchen', payload '{"brightness":229,"color":{"x":0.463162860122743,"y":0.411314131332149},"color_temp":375,"linkquality":51,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:50: MQTT publish: topic 'zigbee2mqtt/gledopto_2_hotte_kitchen', payload '{"brightness":229,"color":{"x":0.463162860122743,"y":0.411314131332149},"color_temp":375,"linkquality":48,"state":"OFF"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-23 00:36:50: MQTT publish: topic 'zigbee2mqtt/gledopto_2_hotte_kitchen', payload '{"brightness":229,"color":{"x":0.463162860122743,"y":0.411314131332149},"color_temp":375,"linkquality":48,"state":"OFF"}'
```

Koenkk, 2020-11-23:
Can you provide me the herdsman debug logging from starting z2m till these errors happen?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kevincaradant, 2020-11-23:
Hi 

I tried to enabled it but not sure if it's case or not.

If I  enable the Z2M debug and I can see it in the log of docker, it's mentioned clearly

But not sure how should I understand this sentence ? 

`Important: this is not logged to the log files and is only available on the STDOUT.`

How can I find the log of `zigbee_herdsman` ? 

I add this in my docker-compose `- DEBUG=zigbee-herdsman*`

So this is my docker-compose file:
```
  version: '3'
  services:
    zigbee2mqtt:
      security_opt:
      - apparmor=unconfined
      container_name: zigbee2mqtt
      image: koenkk/zigbee2mqtt:latest-dev
      volumes:
        - ./data:/app/data
        - /run/udev:/run/udev:ro
      devices:
        - /dev/ttyACM0:/dev/ttyACM0
      restart: unless-stopped
      network_mode: host
      privileged: true
      ports:
      - 9090:9090
      environment:
        - TZ=Europe/Paris
        - DEBUG=zigbee-herdsman*
```

Is it OK ? because I didn't see specifig log about herdsman

Koenkk, 2020-11-23:
You can find it with `docker logs -f zigbee2mqtt`.

kevincaradant, 2020-11-23:
> You can find it with `docker logs -f zigbee2mqtt`.

Yes it's already the logs that I send you since the begin. 

I saw lines like that :
```
docker logs -f zigbee2mqtt
```

Do you have a simple example of zigbee-herdsman logs ? Sorry if it's a nooby question for you but really I don't see what I should send you id the problem occurs again.

Maybe I had under my nose without understand it :)

kevincaradant, 2020-11-23:
Nevermind after two restart docker 

I finally I saw the logs, something like that :)

```
zigbee2mqtt    | 2020-11-23T21:42:06.108Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,47,68,129,0,0,2,7]
zigbee2mqtt    | 2020-11-23T21:42:06.108Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,47,68,129,0,0,2,7]
zigbee2mqtt    | 2020-11-23T21:42:06.109Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [34,92,1,1,0,75,0,141,75,63,0,0,27,24,43,10]
zigbee2mqtt    | 2020-11-23T21:42:06.109Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,47,68,129,0,0,2,7,34,92,1,1,0,75,0,141,75,63,0,0,27,24,43,10]
zigbee2mqtt    | 2020-11-23T21:42:06.109Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,4,42,93,5,0,0,0]
zigbee2mqtt    | 2020-11-23T21:42:06.109Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,47,68,129,0,0,2,7,34,92,1,1,0,75,0,141,75,63,0,0,27,24,43,10,0,4,
```

kevincaradant, 2020-11-25:
Hi 

Today the issue happens again 

```
zigbee2mqtt    | 2020-11-25T19:29:38.270Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b001f7a6063:20837/11 (0,0,13)
zigbee2mqtt    | 2020-11-25T19:29:38.270Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-25T19:29:38.271Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-25T19:29:38.271Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-25T19:29:38.271Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-25T19:29:38.271Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-25T19:29:38.271Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":48234,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":238,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,185,0]}}
zigbee2mqtt    | 2020-11-25T19:29:38.271Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,106,188,11,1,6,0,238,0,30,3,1,185,0,185]
zigbee2mqtt    | 2020-11-25T19:29:38.280Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-25T19:29:38.280Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-25T19:29:38.280Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-25T19:29:38.280Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-25T19:29:38.280Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-25T19:29:38.280Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":58286,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":239,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,186,0]}}
zigbee2mqtt    | 2020-11-25T19:29:38.280Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,174,227,1,1,6,0,239,0,30,3,1,186,0,42]
zigbee2mqtt    | 2020-11-25T19:29:38.290Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-25T19:29:38.290Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-25T19:29:38.290Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-25T19:29:38.290Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-25T19:29:38.290Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-25T19:29:38.290Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":38001,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":240,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,187,0]}}
zigbee2mqtt    | 2020-11-25T19:29:38.290Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,113,148,11,1,6,0,240,0,30,3,17,187,0,134]
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3]
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3]
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3]
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18171,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":241,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,188,0]}}
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,251,70,11,1,6,0,241,0,30,3,17,188,0,216]
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [68,128,0,1,237,43,254,25]
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,237,43,254,25]
zigbee2mqtt    | 2020-11-25T19:29:38.306Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,237] - 43
zigbee2mqtt    | 2020-11-25T19:29:38.307Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":237}
zigbee2mqtt    | 2020-11-25T19:29:38.307Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25]
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-25 20:29:38: MQTT publish: topic 'zigbee2mqtt/gledopto_kitchen', payload '{"brightness":241,"color":{"x":0.461502802724214,"y":0.41096206644152},"color_temp":373,"state":"OFF"}'
zigbee2mqtt    | 2020-11-25T19:29:38.308Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [68,129,0,0,6,0,110,190,11,1,0,81,0,61,186,78]
zigbee2mqtt    | 2020-11-25T19:29:38.308Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,6,0,110,190,11,1,0,81,0,61,186,78]
zigbee2mqtt    | 2020-11-25T19:29:38.309Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,5,24,183,11,1,0,110,190,29,245]
[... cut ...]
zigbee2mqtt    | 2020-11-25T19:29:38.392Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,243,53,254,3]
zigbee2mqtt    | 2020-11-25T19:29:38.392Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,243] - 53
zigbee2mqtt    | 2020-11-25T19:29:38.392Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":243}
zigbee2mqtt    | 2020-11-25T19:29:38.392Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3]
```

I don't know what is important, I took a lot of logs before and after the issue

Regards

kevincaradant, 2020-11-26:
Hi 

Again today right now :)

```
zigbee2mqtt    | 2020-11-26T18:30:51.927Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,40,159,11,1,6,0,19,0,30,3,17,19,0,159]
zigbee2mqtt    | 2020-11-26T18:30:51.938Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T18:30:51.938Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T18:30:51.938Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-26T18:30:51.938Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-26T18:30:51.939Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T18:30:51.939Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":34974,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":20,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,20,0]}}
zigbee2mqtt    | 2020-11-26T18:30:51.939Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,158,136,11,1,6,0,20,0,30,3,17,20,0,62]
zigbee2mqtt    | 2020-11-26T18:30:51.953Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T18:30:51.953Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T18:30:51.953Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-26T18:30:51.953Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-26T18:30:51.953Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T18:30:51.953Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":38001,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":21,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,21,0]}}
zigbee2mqtt    | 2020-11-26T18:30:51.953Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,113,148,11,1,6,0,21,0,30,3,17,21,0,205]
zigbee2mqtt    | 2020-11-26T18:30:51.969Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3]
zigbee2mqtt    | 2020-11-26T18:30:51.969Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3]
zigbee2mqtt    | 2020-11-26T18:30:51.969Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-26T18:30:51.969Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-26T18:30:51.969Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3]
zigbee2mqtt    | 2020-11-26T18:30:51.969Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":2324,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":22,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,22,0]}}
zigbee2mqtt    | 2020-11-26T18:30:51.969Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,20,9,11,1,6,0,22,0,30,3,17,22,0,53]
zigbee2mqtt    | 2020-11-26T18:30:51.970Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [68,128,241,1,21,34]
zigbee2mqtt    | 2020-11-26T18:30:51.970Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,241,1,21,34]
zigbee2mqtt    | 2020-11-26T18:30:51.970Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [241,1,21] - 34
zigbee2mqtt    | 2020-11-26T18:30:51.970Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":241,"endpoint":1,"transid":21}
zigbee2mqtt    | 2020-11-26T18:30:51.970Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T18:30:51.970Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x00124b00218120ac:38001,241,0)
zigbee2mqtt    | 2020-11-26T18:30:51.970Z zigbee-herdsman:controller:endpoint Command 0x00124b00218120ac/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-26 19:30:51: Publish 'set' 'state' to 'gledopto_2_couloir_livingroom' failed: 'Error: Command 0x00124b00218120ac/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-26 19:30:51: Error: Command 0x00124b00218120ac/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))
zigbee2mqtt    |     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
zigbee2mqtt    |     at Generator.next (<anonymous>)
zigbee2mqtt    |     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
[... cut ...] ":0,"len":5,"data":{"type":"Buffer","data":[8,15,11,0,0]}}
zigbee2mqtt    | 2020-11-26T18:30:52.121Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":15,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":0,"statusCode":0}},"address":58286,"endpoint":1,"linkquality":129,"groupID":0}'
```

Hoping that could help

Koenkk, 2020-11-26:
Thanks for the logs, can you try with the following firmware: 
[znp_CC1352P_2_LAUNCHXL_tirtos_ccs_20201126_increased_nwk_buffs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/5604599/znp_CC1352P_2_LAUNCHXL_tirtos_ccs_20201126_increased_nwk_buffs.hex.zip)

kevincaradant, 2020-11-26:
Okay I'm going to do it today :). Do you think it will be possible from your web application one day, or that will be never possible. Just to know ')

Koenkk, 2020-11-26:
Maybe in the future 😄

kevincaradant, 2020-11-26:
> Maybe in the future 😄

Haha alright :)

For the moment I try to update it but i have this mesage in unify

`Unable to open file: C:/Users/Kévin/Downloads/znp.hex`

kevincaradant, 2020-11-26:
Nop always issue. I got them with the first try to switch 8 bulbs :)

Sorry for the huge logs but at least you will find all that you need:

```
zigbee2mqtt    | 2020-11-26T19:46:16.977Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,20,9,11,1,6,0,25,0,30,3,1,17,1,44]
zigbee2mqtt    | 2020-11-26T19:46:16.978Z zigbee-herdsman:controller:endpoint Command 0x00124b0021812873/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 2324 - 11 - 10 - 6 - 11 after 10000ms)
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-26 20:46:16: Publish 'set' 'state' to 'gledopto_3_livingroom' failed: 'Error: Command 0x00124b0021812873/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 2324 - 11 - 10 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-26 20:46:16: Error: Command 0x00124b0021812873/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 2324 - 11 - 10 - 6 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:554:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:497:7)
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-26 20:46:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_3_livingroom' failed: 'Error: Command 0x00124b0021812873/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 2324 - 11 - 10 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"gledopto_3_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | 2020-11-26T19:46:16.985Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0x00124b00218126eb:7264,1)
zigbee2mqtt    | 2020-11-26T19:46:16.986Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b00218126eb:7264/11 (0,0,6)
zigbee2mqtt    | 2020-11-26T19:46:16.986Z zigbee-herdsman:controller:endpoint Command 0x00124b00218126eb/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 7264 - 11 - 14 - 6 - 11 after 10000ms)
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-26 20:46:16: Publish 'set' 'state' to 'gledopto_5_livingroom' failed: 'Error: Command 0x00124b00218126eb/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 7264 - 11 - 14 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-26 20:46:16: Error: Command 0x00124b00218126eb/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 7264 - 11 - 14 - 6 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:554:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:497:7)
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-26 20:46:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_5_livingroom' failed: 'Error: Command 0x00124b00218126eb/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 7264 - 11 - 14 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"gledopto_5_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | 2020-11-26T19:46:16.990Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:46:16.990Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:46:16.990Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-26T19:46:16.990Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-26T19:46:16.991Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T19:46:16.991Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":7264,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":26,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,21,1]}}
[... cut ...] :2994527,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,24,11,0,0]}}
zigbee2mqtt    | 2020-11-26T19:46:27.777Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":24,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":0,"statusCode":0}},"address":34974,"endpoint":11,"linkquality":72,"groupID":0}'
```

kevincaradant, 2020-11-26:
Again others logs. It happens everytime. I don't know if it's this firmware or not but it's weird that every toggle today have this problem.

With the other one it was just sometimes. Maybe just a coincidence ;)

```
zigbee2mqtt    | 2020-11-26T19:52:33.529Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,20,9,11,1,8,0,60,0,30,6,1,47,4,242,0,0,214]
zigbee2mqtt    | 2020-11-26T19:52:33.536Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:52:33.537Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:52:33.537Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-26T19:52:33.537Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-26T19:52:33.537Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T19:52:33.539Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,60,250]
zigbee2mqtt    | 2020-11-26T19:52:33.539Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,60,250]
zigbee2mqtt    | 2020-11-26T19:52:33.539Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,60] - 250
zigbee2mqtt    | 2020-11-26T19:52:33.539Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":60}
zigbee2mqtt    | 2020-11-26T19:52:33.540Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T19:52:33.545Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b00218120a8:40570/11 (1,0,6)
zigbee2mqtt    | 2020-11-26T19:52:33.546Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":40570,"destendpoint":11,"srcendpoint":1,"clusterid":8,"transid":61,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,48,4,242,0,0]}}
zigbee2mqtt    | 2020-11-26T19:52:33.546Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,122,158,11,1,8,0,61,0,30,6,1,48,4,242,0,0,49]
zigbee2mqtt    | 2020-11-26T19:52:33.553Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b0021812816:40744/11 (1,0,6)
zigbee2mqtt    | 2020-11-26T19:52:33.554Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:52:33.554Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:52:33.554Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-26T19:52:33.554Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-26T19:52:33.554Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T19:52:33.554Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":40744,"destendpoint":11,"srcendpoint":1,"clusterid":8,"transid":62,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,49,4,242,0,0]}}
zigbee2mqtt    | 2020-11-26T19:52:33.554Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,40,159,11,1,8,0,62,0,30,6,1,49,4,242,0,0,96]
zigbee2mqtt    | 2020-11-26T19:52:33.556Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,61,251]
zigbee2mqtt    | 2020-11-26T19:52:33.556Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,61,251]
zigbee2mqtt    | 2020-11-26T19:52:33.557Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,61] - 251
zigbee2mqtt    | 2020-11-26T19:52:33.557Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":61}
zigbee2mqtt    | 2020-11-26T19:52:33.557Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T19:52:33.562Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:52:33.562Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2020-11-26T19:52:33.562Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2020-11-26T19:52:33.562Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2020-11-26T19:52:33.562Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-11-26T19:52:33.566Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b00218126eb:7264/11 (1,0,6)
zigbee2mqtt    | 2020-11-26T19:52:33.566Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":7264,"destendpoint":11,"srcendpoint":1,"clusterid":8,"transid":63,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,50,4,242,0,0]}}
zigbee2mqtt    | 2020-11-26T19:52:33.566Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,96,28,11,1,8,0,63,0,30,6,1,50,4,242,0,0,169]
zigbee2mqtt    | 2020-11-26T19:52:33.568Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,62,248]
zigbee2mqtt    | 2020-11-26T19:52:33.568Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,62,248]
zigbee2mqtt    | 2020-11-26T19:52:33.568Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,62] - 248
[... cut ...]
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-26 20:52:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'gledopto_2_couloir_livingroom' failed: 'Error: Command 0x00124b00218120ac/11 genLevelCtrl.moveToLevelWithOnOff({\"level\":242,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 38001 - 11 - 51 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"gledopto_2_couloir_livingroom"},"type":"zigbee_publish_error"}'
```

Koenkk, 2020-11-27:
Looks that there are more timeout and less 241 errors now (?). Before continuing with more firmware tweaks:
- on which channel are you running? (if non specified in configuration.yaml it is 11) 
- can you post a screenshot of a wifi scan (created with an app like https://play.google.com/store/apps/details?id=com.farproc.wifi.analyzer&hl=nl&gl=US)

kevincaradant, 2020-11-27:
Ah yes Indeed there are less 'error 241'. Weirdly, on the latest logs with the spot_kitchen, I saw some 'undefined error 241' but I think it's the same issue than others with just a 'timeout'

At the begin I used channel 11 then I saw with a Wifi scanner than channel 11 was used by neighboors. 

I switched for the channel 25. Then I read somewhere than the channel 25 could be less powerful than others. I already had this issue of timeout with channel 25 so one or two weeks ago I decided to switch for the channel 20. 

It's here where I was surprised to lose every paired devices ... I don't want lose my 52 devices and re-pair them again if possible changing of channel :)

Today I'm still running with the channel 20
5Ghz
[image]

2.4Ghz
[image]

They don't use channel > 14 for me.

realmaltec, 2020-11-27:
hi
at the beginnig: my english ist not verry good and i started with zigbee 2 days ago. so, beginner is the right state for my skills.
sorry for stupid questions, I'll do my best.
i read the comments above and i think i have the described problems too. 
so i will share the little amount of informations that i have. probably it can help to analyze and solve the problem. 

### This is my configuration.yaml

> homeassistant: false
> permit_join: false
> mqtt:
>   client_id: zigbee_pi
>   base_topic: zigbee2mqtt
>   server: 'mqtt://localhost'
> serial:
>   port: '/dev/serial/by-path/platform-3f980000.usb-usb-0:1.2.4:1.0'
> '\n\nadvanced:\n    network_key': GENERATE
> devices:
>   '0x680ae2fffe788da6':
>     friendly_name: '0x680ae2fffe788da6'
>   '0x7cb03eaa0a08d9fc':
>     friendly_name: '0x7cb03eaa0a08d9fc'
>   '0x7cb03eaa0a0933c1':
>     friendly_name: '0x7cb03eaa0a0933c1'
>   '0x7cb03eaa0a0a7855':
>     friendly_name: '0x7cb03eaa0a0a7855'
>   '0x7cb03eaa00b133ff':
>     friendly_name: '0x7cb03eaa00b133ff'
>   '0x7cb03eaa00b1340b':
>     friendly_name: '0x7cb03eaa00b1340b'
>   '0x7cb03eaa0a0a7e01':
     friendly_name: '0x7cb03eaa0a0a7e01'
### startup logfile entries

> info  2020-11-26 15:25:51: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-26.15-25-50' filename: log.txt
 info  2020-11-26 15:25:52: Starting Zigbee2MQTT version 1.16.1 (commit #4978d09)
info  2020-11-26 15:25:52: Starting zigbee-herdsman...
info  2020-11-26 15:25:54: zigbee-herdsman started
info  2020-11-26 15:25:54: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'

By the way, is my firmware on the CC2531 USB dongle an old one? I see  in my logfile the revision 20190608? What is the newest stable Version?'
Under [https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html](url) i can only find the version i use.

### my error message from log.txt

> info  2020-11-26 15:30:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x7cb03eaa00b133ff' failed: 'Error: Command 0x7cb03eaa00b133ff/3 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 9670 - 3 - 13 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x7cb03eaa00b133ff"},"type":"zigbee_publish_error"}'

Could it be the same issue?

thx a lot for the whole bunch of work and supporting us!

Koenkk, 2020-11-27:
@kevincaradant zigbee channel != wifi channel, zigbee channel 20 highly interferes with wifi channel 9 (diablobox), see https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

I'm afraid I have to recommend to switch channel, channel 25 looks the best (and thus having to repair).

kevincaradant, 2020-11-27:
> @kevincaradant zigbee channel != wifi channel, zigbee channel 20 highly interferes with wifi channel 9 (diablobox), see https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html
> 
> I'm afraid I have to recommend to switch channel, channel 25 looks the best (and thus having to repair).

Oh damn okay. But just to be clear, I have exactly the same issue with channel 20 and 25. So really not sure its the cause of this problem. 
I left channel 25 for this issue ').

Koenkk, 2020-11-27:
@kevincaradant good idea, after changing diablobox to channel 1 instead of 9, do you see any improvement?

kevincaradant, 2020-11-27:
> @kevincaradant good idea, after changing diablobox to channel 1 instead of 9, do you see any improvement?

I reply to you tonight (after 5 PM which is the hour where automation with lights is ON :))

At home, it's 03:22 PM so I need to wait around of 2 hours :)

kevincaradant, 2020-11-27:
@Koenkk , just to know, why I never hear until now that the channel 26 exist ? :)

It could be my alternative if I got always issues ? 

Thank you to confirm me that channel 26 is supported by devices and Z2M

kevincaradant, 2020-11-27:
@realmaltec 

Hi, usually we don't merge issue of someone else to avoid multi conversations in the same time.

However, yes you are using  the latest driver. Default 1.2. ( I don't know more recent firmware until this day)

And like me, please check what channel are you using from your wifi and also (I think my issue come from this, the wifi network during the day).
I just saw that I have a reallllllly close neighboor (only the night, that's a little bit tricky for me to see that) which use a wifi channel which can do some interference with mine ...

After to analyze the iwif channel used by you and your neighboors, please use a specific channel for your zigbee coordinator. 
You can specify it in the configuration.yaml file.

Tell us if it's better or not like that .

Koenkk, 2020-11-28:
> Is it possible to increase the range of CC1352-P2 a little bit more, to avoid a little bit more the interferences issues ? :)

With the firmware you are running the CC1352P2 is on 20dbm (which is the max). I don't think power is restricted on channel 25. Can you try switching back to https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin and see if the issue does not re-appear?

kevincaradant, 2020-11-28:
> With the firmware you are running the CC1352P2 is on 20dbm (which is the max). I don't think power is restricted on channel 25. Can you try switching back to [...]

Everything works so well, I'm scared to revert on the aother firmware :P

But Okay, for you, I will do it ;)

kevincaradant, 2020-11-28:
Well I just rollback and I got again so many issues :/ 

```
igbee2mqtt    | Zigbee2MQTT:error 2020-11-28 20:07:35: Publish 'set' 'state' to 'gledopto_kitchen' failed: 'Error: Command 0x00124b001eadcc46/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 35702 - 11 - 41 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 20:07:35: Error: Command 0x00124b001eadcc46/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 35702 - 11 - 41 - 6 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:554:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:497:7)
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-28 20:07:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gledopto_kitchen' failed: 'Error: Command 0x00124b001eadcc46/11 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 35702 - 11 - 41 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"gledopto_kitchen"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 20:07:43: Received Zigbee message from 'router_pc_desktop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,94000],"currentSummReceived":[0,0],"instantaneousDemand":721}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-28 20:07:43: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":87,"power":72.1,"state":"ON"}'
```

The beta firmware was better. Is it normal ?

Koenkk, 2020-11-29:
I've read somewhere that not all devices support channel 26 but I've never seen one yet.

So the fw from: https://github.com/Koenkk/zigbee2mqtt/issues/5043#issuecomment-734440277 works better?

kevincaradant, 2020-11-29:
> I've read somewhere that not all devices support channel 26 but I've never seen one yet.
> 
> So the fw from: [#5043 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/5043#issuecomment-734440277) works better?

Right now, the old one works (only one error but I played with state on/off really quickly so it could be normal ?), really hard to say why it's not so bad as yesterday ... but Yesterday just after to update it, you can see in the previous message what's happens with logs and it was ugly. 
So yes I think the new one was better because I didn't had a timeout with it but I'm a little bit confused and I don't know if it's normal that your beta firmware seems to work better. In all the cases I prefer it ;)

realmaltec, 2020-11-30:
> @realmaltec
> 
> Hi, usually we don't merge issue of someone else to avoid multi conversations in the same time. But maybe it's the same issue so we can analyze it here. If it's not the same issue, I think you will need to create a new one ;)
> 
> However, yes you are using the latest firmware. Default 1.2. ( I don't know more recent firmware for CC2531 until this day)
> 
> And like me, please check what channel are you using from your wifi and also (I think my issue come from this, the wifi network during the day of mine and neighbours).
> I just saw that I have a reallllllly close neighbour (only the night, that's a little bit tricky for me to see that) which use a wifi channel which can do some interference with mine...
> 
> After to analyze the wifi channel used by you and your neighboors, please use a specific / optimized channel for your zigbee coordinator.
> Use this picture to help you to know what channel you should use:
> [image]
> To summarize, in 90% cases, you have to choose between the channel 15 / 20 / 25 for your coordinator.
> Then you can specify it in the configuration.yaml file.
> 
> Tell us if it's better or not like that .
@Koenkk 
hi,
i tried the last days to reproduce the bug without success. 
Its working fine, but i didn`t changed anything?!  I`ll wait an try to analyze it if it comes back and then i will change the Channel as described in your answer. (I`m usinf Wi-Fi on Channel 11 and a scan shows no traffic in the Range of the channels above 21. So i decided for the future to use Channel 25 for the zigbee environment. )
thx for your advice

Koenkk, 2020-11-30:
@kevincaradant can you try with this fw (should make it better or stay equal) to the fw I provided earlier. 
[znp_CC1352P_2_LAUNCHXL_tirtos_ccs_20201130_increased_nwk_buffs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/5617872/znp_CC1352P_2_LAUNCHXL_tirtos_ccs_20201130_increased_nwk_buffs.hex.zip)

kevincaradant, 2020-11-30:
Hi @Koenkk 

I just updated, right now with your latest firmware, restarted Z2M

And I still got issues timeout. My kitchen bulbs took 45 seconds to start.

```
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-30 21:08:51: Publish 'set' 'state' to 'controller_leds_tv_livingroom' failed: 'Error: Command 0x00124b001fb4d55c/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 48750 - 11 - 3 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-30 21:08:51: Error: Command 0x00124b001fb4d55c/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 48750 - 11 - 3 - 6 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:554:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:497:7)
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-30 21:08:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'controller_leds_tv_livingroom' failed: 'Error: Command 0x00124b001fb4d55c/11 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 48750 - 11 - 3 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"controller_leds_tv_livingroom"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-30 21:08:53: Received Zigbee message from 'router_pc_desktop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,117000],"currentSummReceived":[0,0],"instantaneousDemand":1102}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-30 21:08:53: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":129,"power":110.2,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-30 21:08:55: Received Zigbee message from 'router_pc_desktop', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,117000],"currentSummReceived":[0,0],"instantaneousDemand":1097}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-30 21:08:55: MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"linkquality":60,"power":109.7,"state":"ON"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-30 21:09:00: Publish 'set' 'state' to 'controller_leds_kitchen' failed: 'Error: Command 0x00124b001f7a6063/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 20837 - 11 - 11 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-30 21:09:00: Error: Command 0x00124b001f7a6063/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 20837 - 11 - 11 - 6 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:554:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:497:7)
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-30 21:09:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'controller_leds_kitchen' failed: 'Error: Command 0x00124b001f7a6063/11 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 20837 - 11 - 11 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"controller_leds_kitchen"},"type":"zigbee_publish_error"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-30 21:09:00: Publish 'set' 'state' to 'gledopto_kitchen' failed: 'Error: Command 0x00124b001eadcc46/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 35702 - 11 - 13 - 6 - 11 after 10000ms)'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-30 21:09:00: Error: Command 0x00124b001eadcc46/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 35702 - 11 - 13 - 6 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:554:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:497:7)
[... cut ...]
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-30 21:09:00: Error: Command 0x00124b00218124fd/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 18171 - 11 - 14 - 6 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:554:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:497:7)
```

Regards

Koenkk, 2020-12-01:
Can you try adding the following to your `configuration.yaml`:

```
advanced:
  adapter_concurrent: 1
```

And check if things stay stable?

kevincaradant, 2020-12-01:
> adapter_concurrent: 1

I saw that today in an other thread where the man put `adapter_concurrent: 36` as valkue. What is this parameter ? Ok I will try to add it and see if it's more stable. Currently I'm running under the latest firmware that you give me ;)

sjorge, 2020-12-01:
From https://www.zigbee2mqtt.io/information/configuration.html

```
  # Optional: configure adapter concurrency (e.g. 2 for CC2531 or 16 for CC26X2R1) (default: null, uses recommended value)
  adapter_concurrent: null
```

kevincaradant, 2020-12-01:
Hi @sjorge 

Indeed, I already read this documentation but as I said, no mention about what does it ? or why we should use it ? :D

But your edit message is nice, I understand at least, this parameter, avoid some trouble using devices together in the same time.

It's too early to say anything but at least tonight, no issue at all (cross fingers) for the next days.

Thank you ;)

sjorge, 2020-12-01:
It how many actions in parallel the coordinator will try to execute. (As I understand it, it does not effect incoming messages, just outgoing)

For me dropping seems to result in the initial reporting messages after configure of trådfri bulbs to be more spread out (would make sense if only one bulb gets configure instead of two at once) and I avoid the odd buffet_full that I would get otherwise.

With it set to 2, I’d have 1/3 change of hitting buffer full of 4 bulbs announced at once. Was playing with it on 1 and toggled the set of bulbs 8 times without issue.

kevincaradant, 2020-12-01:
Nice explanations thanks !, I think this could be help a lot indeed. All my issues come from when automations play with more than one device at the same time :)

kevincaradant, 2020-12-09:
Well @Koenkk. I think I'm going to close this issue for the moment because, no issue since one week.
Today I'm running under 60 devices and it's still stable.

If someone need to know what I done to fix this problem.
The fix could be one of these actions (I don't know whitch one):
   - Upgrade my NUC From Ubuntu 16.04 => 20.04
   - Coordinator more far away from the Wifi router (2 meters instead 50cm) 
   - Add USB Cable 3.0 (4 meters in total)
   - Use the firmware than @Koenkk gives some messages before this one.
   - Use `availability_timeout: 60` option ? (if it doesn't help on timeout issue, it's seems to help me about my end-devices Xiaomi which left sometimes the network after few days.
   - Use the `adapter_concurrency: 1` option
   - Avoid at the maximum to use Wifi 2.4Ghz at home. Use 5.0Ghz instead. But I have always 4 devices in 2.4Ghz (vaccum, Air Conditioner) because they doesn't support 5.0Ghz.
   - Check my network Zigbee is not superposed to the Wifi network (mine and neighboors)
   - My prayer ? :D

closed 2020-12-09 by kevincaradant

desinox, 2022-06-14:
Hey, 
I have an similar issue, I run hass os on an raspi 4 with an conbee II usb stick.

The Errors I get are all similar to that: 
`2022-06-14 03:38:22Publish 'set' 'color_temp' to 'PhilipsDeckenlampe2' failed: 'Error: Command 0x001788010b54a845/11 lightingColorCtrl.moveToColorTemp({"colortemp":383,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-06-14 03:38:22Publish 'set' 'color_temp' to 'Schreibtischlampe' failed: 'Error: Command 0x0017880109aefc44/11 lightingColorCtrl.moveToColorTemp({"colortemp":383,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-06-14 03:38:22Publish 'set' 'brightness' to 'PhilipsDeckenlampe4' failed: 'Error: Command 0x0017880109d95485/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-06-14 03:38:22Publish 'set' 'brightness' to 'PhilipsDeckenlampe3' failed: 'Error: Command 0x0017880109e15ebf/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-06-14 03:38:27Publish 'set' 'brightness' to 'PhilipsDeckenlampe4' failed: 'Error: Command 0x0017880109d95485/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-06-14 03:38:27Publish 'set' 'brightness' to 'PhilipsDeckenlampe3' failed: 'Error: Command 0x0017880109e15ebf/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error 2022-06-14 03:38:49Publish 'set' 'color_temp' to 'Schreibtischlampe' failed: 'Error: Command 0x0017880109aefc44/11 lightingColorCtrl.moveToColorTemp({"colortemp":383,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

I tried changing channel and the adapter_concurrency to 1 and updated the conbee stick to the newest firmware but the error still remains. 
Link Quality ist on all devices at 170 or higher.

I hope anyone can help me, this is very frustrating...