Koenkk/zigbee2mqtt #5051: After started Z2M =>  Failed to execute LQI for 'XXXX' on Gledopto devices

kevincaradant, 2020-11-22:
Hi

When I started Zigbe2mqtt, everything is OK (that solved the issue about unavailable on HA after a restart of HA) but not the LQI report.

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 12:24:15: Starting network scan (includeRoutes 'false')
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_1_hotte_kitchen'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_2_hotte_kitchen'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_1_livingroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_3_livingroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_2_livingroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_5_livingroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_4_livingroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_kitchen'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_2_couloir_livingroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:15: Failed to execute LQI for 'gledopto_1_couloir_livingroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:16: Failed to execute LQI for 'controller_leds_kitchen'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:16: Failed to execute LQI for 'gledopto_3_bathroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:16: Failed to execute LQI for 'gledopto_2_bathroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:16: Failed to execute LQI for 'gledopto_1_bathroom'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-22 12:24:16: Failed to execute LQI for 'controller_leds_tv_livingroom'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-22 12:24:16: Network scan finished
```

I'm running on the latest version of Zigbee2MQTT (1.16.1-dev).

I removed `availability_timeout: 60` from configuration.yaml, is it wrong or right to add it ? Not sure to undertsand the real utility of it.

And now, not sure why (maybe the avaibility feature that I disabled ?) I have only 0 lqi on each gledopto devices in HA.
Even after to switch ON / OFF them ;)

[image]

Thank you to look about it

Koenkk, 2020-11-22:
Most gledopto devices don't support LQI scans so this is expected.

kevincaradant, 2020-11-22:
Alright, why before I had LQI signal ?
The "availability_timeout" could be the reason ?  

So we can do nothing ? I will have always 0 LQI as signal in HA ? 

Thank you :)

Koenkk, 2020-11-23:
@kevincaradant the linkquality should be updated each time you send a command to a device (even for gledopto, just tested with my gledopto led controller). Easiest way to test is to open the frontend and to turn on/off your device there, each time the LQI value should change. Make sure you are on the dev branch to test the latest z2m version (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

[image]

kevincaradant, 2020-11-23:
Hi

yes I know about the rules to get LQI changing the state :)

But I had some devices which doesn't report value in Z2M :). My Gledopto Led Controller have it (LQI) . But the Gledopto bulbs doesn't :(. Before I had values so I don't understand what changed. The only feature possible is that I removed `availability_timeout: 60` but really not sure it has a link with this issue

[image]

[image]

And that's the issue that I wanted mention :)

Koenkk, 2020-11-23:
What is the model of these spots?

kevincaradant, 2020-11-23:
Not sure to understand your question because you yourself answer it but yes it's a GL-S-007Z =>  GU10 bulb. It's a spot  :)

Koenkk, 2020-11-23:
The reason why you don't get LQI is because these devices don't provide a default response (acknowledgment to a command, this is where the LQI is taken from): https://github.com/Koenkk/zigbee-herdsman-converters/blob/7201ea3820ea780a9dcfcbdfaf1da8b4a5d10ab1/devices.js#L6904 . I'm not sure if it was added correctly there, you can try to remove it and see if you can still properly control the bulb. 

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

kevincaradant, 2020-11-23:
Ok I need times to understand what I need to edit / change. 

Firstly I understood that I need to remove the code related to my device in `device.js` and . Need to see how to change `devices.js` via docker-compose. 
If I remove my device from devices.js, I should have LQI and still be able to use my bulbs ? 

I never modify devices.js, my knowledges are alight on this subject.

Koenkk, 2020-11-24:
Don't remove the complete device (as this will make your device appear as unsupported), but only remove the following line: `meta: {disableDefaultResponse: true},`

kevincaradant, 2020-11-25:
Hi

It's working now after to have removed `{disableDefaultResponse: true},` :)

So I don't know what I should understand but it works like that. Maybe you could modify it directly from your devices.js to avoid to have ours on our side ;)

Koenkk, 2020-11-26:
Removed it, I expect all devices having disableDefaultResponse to have this issue.

kevincaradant, 2020-11-26:
Cool, tell me when it will be available. I think tomorrow during a nightly build with docker? I upgraded my docker two hours ago but its doesn't work yet in the latest-dev tag :)

Koenkk, 2020-11-26:
in +- 1 hour from now.

kevincaradant, 2020-11-27:
It's works as expected on the dev branch. I think we can close this issue. I will open it again if I see something weird

kevincaradant, 2022-01-16:
Hi @Koenkk 

Sorry to reopen it but someone has change something about this issue you think ? 

I got 'NuL' as value.

Logs: 
```
Debug Publishing 'set' 'state' to 'Z2M_livingroom_lights'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_livingroom', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"linkquality":76,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/Z2M_livingroom_lights', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_livingroom', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"linkquality":149,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_3_livingroom', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"linkquality":null,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_4_livingroom', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"linkquality":null,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_5_livingroom', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"linkquality":null,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir_livingroom', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"linkquality":null,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/gledopto_2_couloir_livingroom', payload '{"brightness":110,"color":{"x":0.4632,"y":0.4113},"color_mode":"color_temp","color_temp":375,"linkquality":null,"state":"OFF"}'
Info 
```

Thank

kevincaradant, 2022-01-16:
Oh that's weird, it's because if I switch ON / OFF ( one by one ) like I did on the gledopto_1 and gledopto_2 it's working.

But now if I switch ON / OFF, the Z2M group, it's not working and it won't updated the LQI value.

I don't know if it's helping or not.

kevincaradant, 2022-02-16:
Still having issue. Regards

Koenkk, 2022-02-16:
What are the model you get linkquality null for?

kevincaradant, 2022-02-16:
I got this issue on 
GL-S-007Z
But I'm not sure it's related to the model. As I exaplained in the previous message, I got this issue only when it's used as a group of lights. 
Livingroom toogle = 6 gledopto lights.

If I trigger them one by one, I got the LQI on the Z2M UI ;)

I hope I was clear

Koenkk, 2022-02-16:
When a light is controlled to a group the LQI will not be updated. This is because no command is send back to z2m when controlled to a group (which is the case when controlled individually)

kevincaradant, 2022-02-16:
OK thank you

That's weird, it's working for the (hottes lights, the  two first lights on my picture) , it's also a group and I didn't ping them individually to control them, I got the LQI as wanted. 
WHat al I missing ?

Koenkk, 2022-02-18:
> OK so no workaround possible to avoid this ? It's the normal behavior for you ?

yes this is expected behaviour. (as explained in https://github.com/Koenkk/zigbee2mqtt/issues/5051#issuecomment-1042281136)