Koenkk/zigbee2mqtt #5053: Xiaomi Aqara RTCGQ01LM/RTCGQ11LM Occupancy Sensors stop working

guardiande, 2020-11-22:
## What happened

**Disclaimer:** I know that there are already many issues around those devices.

All three occupany sensors I have (1 RTCGQ01LM and 2 RTCGQ11LM) stopped working more or less at the same time.

## What did you expect to happen

All occupany sensors to report occupany without interruptions.

## How to reproduce it (minimal and precise)

I cannot reproduce the behaviour. I happened in the past regularly and did not for a longer time now.

Here are my observations:

* Resetting the controller does not help.
* Restarting Z2M does not help.
* Unplugging the controller and restarting Z2M seems to help for some time sometimes.

What seemed to have helped:

* I always have my OSRAM smart plug under suspicion. Therefore I unplugged it.
* Unplugged the stick and restarted Z2M.
* Shortly pressed the button on the occupancy sensors.

Here's a sniff:

[image]

And here Z2M logs:

```
2020-11-22.09-44-28/log.txt:info  2020-11-22 10:16:25: MQTT publish: topic 'zigbee2mqtt/Motion3', payload '{"battery":100,"linkquality":0,"occupancy":false,"voltage":3005}'
2020-11-22.10-24-21/log.txt:info  2020-11-22 10:24:25: MQTT publish: topic 'zigbee2mqtt/Motion3', payload '{"battery":100,"linkquality":10,"occupancy":false,"voltage":3005}'
2020-11-22.13-06-58/log.txt:info  2020-11-22 13:07:03: MQTT publish: topic 'zigbee2mqtt/Motion3', payload '{"battery":100,"linkquality":10,"occupancy":false,"voltage":3005}'
2020-11-22.13-07-19/log.txt:info  2020-11-22 13:07:23: MQTT publish: topic 'zigbee2mqtt/Motion3', payload '{"battery":100,"linkquality":10,"occupancy":false,"voltage":3005}'
2020-11-22.13-07-19/log.txt:info  2020-11-22 13:09:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Motion3","ieee_address":"0x00158d0002c0e39e"},"type":"device_announce"}'
2020-11-22.13-07-19/log.txt:info  2020-11-22 13:09:15: MQTT publish: topic 'zigbee2mqtt/Motion3', payload '{"battery":100,"linkquality":94,"occupancy":false,"voltage":3005}'
2020-11-22.13-07-19/log.txt:info  2020-11-22 13:09:21: MQTT publish: topic 'zigbee2mqtt/Motion3', payload '{"battery":100,"linkquality":47,"occupancy":true,"voltage":3005}'
```

Note:

* Observe that the first event in Z2M from the sensor occurred at 13:07:23 after restarting Z2M.
* The rejoin happened at 13:09:10 (visible in dump and logs).
* There were several unsuccessful transmit in the dump before 13:07:23.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version: 20190619

tetti-de, 2020-11-22:
I had the same problem with my sensor. Conclusion: battery too weak.

guardiande, 2020-11-22:
@tetti-de, did you happen to check the battery level or voltage level?

I record them since over a year, I don't remember when I first activate the devices though... It seems to me that the battery attribute (almost always at 100%) and the voltage attribute (low variation) are not very useful for determining battery depletion.

[image]

tetti-de, 2020-11-22:
@guardiande Yes I did. Battery level 51%, voltage 2915. The new CR2450 shows level 100% and voltage 3235 (today 3135 after 2 month of operation)
The original battery starts at 100%, 3075 in march 2020.  My sensor stopped proper working end of september 2020

marc-gist, 2020-11-24:
One of two things in my opinion.  Signal strength (add a router near it, or move it for a day or two near your  Coordinator).
If its not signal. I have a number of these (6). and One would constantly drop off after a day or two.  new battery and moving it close didn't work. I threw it out :) so could just be a bad one.

gemignani, 2020-11-26:
I have 6 of the same devices working with zigbee2mqtt for over 18 months and recently one RTCGQ11LM dropped and it won't stay on the network, I've tried all the usual things: not sharing wifi channels, bring close to cc2531 to join, reboot / restart.
Unfortunately the sensor always stop reporting after 4-6 hours, but if I leave them just close to my PC desk, they work fine...

igosoft, 2020-11-28:
Similar to this one #4999

closed 2021-01-05 by github-actions[bot]