Koenkk/zigbee2mqtt #5054: Stop adding `_light` qualifier to the end of light entities in hassio

ChipWolf, 2020-11-22:
## Is your feature request related to a problem? Please describe
Currently, any lights which are added to hass.io from zigbee2mqtt are suffixed with `_light`.
The entity is added to the `light.` namespace which leaves all lights with a redundant qualifier. _(ie. `light.bedroom_lamp_light`)_

## Describe the solution you'd like
I would like this `_light` qualifier to not be added to new lights.

## Describe alternatives you've considered
Manually modifying each light or manually defining lights in hass.io config.

## Additional context
The automatic addition of entities to hass.io from zigbee2mqtt is very useful, however, when other integrations also make use of these names and they're included in the display names it can make everything a little messy with overrides and the occasional **Alexa, turn off the bedroom lamp light**

Koenkk, 2020-11-23:
Implemented.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-23 by Koenkk

SamJongenelen, 2020-12-01:
Ah, good to know. 

Will this break all my automations using the suffix included in the entityid, homeassistant?

Koenkk, 2020-12-02:
@SamJongenelen no, Home Assistant will only use this name on initial discovery.