Koenkk/zigbee2mqtt #5056: Xiaomi MCCGQ11LM not reflected corrrectly in Home Assistant

taugusti, 2020-11-22:
## What happened
I have several Xiaomi MCCGQ11LM door/window switches that does not report "contact" in the _contact device in hass.
in the z2m plugin it looks fine:
[image]
but in the "developer tools"-section it is not reflected:
[image]
Please note that the "contact" attribute reflects the state correctly. Opening/ closing the window does not change the value of the device - only the contact attribute.
All other devices (Xiaomi and Ikea) reflect their status in the value and not only in the attributes.

Device discovery mgs on mqtt:
topic: homeassistant/binary_sensor/0x00158d0004a02717/contact/config
availability:
  - topic: zigbee2mqtt/bridge/state
  - topic: zigbee2mqtt/Badeværelse vindue/availability
device:
  identifiers:
    - zigbee2mqtt_0x00158d0004a02717
  manufacturer: Xiaomi
  model: Aqara door & window contact sensor (MCCGQ11LM)
  name: Badeværelse vindue
  sw_version: Zigbee2MQTT 1.16.1
device_class: door
json_attributes_topic: zigbee2mqtt/Badeværelse vindue
name: Badeværelse vindue contact
payload_off: true
payload_on: false
state_topic: zigbee2mqtt/Badeværelse vindue
unique_id: 0x00158d0004a02717_contact_zigbee2mqtt
value_template: '{{ value_json.contact }}'

open window:
topic: zigbee2mqtt/Badeværelse vindue
battery: 100
contact: false
linkquality: 177
voltage: 3065

close window:
topic: zigbee2mqtt/Badeværelse vindue
battery: 100
contact: true
linkquality: 177
voltage: 3065

I have compared to a Xiaomi pir-sensor but I cannot find anything that seems to be linked to my problem.

Looking at the landsman log also does not give me any clues.
Here the config msg for the device above:
https://pastebin.com/1kmcLxMJ

Here the complete log (note that there are unfortunately no state changes for the device in this log)
https://pastebin.com/snMxzkJ8

## What did you expect to happen
That the contact attribute is reflected in the device value.

## How to reproduce it (minimal and precise)
1. Pair a Xiaomi MCCGQ11LM
2. Open / close the window / door 
3. See that the _contact device does not reflect the state of the door / window switch.
4. check that the contact appribute on the _contact device reflects the state of the door / window switch.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: Conbee II
Adapter firmware version: deCONZ_ConBeeII_0x26580700.bin

Koenkk, 2020-11-22:
This looks like a Home Assistant bug to me.

taugusti, 2020-11-22:
I have just been instructed to report it here by @ciotlosm on the mqtt plugin-repo.
If neither there nor here then where?

ciotlosm, 2020-11-23:
@taugusti I've redirected you here as the repository where you raised the initial issue is for the add-on that does mostly wrapping of config and Supervisor integration. 

The mqtt integration is handled mostly by the Z2M core (this repo), however if it is indeed a Home Assistant bug, you can use https://github.com/home-assistant/core/issues to search for similar issues or report your issue there. 

We're trying to redirect users to raise issues in the right place to get most benefit for getting it resolved.

taugusti, 2020-11-23:
> @taugusti I've redirected you here as the repository where you raised the initial issue is for the add-on that does mostly wrapping of config and Supervisor integration.
>
> The mqtt integration is [...]

Thanks. I will try there.

closed 2020-11-23 by taugusti